open-power/op-build #690: Backtrace while shuttting down to kexec

shenki, 2016-10-12:
Kernel 4.8, with an experimental xhci patch:

```
[   87.188320] DELAYING BEFORE HALTING
[  108.192819] INFO: rcu_sched self-detected stall on CPU
[  108.193265] 	5-...: (2098 ticks this GP) idle=b43/140000000000001/0 softirq=105/105 fqs=1050 
[  108.193846] 	 (t=2100 jiffies g=3 c=2 q=2)
[  108.194254] Task dump for CPU 5:
[  108.194670] usb-storage     R  running task        0  3385      2 0x00000804
[  108.195198] Call Trace:
[  108.195678] [c000001fd0bdb0b0] [c0000000000864e8] sched_show_task+0xe4/0x118 (unreliable)
[  108.196322] [c000001fd0bdb120] [c000000000bee988] rcu_dump_cpu_stacks+0xdc/0xec
[  108.197130] [c000001fd0bdb170] [c0000000000aef14] rcu_check_callbacks+0x2f4/0x794
[  108.197833] [c000001fd0bdb290] [c0000000000b15f4] update_process_times+0x3c/0x74
[  108.198556] [c000001fd0bdb2c0] [c0000000000c0a28] tick_sched_handle.isra.3+0x60/0x6c
[  108.199314] [c000001fd0bdb2f0] [c0000000000c0a80] tick_sched_timer+0x4c/0x84
[  108.200082] [c000001fd0bdb330] [c0000000000b2164] __hrtimer_run_queues+0xe4/0x184
[  108.200897] [c000001fd0bdb3b0] [c0000000000b241c] hrtimer_interrupt+0xb0/0x1f4
[  108.201742] [c000001fd0bdb470] [c000000000017010] __timer_interrupt+0x7c/0xdc
[  108.202592] [c000001fd0bdb4b0] [c0000000000171b0] timer_interrupt+0xa8/0xc0
[  108.203542] [c000001fd0bdb4e0] [c000000000002328] decrementer_common+0x128/0x180
[  108.204467] --- interrupt: 901 at _raw_spin_lock_irqsave+0x64/0x90
[  108.204467]     LR = _raw_spin_lock_irqsave+0x5c/0x90
[  108.206354] [c000001fd0bdb7d0] [c000001fd6a9a000] 0xc000001fd6a9a000 (unreliable)
[  108.207424] [c000001fd0bdb810] [c000000000a59bc4] xhci_urb_enqueue+0x418/0x568
[  108.208524] [c000001fd0bdb8e0] [c000000000a32370] usb_hcd_submit_urb+0x84c/0x880
[  108.209653] [c000001fd0bdba20] [c000000000a33438] usb_submit_urb+0x444/0x494
[  108.210799] [c000001fd0bdbac0] [c000000000a68424] usb_stor_msg_common+0x98/0x158
[  108.211986] [c000001fd0bdbb20] [c000000000a68878] usb_stor_bulk_transfer_buf+0x58/0xa4
[  108.212821] INFO: rcu_sched detected stalls on CPUs/tasks:
[  108.215219] [c000001fd0bdbb60] [c000000000a68f90] usb_stor_Bulk_transport+0xe0/0x35c
[  108.217189] [c000001fd0bdbbc0] [c000000000a693f0] usb_stor_invoke_transport+0x38/0x524
[  108.219202] [c000001fd0bdbca0] [c000000000a68090] usb_stor_transparent_scsi_command+0x18/0x2c
[  108.221268] [c000001fd0bdbcc0] [c000000000a6aa94] usb_stor_control_thread+0x29c/0x2bc
[  108.223383] [c000001fd0bdbd80] [c00000000007c1a8] kthread+0x100/0x108
[  108.225523] [c000001fd0bdbe30] [c00000000000986c] ret_from_kernel_thread+0x5c/0x70
[  108.227741] 	5-...: (2098 ticks this GP) idle=b43/140000000000001/0 softirq=105/105 fqs=1050 
[  108.230488] 	(detected by 4, t=2103 jiffies, g=3, c=2, q=2)
[  108.232156] Task dump for CPU 5:
[  108.234141] usb-storage     R  running task        0  3385      2 0x00000804
[  108.235907] Call Trace:
[  108.237623] [c000001fd0bdb850] [c000001ffe10cd00] 0xc000001ffe10cd00 (unreliable)
[  108.239499] [c000001fd0bdba20] [c000000000a33438] usb_submit_urb+0x444/0x494
[  108.241355] [c000001fd0bdbac0] [c000000000a68424] usb_stor_msg_common+0x98/0x158
[  108.243453] [c000001fd0bdbb20] [c000000000a68878] usb_stor_bulk_transfer_buf+0x58/0xa4
[  108.246221] [c000001fd0bdbb60] [c000000000a68f90] usb_stor_Bulk_transport+0xe0/0x35c
[  108.249005] [c000001fd0bdbbc0] [c000000000a693f0] usb_stor_invoke_transport+0x38/0x524
[  108.251835] [c000001fd0bdbca0] [c000000000a68090] usb_stor_transparent_scsi_command+0x18/0x2c
[  108.254733] [c000001fd0bdbcc0] [c000000000a6aa94] usb_stor_control_thread+0x29c/0x2bc
[  108.257655] [c000001fd0bdbd80] [c00000000007c1a8] kthread+0x100/0x108
[  108.260595] [c000001fd0bdbe30] [c00000000000986c] ret_from_kernel_thread+0x5c/0x70
[  112.002821] NMI watchdog: BUG: soft lockup - CPU#6 stuck for 23s! [usb-storage:3410]
[  112.002824] NMI watchdog: BUG: soft lockup - CPU#5 stuck for 23s! [usb-storage:3385]
[  112.002828] NMI watchdog: BUG: soft lockup - CPU#0 stuck for 23s! [usb-storage:3380]
[  112.002833] Modules linked in:
[  112.002834] Modules linked in:
[  112.002834] 
[  112.002835] 
[  112.002839] CPU: 5 PID: 3385 Comm: usb-storage Not tainted 4.8.0-00002-g6829d3a34422 #47
[  112.002842] CPU: 0 PID: 3380 Comm: usb-storage Not tainted 4.8.0-00002-g6829d3a34422 #47
[  112.002845] task: c000001fd0b13f00 task.stack: c000001fd0bd8000
[  112.002847] task: c000001fd0b0f680 task.stack: c000001fd0bc4000
[  112.002850] NIP: c000000000bedba0 LR: c000000000bedb98 CTR: 0000000000000000
[  112.002852] NIP: c000000000bedb9c LR: c000000000bedb98 CTR: 0000000000000000
[  112.002855] REGS: c000001fd0bdb550 TRAP: 0901   Not tainted  (4.8.0-00002-g6829d3a34422)
[  112.002858] REGS: c000001fd0bc7550 TRAP: 0901   Not tainted  (4.8.0-00002-g6829d3a34422)
[  112.002860] MSR: 900000000280b033 
[  112.002861] MSR: 900000000280b033 
[  112.002862] <
[  112.002863] <
[... 224 lines cut ...]
[  171.289129] [c000001fd0bdba20] [c000000000a33438] usb_submit_urb+0x444/0x494
[  171.291213] [c000001fd0bdbac0] [c000000000a68424] usb_stor_msg_common+0x98/0x158
[  171.293462] [c000001fd0bdbb20] [c000000000a68878] usb_stor_bulk_transfer_buf+0x58/0xa4
[  171.295723] [c000001fd0bdbb60] [c000000000a68f90] usb_stor_Bulk_transport+0xe0/0x35c
[  171.297755] [c000001fd0bdbbc0] [c000000000a693f0] usb_stor_invoke_transport+0x38/0x524
[  171.299796] [c000001fd0bdbca0] [c000000000a68090] usb_stor_transparent_scsi_command+0x18/0x2c
[  171.301833] [c000001fd0bdbcc0] [c000000000a6aa94] usb_stor_control_thread+0x29c/0x2bc
[  171.303895] [c000001fd0bdbd80] [c00000000007c1a8] kthread+0x100/0x108
[  171.305937] [c000001fd0bdbe30] [c00000000000986c] ret_from_kernel_thread+0x5c/0x70
[  171.307941] 	5-...: (8401 ticks this GP) idle=b43/140000000000001/0 softirq=105/105 fqs=4199 
[  171.309965] 	(detected by 0, t=8411 jiffies, g=3, c=2, q=2)
[  171.312004] Task dump for CPU 5:
[  171.314027] usb-storage     R  running task        0  3385      2 0x00000804
[  171.316094] Call Trace:
[  171.318044] [c000001fd0bdb850] [c000001ffe10cd00] 0xc000001ffe10cd00 (unreliable)
[  171.320119] [c000001fd0bdba20] [c000000000a33438] usb_submit_urb+0x444/0x494
[  171.322194] [c000001fd0bdbac0] [c000000000a68424] usb_stor_msg_common+0x98/0x158
[  171.324443] [c000001fd0bdbb20] [c000000000a68878] usb_stor_bulk_transfer_buf+0x58/0xa4
[  171.326552] [c000001fd0bdbb60] [c000000000a68f90] usb_stor_Bulk_transport+0xe0/0x35c
[  171.328645] [c000001fd0bdbbc0] [c000000000a693f0] usb_stor_invoke_transport+0x38/0x524
[  171.330731] [c000001fd0bdbca0] [c000000000a68090] usb_stor_transparent_scsi_command+0x18/0x2c
[  171.332824] [c000001fd0bdbcc0] [c000000000a6aa94] usb_stor_control_thread+0x29c/0x2bc
[  171.332828] [c000001fd0bdbd80] [c00000000007c1a8] kthread+0x100/0x108
[  171.332833] [c000001fd0bdbe30] [c00000000000986c] ret_from_kernel_thread+0x5c/0x70
```

shenki, 2016-10-12:
4.4.22-openpower1

```
[  193.528305] xhci_hcd 0003:09:00.0: remove, state 1^M
[  193.528501] usb usb2: USB disconnect, device number 1^M
[  193.528774] xhci_hcd 0003:09:00.0: USB bus 2 deregistered^M
[  193.528994] xhci_hcd 0003:09:00.0: remove, state 1^M
[  193.529172] usb usb1: USB disconnect, device number 1^M
[  193.529364] usb 1-1: USB disconnect, device number 2^M
[  193.713253] usb 1-3: USB disconnect, device number 3^M
[  193.713577] usb 1-3.1: USB disconnect, device number 4^M
[  193.893400] usb 1-3.2: USB disconnect, device number 5^M
[  194.092817] usb 1-3.3: USB disconnect, device number 6^M
[  194.353350] usb 1-3.4: USB disconnect, device number 7^M
[  194.619540] xhci_hcd 0003:09:00.0: Host not halted after 16000 microseconds.^M
[  194.620134] xhci_hcd 0003:09:00.0: USB bus 1 deregistered^M
[  194.620980] kexec_core: Starting new kernel^M
```

v4.8 with mdelay patch and xhci_pci_remove for xhci device shutdown
```
[   73.598143] xhci_hcd 0021:09:00.0: remove, state 1^M
[   73.598343] usb usb2: USB disconnect, device number 1^M
[   73.598625] DELAYING BEFORE HALTING^M
[   73.598805] DELAYING BEFORE HALTING^M
[   73.598987] DELAYING BEFORE HALTING^M
[   73.599170] DELAYING BEFORE HALTING^M
[   73.599356] DELAYING BEFORE HALTING^M
[   73.599542] DELAYING BEFORE HALTING^M
[   73.599730] DELAYING BEFORE HALTING^M
[   73.654715] xhci_hcd 0021:09:00.0: Host not halted after 16000 microseconds.^M
[   73.655094] xhci_hcd 0021:09:00.0: Host controller not halted, aborting reset.^M
[   73.655346] xhci_hcd 0021:09:00.0: USB bus 2 deregistered^M
[   73.655823] xhci_hcd 0021:09:00.0: remove, state 1^M
[   73.656079] usb usb1: USB disconnect, device number 1^M
[   73.656351] usb 1-1: USB disconnect, device number 2^M
[   73.873103] usb 1-3: USB disconnect, device number 3^M
[   73.873508] usb 1-3.1: USB disconnect, device number 4^M
[   74.053194] usb 1-3.2: USB disconnect, device number 5^M
[   74.323185] usb 1-3.3: USB disconnect, device number 6^M
[   74.713183] usb 1-3.4: USB disconnect, device number 7^M
[   74.993339] xhci_hcd 0021:09:00.0: USB bus 1 deregistered^M
[   74.994293] kexec_core: Starting new kernel^M
```

stewart-ibm, 2016-11-03:
I assume this is open so that we track our USB issues and have them resolved by the time we switch to 4.9 kernel?

shenki, 2016-11-03:
I think we can close it. We're tracking the eventual upstream fix in #307.

closed 2016-11-03 by shenki