open-power/op-build #701: witherpsoon configuration does not build

shenki, 2016-10-21:
The OCC part of the build attempts to access IBM internal paths. This is blocking the build.

```
/occ-64fcfd7e3f3899e761d21eda204b53114d494aaa/src/occ_405 -DSTRAIGHT_TO_OBS_HACK=1 -DIMAGE_NAME=occBootLoader -DSSX_TIMER_SUPPORT=1 -DSSX_THREAD_SUPPORT=1 -DPPC405_MMU_SUPPORT=1 -DSSX_TRACE_SUPPORT=1 -DSSX_TRACE_HASH_PREFIX=19882 -DSSX_USE_INIT_SECTION=0 -DUSE_SSX_APP_CFG_H=1 -D__SSX__=1 -DCONFIGURE_PTS_SLW=0  -o /var/lib/jenkins/workspace/opbuild-build/master-next/Platform/witherspoon/output/build/occ-64fcfd7e3f3899e761d21eda204b53114d494aaa/obj/occBootLoader/bootMain.o bootMain.c
10:55:58 sh: 1: /afs/awd/projects/cte/tools/ppcgcc/prod/bin/powerpc-linux-gcc: not found
10:55:58 error from cpp
10:55:58 cpp returned 127
10:55:58 sh: 1: /afs/awd/projects/cte/tools/ppcgcc/prod/bin/powerpc-linux-gcc: not found
10:55:58 img_defs.mk:240: recipe for target '/var/lib/jenkins/workspace/opbuild-build/master-next/Platform/witherspoon/output/build/occ-64fcfd7e3f3899e761d21eda204b53114d494aaa/obj/occBootLoader/bootMain.o' failed
```

@ploetzma said that @wilbryan is working on this.

ploetzma, 2016-10-21:
@wilbryan said they need something like this:
https://github.com/open-power/occ/blob/master_p8/src/ssx/pgp/ssx.mk#L75

in this place:
https://github.com/open-power/occ/blob/64fcfd7e3f3899e761d21eda204b53114d494aaa/src/occBootLoader/img_defs.mk#L117

Last I heard he was attempting to recreate and test the fix

wilbryan, 2016-10-24:
What I believe should fix this issue is in the occ repository. From what I understand, master-next will be updated with the latest commit ID in occ.mk, right?

williamspatrick, 2016-10-25:
Merged occ update that is suppose to fix this problem.

shenki, 2016-10-25:
It is still failing at the OCC build step. In this case it appears the PPE compiler is crashing

```
make -I output/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/src/occ_gpe1 -C output/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/src/lib/occlib
make[3]: Entering directory 'output/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/src/lib/occlib'
mkdir -p output/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/obj/occ_gpe1/occlib
output/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/obj/ppetools/ppetracepp output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/bin/powerpc-eabi-gcc -c -g -gpubnames -gdwarf-3 -Wall -Werror -fsigned-char -ffunction-sections -fdata-sections -msoft-float -mcpu=ppe42 -meabi -ffreestanding -fno-common -fno-inline-functions-called-once -pipe -Os -Ioutput/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/src/occ_gpe1 -Ioutput/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/src/occ_gpe1/.. -Ioutput/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/src/ppe/pk/kernel -Ioutput/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/src/ppe/pk/ppe42 -Ioutput/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/src/ppe/pk/trace -Ioutput/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/src/ppe/pk/gpe -Ioutput/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/src/ppe/pk/../../include -Ioutput/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/src/ppe/pk/../../include/registers -Ioutput/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/src/lib/occlib -Ioutput/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/src/lib/common -Ioutput/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/src/occ_405/incl -DIMAGE_NAME=occ_gpe1 -DPK_TIMER_SUPPORT=1 -DPK_THREAD_SUPPORT=1 -DPK_TRACE_SUPPORT=1 -DPK_TRACE_HASH_PREFIX=31727 -DUSE_PK_APP_CFG_H=1 -D__PK__=1 -S -o output/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/obj/occ_gpe1/occlib/ipc_core.s ipc_core.c
*** buffer overflow detected ***: output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1 terminated
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x77725)[0x2ab6e62c7725]
/lib/x86_64-linux-gnu/libc.so.6(__fortify_fail+0x5c)[0x2ab6e636889c]
/lib/x86_64-linux-gnu/libc.so.6(+0x1168a0)[0x2ab6e63668a0]
/lib/x86_64-linux-gnu/libc.so.6(+0x115e09)[0x2ab6e6365e09]
/lib/x86_64-linux-gnu/libc.so.6(_IO_default_xsputn+0x80)[0x2ab6e62cb5e0]
/lib/x86_64-linux-gnu/libc.so.6(_IO_vfprintf+0x7bd)[0x2ab6e629d8ed]
/lib/x86_64-linux-gnu/libc.so.6(__vsprintf_chk+0x84)[0x2ab6e6365e94]
/lib/x86_64-linux-gnu/libc.so.6(__sprintf_chk+0x7d)[0x2ab6e6365ded]
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1[0xad96c6]
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1(_Z15final_scan_insnP7rtx_defP8_IO_FILEiiPi+0x925)[0x663eb5]
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1(_Z5finalP7rtx_defP8_IO_FILEi+0x226)[0x664906]
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1[0x664b76]
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1(_Z16execute_one_passP8opt_pass+0x41f)[0x7a151f]
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1(_Z17execute_pass_listP8opt_pass+0x16)[0x7a1876]
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1(_Z17execute_pass_listP8opt_pass+0x28)[0x7a1888]
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1(_Z17execute_pass_listP8opt_pass+0x28)[0x7a1888]
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1[0x5bdab6]
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1[0x5be84e]
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1(_Z25finalize_compilation_unitv+0x85)[0x5bfb55]
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1(_Z27c_write_global_declarationsv+0x1cc)[0x4e1b0c]
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1[0x83a595]
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1(_Z11toplev_mainiPPc+0xa98)[0x83c268]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x2ab6e6270830]
output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1(_start+0x29)[0x4cce49]
======= Memory map: ========
00400000-0121e000 r-xp 00000000 fc:00 636562                             output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1
0141d000-0141f000 r--p 00e1d000 fc:00 636562                             output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1
0141f000-01425000 rw-p 00e1f000 fc:00 636562                             output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1
[... cut ...]
make[1]: *** [output/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/obj/occ_gpe1/occ_gpe1.out] Error 2
make[1]: Leaving directory 'output/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/src'
package/pkg-generic.mk:201: recipe for target 'output/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/.stamp_built' failed
make: *** [output/build/occ-a4e04571d60b58c0c778c870bc215bf467085fb6/.stamp_built] Error 2
make: Leaving directory 'buildroot'
Build step 'Execute shell' marked build as failure
```

wilbryan, 2016-10-25:
What's strange is that we're able to build the witherspoon PNOR locally on our gfw machines. @dgilbert999 will take a look.

stewart-ibm, 2016-10-25:
@wilbryan It probably works for you because the gfw machines are running positively ancient compilers that don't do any hardening of binaries.

see:
```*** buffer overflow detected ***: output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/libexec/gcc/powerpc-eabi/4.9.2/cc1 terminated```

You'll hit the same thing, just instead of crashing out it'll overwrite some memory, hopefully not affecting your binary results

dgilbert999, 2016-10-28:
I believe I've found the issue with the buffer overflow. I will do a pull request to ppe42-gcc

williamspatrick, 2016-10-29:
I merged your pull request. I don't think this package is one we automatically update yet so you will need to create a pull request to op-build also.

ploetzma, 2016-11-08:
Has this been fixed?

wghoffa, 2016-11-08:
I spoke with @dgilbert999 - It should be fixed with this: https://github.com/open-power/op-build/pull/714 which has been merged.

stewart-ibm, 2016-11-10:
This may also now be the case with the gcc6 switch, as that does only add p8 patches for Hostboot.

Of course, it passes all of the regression tests on pull requests as witherspoon isn't built there, so perhaps it's not my problem until it fails there :)

stewart-ibm, 2016-11-17:
So, this particular issue seems fixed now, with witherspoon building in a number of places.

I'll open a new issue for Witherspoon not using GCC6 though.

closed 2016-11-17 by stewart-ibm