open-power/op-build #703: After an HMI Malfunction alert NRE system failed to boot from primary side and boots from golden side.

pridhiviraj, 2016-10-24:
Install a pnor image on habanero system which is build from op-build master next branch, And execute a HMI Malfunction alert from host OS. System failed to boot from primary side saying enough hardware is not available(Both processors and mem buf's)  in the system and boots from the golden side pnor.

01:53:35 [pexpect]#./putscom -c 00000000 1c013100 1000000000000000
01:53:35 ./putscom -c 00000000 1c013100 1000000000000000
01:53:35   3.82673|Ignoring boot flags, incorrect version 0x0
01:53:35   3.91722|ISTEP  6. 3
01:53:35   4.36792|ISTEP  6. 4
01:53:35   4.36871|ISTEP  6. 5
01:53:35   9.11880|HWAS|PRESENT> DIMM[03]=AAAA0000AAAA0000
01:53:35   9.11880|HWAS|PRESENT> Membuf[04]=CC00000000000000
01:53:35   9.11881|HWAS|PRESENT> Proc[05]=8000000000000000
01:53:35   9.24349|ISTEP  6. 6
01:53:35  10.30301|================================================
01:53:35  10.30302|Error reported by unknown (0xE500)
01:53:35  10.30302|  <none>
01:53:35  10.30302|  ModuleId   0x0b unknown
01:53:35  10.30302|  ReasonCode 0xe504 unknown
01:53:35  10.30303|  UserData1  unknown : 0x0005000000000103
01:53:35  10.30303|  UserData2  unknown : 0xe2da000000000000
01:53:35  10.30303|User Data Section 0, type UD
01:53:35  10.30303|  Subsection type 0x04
01:53:35  10.30304|  ComponentId errl (0x0100)
01:53:35  10.30304|User Data Section 1, type UD
01:53:35  10.30304|  Subsection type 0x06
01:53:35  10.30305|  ComponentId errl (0x0100)
01:53:35  10.30305|  CALLOUT
01:53:35  10.30305|  HW CALLOUT
01:53:35  10.30305|  Reporting CPU ID: 15
01:53:35  10.30306|  Called out entity:
01:53:35  10.30306|User Data Section 2, type UD
01:53:35  10.30306|  Subsection type 0x33
01:53:35  10.30307|  ComponentId unknown (0xe500)
01:53:35  10.30307|User Data Section 3, type UD
01:53:35  10.30307|  Subsection type 0x01
01:53:35  10.30308|  ComponentId unknown (0xe500)
01:53:35  10.30308|  STRING
01:53:35  10.30308|  
01:53:35  10.30308|User Data Section 4, type UD
01:53:35  10.30309|  Subsection type 0x15
01:53:35  10.30309|  ComponentId hb-trace (0x3100)
01:53:35  10.30309|User Data Section 5, type UD
01:53:35  10.30309|  Subsection type 0x01
01:53:35  10.30310|  ComponentId unknown (0xe500)
01:53:35  10.30310|  STRING
01:53:35  10.30310|  
01:53:35  10.30310|User Data Section 6, type UD
01:53:35  10.30311|  Subsection type 0x03
01:53:35  10.30311|  ComponentId errl (0x0100)
01:53:35  10.30311|User Data Section 7, type UD
01:53:35  10.30312|  Subsection type 0x01
01:53:35  10.30312|  ComponentId errl (0x0100)
01:53:35  10.30312|  STRING
01:53:35  10.30312|  Hostboot Build ID: 
01:53:35  10.30313|User Data Section 8, type UD
01:53:35  10.30313|  Subsection type 0x04
01:53:35  10.30313|  ComponentId errl (0x0100)
01:53:35  10.30314|================================================
01:53:35  19.45794|System Shutting Down To Perform Reconfiguration
01:53:35  19.45795|IPMI: Initiate power cycle
01:53:35  19.45883|Stopping istep dispatcher
01:53:35  29.25931|IPMI: shutdown complete
01:53:35 
01:53:35   3.81797|Ignoring boot flags, incorrect version 0x0
01:53:35   3.90368|ISTEP  6. 3
01:53:35   4.35336|ISTEP  6. 4
01:53:35   4.35443|ISTEP  6. 5
01:53:35   9.10055|HWAS|PRESENT> DIMM[03]=AAAA0000AAAA0000
01:53:35   9.10056|HWAS|PRESENT> Membuf[04]=CC00000000000000
01:53:35   9.10056|HWAS|PRESENT> Proc[05]=8000000000000000
01:53:35   9.22599|ISTEP  6. 6
[... 311 lines cut ...]
01:53:35 [8393527940,5] Free space in HEAP memory regions:
01:53:35 [8393634191,5] Region ibm,firmware-heap free: a23fd0
01:53:35 [8406728730,5] Region ibm,firmware-allocs-memory@0 free: 10cd70
01:53:35 [8430483415,5] Total free: 11734336
01:53:35 [8462502966,5] INIT: Starting kernel at 0x20010000, fdt at 0x3055f208 (size 0x15750)
01:53:35 
01:53:35 
zImage starting: loaded at 0x0000000020010000 (sp: 0x0000000020ec9ed8)
01:53:35 
Allocating 0x1575554 bytes for kernel ...
01:53:35 
gunzipping (0x0000000000000000 <- 0x000000002001d000:0x0000000020ec7459)...done 0x13f0300 bytes
01:53:35 
01:53:35 
Linux/PowerPC load: 
01:53:35 
Finalizing device tree... flat tree at 0x20ed6520
01:53:35 
 -> smp_release_cpus()
01:53:35 spinning_secondaries = 79
01:53:35  <- smp_release_cpus()
01:53:35  <- setup_system()
01:53:35 
01:53:35 (B)0[1;24r[m[?7h[?1h=[H[J[H[J Petitboot (v1.3.1-828e2bd)
[1B [0mqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq

dcrowell77, 2016-10-24:
@zane131 Please confirm callouts for injected error with Henderson/Mueller.  I suspect this is just working as designed.

stewart-ibm, 2018-08-07:
Golden side never worked correctly as far as I could see, and there's practically going to be no updates to how it works on p8 ever.

I don't think it's worth keeping this bug open.

closed 2018-08-07 by stewart-ibm