open-power/op-build #709: OCC Reset command just hangs on OpenPower habanero system and occ's are entered in Disabled state.

pridhiviraj, 2016-10-27:
Install a pnor image which is built from op-build master-next branch and try to do a OCC Reset on a OpenPower habanero system and command just hangs and occ's are entered into Disabled state on first attempt itself. 

 service opal-prd status
* opal-prd.service - OPAL PRD daemon
   Loaded: loaded (/lib/systemd/system/opal-prd.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2016-10-27 10:11:39 EDT; 1min 0s ago
     Docs: man:opal-prd(8)
 Main PID: 4325 (opal-prd)
    Tasks: 1 (limit: 11059)
   CGroup: /system.slice/opal-prd.service
           `-4325 /usr/sbin/opal-prd --pnor /dev/mtd0

ipmitool fru print 43
 Product Name          : OpenPOWER Firmware
 Product Version       : open-power-habanero-e60dd7a
 Product Extra         : 	buildroot-211bd05
 Product Extra         : 	skiboot-5.3.7
 Product Extra         : 	hostboot-5a94e29
 Product Extra         : 	linux-4.4.24-openpower1-bf3db29
 Product Extra         : 	petitboot-v1.3.1-828e2bd
 Product Extra         : 	habanero-xml-5565b8f-3d88639
 Product Extra         : 	occ-bb092e9
 Product Extra         : 	hostboot-binaries-56532f5
 Product Extra         : 	capp-uco

ipmitool sdr elist | grep -i occ
OCC Active       | 08h | ok  | 210.0 | Device Enabled
opal-prd occ reset

On another terminal just check the occ's status again.
ipmitool sdr elist | grep -i occ
OCC Active       | 08h | ok  | 210.0 | Device Disabled

OCC's are entered into disabled state.

The above all tested on a fresh reboot it self.
 opal-prd --version
opal-prd opal-prd-5.3.3

Tried same issue with upstream opal-prd from skiboot
./opal-prd --version
opal-prd opal-prd-5.4.0-rc2

~/skiboot/external/opal-prd# ./opal-prd occ reset

ipmitool sdr elist | grep -i occ
OCC Active       | 08h | ok  | 210.0 | Device Disabled

occ reset command just hangs and occ's are in disabled state.

pridhiviraj, 2016-10-27:
[syslog.txt](https://github.com/open-power/op-build/files/556143/syslog.txt)

pridhiviraj, 2016-10-27:
[opal_msglog.txt](https://github.com/open-power/op-build/files/556146/opal_msglog.txt)

dcrowell77, 2016-10-27:
Oct 27 10:15:33 pkvmhab012 opal-prd: HBRT: PNOR:>>RtPnor::readFromDevice: i_offset=0x0, i_procId=0 sec=11 size=0x20000 ecc=1
Oct 27 10:15:33 pkvmhab012 opal-prd: HBRT: PNOR:RtPnor::readFromDevice: removing ECC...
Oct 27 10:15:33 pkvmhab012 opal-prd: HBRT: PNOR:RtPnor::readFromDevice> Uncorrectable ECC error : chip=0,offset=0x0

We get an error trying to access the HBERRL section of PNOR.  I don't see any error messages from the OPAL code so I don't have any guesses.  I would've thought we'd see any issues during initial boot, but maybe not.  I suggest you reflash the pnor.

jazurin, 2016-10-27:
@pridhiviraj This looks similar to: https://github.com/open-power/op-build/pull/696
Does this reproduce every time?

stewart-ibm, 2016-11-03:
I'm 99% sure this is the same as what the issue was in pull req #696 and https://github.com/open-power/skiboot/issues/44 which is resolved in skiboot-5.4.0-rc3.

Please re-open if reproducible on rc3

closed 2016-11-03 by stewart-ibm

pridhiviraj, 2016-11-03:
Tested with skiboot 5.4.0-rc3 , occ reset works fine.