open-power/op-build #718: opal-prd not clearing IPOLL MASK REGISTER on an OpenPower hardware.

pridhiviraj, 2016-11-02:
cd /tmp/skiboot/external/xscom-utils/;
./getscom -c 0x0 0x01020013
cd /tmp/skiboot/external/xscom-utils/;
[pexpect]#./getscom -c 0x0 0x01020013
1000000000000000
[pexpect]#service opal-prd start
service opal-prd start
[pexpect]#./getscom -c 0x0 0x01020013
./getscom -c 0x0 0x01020013
1000000000000000
[pexpect]#Traceback (most recent call last):
  File "<string>", line 16, in <module>
  File "/home/pridhiviraj/new_optests/op-test-framework/ci/source/op_opal_fvt.py", line 347, in test_prd_driver
    return opTestPrdDriver.testPrdDriver()
  File "/home/pridhiviraj/new_optests/op-test-framework/testcases/OpTestPrdDriver.py", line 113, in testPrdDriver
    raise OpTestError(l_msg)
common.OpTestError.OpTestError: IPOLL MASK REGISTER is not getting cleared by opal-prd

With latest CI run this test case is failing, which is not expected.

Below are the registers:
    # PRD driver specific registers
    IPOLL_MASK_REGISTER = "0x01020013"
    IPOLL_MASK_REGISTER_CONTENT = "0000000000000000"

When opal-prd runs on open power system, expected IPOLL_MASK_REGISTER value is IPOLL_MASK_REGISTER_CONTENT ("0000000000000000"), but instead of it still we are getting "1000000000000000" which is not expected.

Skiboot level
cat /sys/firmware/opal/msglog | grep -i skiboot
[   45.107135142,5] SkiBoot skiboot-5.4.0-rc2-10-ge6099da starting...

PNOR Level:
ipmitool fru print 43
 Product Name          : OpenPOWER Firmware
 Product Version       : open-power-habanero-f1b7315
 Product Extra         : 	buildroot-211bd05
 Product Extra         : 	skiboot-5.3.7
 Product Extra         : 	hostboot-5669e5b
 Product Extra         : 	linux-4.4.27-openpower1-d1d686e
 Product Extra         : 	petitboot-v1.3.1-5a8e394
 Product Extra         : 	habanero-xml-5565b8f-a5c92c8
 Product Extra         : 	occ-bb092e9
 Product Extra         : 	hostboot-binaries-56532f5
 Product Extra         : 	capp-uco

service opal-prd status
* opal-prd.service - OPAL PRD daemon
   Loaded: loaded (/lib/systemd/system/opal-prd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2016-10-30 09:43:56 EDT; 2 days ago
     Docs: man:opal-prd(8)
 Main PID: 4559 (opal-prd)
    Tasks: 1 (limit: 12288)
   CGroup: /system.slice/opal-prd.service
           `-4559 /usr/sbin/opal-prd --pnor /dev/mtd0

Oct 30 20:41:53 pkvmhab012 opal-prd[4559]: HBRT: PRDF:>>PegasusConfigurator::build()
Oct 30 20:41:53 pkvmhab012 opal-prd[4559]: HBRT: PRDF:<<PegasusConfigurator::build()
Oct 30 20:41:53 pkvmhab012 opal-prd[4559]: HBRT: PRDF:<<PRDF::noLock_initialize()
Oct 30 20:41:53 pkvmhab012 opal-prd[4559]: HBRT: ERRL:>>ErrlManager::ErrlManager constructor.
Oct 30 20:41:53 pkvmhab012 opal-prd[4559]: HBRT: ERRL:iv_hiddenErrorLogsEnable = 0x0
Oct 30 20:41:53 pkvmhab012 opal-prd[4559]: HBRT: ERRL:>>setupPnorInfo
Oct 30 20:41:53 pkvmhab012 opal-prd[4559]: HBRT: PNOR:>>RtPnor::getSectionInfo
Oct 30 20:41:53 pkvmhab012 opal-prd[4559]: HBRT: PNOR:>>RtPnor::readFromDevice: i_offset=0x0, i_procId=0 sec=11 size=0x20000 ecc=1
Oct 30 20:41:53 pkvmhab012 opal-prd[4559]: HBRT: PNOR:RtPnor::readFromDevice: removing ECC...
Oct 30 20:41:53 pkvmhab012 opal-prd[4559]: HBRT: PNOR:RtPnor::readFromDevice> Uncorrectable ECC error : chip=0,offset=0x0

dcrowell77, 2016-11-02:
@stewart-ibm @zane131

pridhiviraj, 2016-11-02:
For more info on prd driver test case.
https://github.com/open-power/op-test-framework/blob/master/testcases/OpTestPrdDriver.py

stewart-ibm, 2016-11-02:
@pridhiviraj This could be related to https://github.com/open-power/skiboot/issues/44 - which was a bug with PNOR access to partitions with ECC. Can you try with skiboot 5.4.0-rc3 ?

pridhiviraj, 2016-11-03:
@stewart-ibm tried with latest skiboot 5.4.0-rc3, PRD driver test case works fine without any failure.
Thanks

closed 2016-11-04 by stewart-ibm