open-power/op-build #725: Petitboot menu loses shows no entries fails on every maas deployments for Firestone following PSU Fault 2 and PSU Fault 1

larrymi, 2016-11-04:
The issue described below has been recreated. Basically, after multiple maas deployments, the boot order is gone. 

When it first occurred on 10/18, there was a single entry for the disk but no entries  for the network devices. This time there is not a single entry.

It should be noted that for the first occurrence that the issue went away after doing a hard power cycle from remote PDUs on 10/25. In other words, we did the equivalent of unplugging the system and plugging it back in.

The maas deployments are with MAAS 2.1 but it was originally hit while using MAAS 2.0. A curtin preseed workaround has been in place since 4/13 as noted in https://bugs.launchpad.net/maas/+bug/1558747.

System last deployed successfully at ~00:05UTC per maas event log: 
----------------------------------------------------------------------------------------------------------
Node changed status - From 'Deploying' to 'Deployed'	Fri, 04 Nov. 2016 00:05:20
----------------------------------------------------------------------------------------------------------

The subsequent install fails:
----------------------------------------------------------------------------------------------------------
Node changed status - From 'Deploying' to 'Failed deployment'	Fri, 04 Nov. 2016 02:46:28
Marking node failed - Machine operation 'Deploying' timed out after 40 minutes.	Fri, 04 Nov. 2016 02:46:28
----------------------------------------------------------------------------------------------------------

Per the event log extracted from the web console, those events below are the last one shown (extracted at 1130 UTC). Strangely, there are no other events shown after 7634 and 7633 which could be indication of underlying issue that leads to current state of the system. Timestamp do seem to match timestamps on the maas event logs (last successful deployment and first failed deployment).
----------------------------------------------------------------------------------------------------------
7634	11/04/2016 00:41:30	PSU Fault 1	Power Supply	Power Supply Input Lost (AC/DC) - Asserted
7633	11/04/2016 00:09:34	PSU Fault 2	Power Supply	Power Supply Input Lost (AC/DC) - Deasserted
7632	11/04/2016 00:04:03	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7631	11/04/2016 00:03:55	FW Boot Progress	BIOS POST Progress	Progress-PCI resource configuration. - Asserted
7630	11/04/2016 00:03:51	FW Boot Progress	BIOS POST Progress	Progress-User authentication. - Asserted
7629	11/04/2016 00:03:25	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7628	11/04/2016 00:03:24	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7627	11/04/2016 00:03:16	FW Boot Progress	BIOS POST Progress	Progress-Memory initialization. - Asserted
7626	11/04/2016 00:02:54	FW Boot Progress	BIOS POST Progress	Progress-User authentication. - Asserted
7625	11/03/2016 23:56:26	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7624	11/03/2016 23:56:21	FW Boot Progress	BIOS POST Progress	Progress-PCI resource configuration. - Asserted
7623	11/03/2016 23:56:15	FW Boot Progress	BIOS POST Progress	Progress-User authentication. - Asserted
7622	11/03/2016 23:55:47	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7621	11/03/2016 23:55:45	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7620	11/03/2016 23:55:38	FW Boot Progress	BIOS POST Progress	Progress-Memory initialization. - Asserted
7619	11/03/2016 23:55:16	FW Boot Progress	BIOS POST Progress	Progress-User authentication. - Asserted
7618	11/03/2016 23:15:08	PSU Fault 2	Power Supply	Power Supply Input Lost (AC/DC) - Asserted
7617	11/03/2016 17:31:59	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7616	11/03/2016 17:31:52	FW Boot Progress	BIOS POST Progress	Progress-PCI resource configuration. - Asserted
7615	11/03/2016 17:31:47	FW Boot Progress	BIOS POST Progress	Progress-User authentication. - Asserted
7614	11/03/2016 17:31:21	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7613	11/03/2016 17:31:19	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7612	11/03/2016 17:31:11	FW Boot Progress	BIOS POST Progress	Progress-Memory initialization. - Asserted
7611	11/03/2016 17:30:50	FW Boot Progress	BIOS POST Progress	Progress-User authentication. - Asserted
7610	11/03/2016 17:22:28	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7609	11/03/2016 17:22:19	FW Boot Progress	BIOS POST Progress	Progress-PCI resource configuration. - Asserted
7608	11/03/2016 17:22:15	FW Boot Progress	BIOS POST Progress	Progress-User authentication. - Asserted
7607	11/03/2016 17:21:49	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7606	11/03/2016 17:21:48	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7605	11/03/2016 17:21:40	FW Boot Progress	BIOS POST Progress	Progress-Memory initialization. - Asserted
7604	11/03/2016 17:21:19	FW Boot Progress	BIOS POST Progress	Progress-User authentication. - Asserted
7603	11/03/2016 16:52:23	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7602	11/03/2016 16:52:14	FW Boot Progress	BIOS POST Progress	Progress-PCI resource configuration. - Asserted
7601	11/03/2016 16:52:10	FW Boot Progress	BIOS POST Progress	Progress-User authentication. - Asserted
7600	11/03/2016 16:51:44	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7599	11/03/2016 16:51:43	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7598	11/03/2016 16:51:35	FW Boot Progress	BIOS POST Progress	Progress-Memory initialization. - Asserted
7597	11/03/2016 16:51:13	FW Boot Progress	BIOS POST Progress	Progress-User authentication. - Asserted
7596	11/03/2016 16:42:52	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7595	11/03/2016 16:42:44	FW Boot Progress	BIOS POST Progress	Progress-PCI resource configuration. - Asserted
7594	11/03/2016 16:42:40	FW Boot Progress	BIOS POST Progress	Progress-User authentication. - Asserted
7593	11/03/2016 16:42:14	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7592	11/03/2016 16:42:13	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7591	11/03/2016 16:42:05	FW Boot Progress	BIOS POST Progress	Progress-Memory initialization. - Asserted
7590	11/03/2016 16:41:44	FW Boot Progress	BIOS POST Progress	Progress-User authentication. - Asserted
7589	11/03/2016 15:29:39	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
7588	11/03/2016 15:29:31	FW Boot Progress	BIOS POST Progress	Progress-PCI resource configuration. - Asserted
7587	11/03/2016 15:29:27	FW Boot Progress	BIOS POST Progress	Progress-User authentication. - Asserted
7586	11/03/2016 15:29:02	FW Boot Progress	BIOS POST Progress	Progress-Secondary processor(s) initialization. - Asserted
[... 585 lines cut ...]
[    3.857923] at24 2-0051: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    3.858026] at24 3-0050: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    3.858093] at24 3-0051: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    3.858328] at24 4-0050: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    3.858436] at24 5-0050: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    3.858504] at24 5-0051: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    3.878886] at24 6-0050: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    3.879423] at24 6-0051: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    3.880421] md: linear personality registered for level -1
[    3.880727] md: raid0 personality registered for level 0
[    3.881164] md: raid1 personality registered for level 1
[    3.881621] md: raid10 personality registered for level 10
 Petitboot (dev.20160310)                     8335-GTA         0000000000000000
 ──────────────────────────────────────────────────────────────────────────────

  System information
  System configuration
  Language
  Rescan devices
  Retrieve config from URL
 *Exit to shell           

 ──────────────────────────────────────────────────────────────────────────────
 Enter=accept, e=edit, n=new, x=exit, l=language, h=help
 Welcome to Petitboot

larrymi, 2016-11-04:
Looking back in the event log, I don't see any PSU Fault 1 or PSU Fault 2 so exact relevance of these errors is not clear.  However, that they occurred between the last good deployment and the first bad one does raise a flag.

sammj, 2016-11-06:
Please post `/var/log/messages` and `/var/log/petitboot/pb-discover.log` - perhaps Petitboot is not seeing the disks.

larrymi, 2016-11-08:
@sammj please see the attached logs that you requested.
[logs.tar.gz](https://github.com/open-power/op-build/files/578200/logs.tar.gz)

sammj, 2016-11-08:
From the logs not even the kernel appears to think this machine has disks attached. Do you know exactly what should be appearing, and how it is attached (eg. as part of a RAID array)?

Also please post `/sys/firmware/opal/msglog` in case there's a more fundamental problem.

larrymi, 2016-11-09:
@sammj in the logs that I included, there should be an lshw file. That contains the disks that were previously discovered.

larrymi, 2016-11-15:
@sammj  here's the /sys/firmware/opal/msglog
[sysfirmwareopalmsglog.log.gz](https://github.com/open-power/op-build/files/592367/sysfirmwareopalmsglog.log.gz)

sammj, 2016-11-23:
```
[7279045336,7] PHB3: Timeout waiting for electrical link
[7279047461,7] PHB3: DLP train control: 0x0fd0001101000000
[7279049941,7] PHB3: Slot freset: Retrying
[7279051718,7] PHB3: Slot freset: Asserting PERST
[7330251688,7] PHB4: Timeout waiting for link up
```

If your disks are behind one of these PHBs this could be the issue - if possible can you update to a more recent Skiboot version and retest?

larrymi, 2017-01-08:
I have tried to build the last pflash and I am getting an error when I try to  backup my existing skiboot:

To build it, I did the following:
"git clone https://github.com/open-power/skiboot.git"
On the target ppc64el system, I then ran make:
"cd skiboot/external/pflash"
"make"

Then this was from trying to backup the existing skiboot:
===================================================
ubuntu@entei:~/skiboot/external/pflash$ ./pflash -r zImage.backup -P BOOTKERNEL
Couldn't initialise architecture flash structures
===================================================

Then, I had planned to run command below to upgrade to v1.12:
===================================================
./pflash -e -p zImage.epapr -P BOOTKERNEL
===================================================

With the backup failing, I don't want to risk flashing to the latest until we understand this failure.

@sammj please take a look at the error above. Any idea as to what could be the issue and/or suggestion for how to go about updating?

stewart-ibm, 2017-01-09:
@larrymi you will need to run pflash as root

YiannisGkoufas, 2018-07-24:
Hi,
I have started to see the same behavior, without maas deployment.
On reboot suddenly it doesn't see any bootable partitions (or partitions in general).
I am attaching the logs I collected using pb-sos.
Would really appreciate any help/suggestion.
[diag.zip](https://github.com/open-power/op-build/files/2224336/diag.zip)

YiannisGkoufas, 2018-07-26:
Hi Samuel,

thanks for the info. I didn't restart petitboot manually, just rebooted the node.
I updated the config and these are logs I am getting now.

[diag.zip](https://github.com/open-power/op-build/files/2232758/diag.zip)