open-power/op-build #730: Witherspoon fails to build pnor with BOOTKERNEL image too large

ploetzma, 2016-11-07:
With latest master-next, we get an  error that the bootkernel image is too large. Do we need to increase the bootkerel size?

```ERROR: main::checkSpaceConstraints: Image provided (/esw/san2/maploetz/p9github/op-build/output/images/zImage.epapr) has size (15743392) which is greater than allocated space (15728640) for section=BOOTKERNEL.  Aborting! at /esw/san2/maploetz/p9github/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//buildpnor.pl line 698.
```

stewart-ibm, 2016-11-08:
Did you do a clean build? check target/lib/modules for modules for other kernel versions. This is usually what causes this.

ploetzma, 2016-11-08:
This was a build after doing op-build distclean, so yes it should be clean. I only see 4.9.0-rc3 in my targets/lib/modules dir
maploetz@gfw155:/esw/san2/maploetz/p9github/op-build/output/target/lib/modules$ ls -la
total 12
drwxr-xr-x 3 maploetz maploetz 4096 Nov  7 12:45 .
drwxr-xr-x 5 maploetz maploetz 4096 Nov  7 13:24 ..
drwxr-xr-x 3 maploetz maploetz 4096 Nov  7 13:24 4.9.0-rc3

ploetzma, 2016-11-08:
I reverted back to 4.8 today and was able to build a witherspoon config.

stewart-ibm, 2016-11-10:
I get 15661480 for a witherspoon_defconfig with gcc6 (a whole 67160 of free space there).

So we seem to be cutting it fairly close.

stewart-ibm, 2016-11-10:
fixed with move to XZ compressing the 4.9 kernel for witherspoon.

closed 2016-11-10 by stewart-ibm