open-power/op-build #746: opal-prd[3226]: unhandled signal 11 at 0000000000000000 nip 00003fff9474fcb0 lr 00003fff9474fa48 code 30001

pridhiviraj, 2016-11-15:
System is having latest PNOR Level and Skiboot Level. Not seeing this for every reboot. opal-prd getting exited with below message.
 opal-prd[3226]: unhandled signal 11 at 0000000000000000 nip 00003fff9474fcb0 lr 00003fff9474fa48 code 30001

service opal-prd status
Redirecting to /bin/systemctl status  opal-prd.service
● opal-prd.service - OPAL PRD daemon
   Loaded: loaded (/usr/lib/systemd/system/opal-prd.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Tue 2016-11-15 11:17:04 IST; 5h 34min ago
  Process: 3226 ExecStart=/usr/sbin/opal-prd --pnor /dev/mtd0 (code=killed, signal=SEGV)
 Main PID: 3226 (code=killed, signal=SEGV)

Nov 15 11:17:04 localhost.localdomain opal-prd[3226]: HBRT: FAPI:dimmBadDqBitmapFuncs.C: <<dimmGetBadDqBitmap
Nov 15 11:17:04 localhost.localdomain opal-prd[3226]: HBRT: FAPI:dimmBadDqBitmapFuncs.C: >>dimmSetBadDqBitmap. centaur.mba     k0:n0:s0:p04:c1    :0:0:0
Nov 15 11:17:04 localhost.localdomain opal-prd[3226]: HBRT: FAPI:dimmBadDqBitmapAccessHwp.C: >>dimmBadDqBitmapAccessHwp: Getting bitmap
Nov 15 11:17:04 localhost.localdomain opal-prd[3226]: HBRT: FAPI:dimmBadDqBitmapAccessHwp.C: <<dimmBadDqBitmapAccessHwp
Nov 15 11:17:04 localhost.localdomain opal-prd[3226]: HBRT: FAPI:dimmBadDqBitmapAccessHwp.C: >>dimmBadDqBitmapAccessHwp: Setting bitmap
Nov 15 11:17:04 localhost.localdomain opal-prd[3226]: HBRT: FAPI:dimmBadDqBitmapAccessHwp.C: dimmBadDqBitmapAccessHwp: Reconfigure needed, Bad DQ set
Nov 15 11:17:04 localhost.localdomain opal-prd[3226]: SCOM: read: chip 0x80000004, addr 0x20115ce, val 0x400f800000000000
Nov 15 11:17:04 localhost.localdomain systemd[1]: opal-prd.service: main process exited, code=killed, status=11/SEGV
Nov 15 11:17:04 localhost.localdomain systemd[1]: Unit opal-prd.service entered failed state.
Nov 15 11:17:04 localhost.localdomain systemd[1]: opal-prd.service failed.

 opal-prd occ reset
CTRL: Failed to connect to prd daemon: Connection refused

Nov 15 10:35:14 localhost systemd: Starting Basic System.
Nov 15 10:35:14 localhost systemd: Starting NTP client/server...
Nov 15 10:35:14 localhost systemd: Starting Authorization Manager...
Nov 15 10:35:14 localhost systemd: Starting Dump dmesg to /var/log/dmesg...
Nov 15 10:35:14 localhost systemd: Starting Permit User Sessions...
Nov 15 10:35:14 localhost systemd: Started OPAL PRD daemon.
Nov 15 10:35:14 localhost systemd: Starting OPAL PRD daemon...
Nov 15 10:35:14 localhost systemd: Started irqbalance daemon.
Nov 15 10:35:14 localhost chronyd[3245]: chronyd version 2.1.1 starting (+CMDMON +NTP +REFCLOCK +RTC +PRIVDROP +DEBUG +ASYNCDNS +IPV6 +SECHASH)
Nov 15 10:35:14 localhost systemd: Starting irqbalance daemon...
Nov 15 10:35:14 localhost systemd: Starting Initialize hardware monitoring sensors...
Nov 15 10:35:14 localhost systemd: Starting ppc64-diag rtas_errd (platform error handling) Service...
Nov 15 10:35:14 localhost systemd: Starting opal_errd (PowerNV platform error handling) Service...
Nov 15 10:35:14 localhost systemd: Started D-Bus System Message Bus.
Nov 15 10:35:14 localhost modprobe: modprobe: WARNING: Module jc42 not found.
Nov 15 10:35:14 localhost opal_errd: opal_errd daemon is not supported on this platform
Nov 15 10:35:14 localhost polkitd[3240]: Started polkitd version 0.112
Nov 15 10:35:14 localhost opal-prd: I2C: Found Chip: 80000000 engine 0 port 0
Nov 15 10:35:14 localhost opal-prd: I2C: Found Chip: 80000000 engine 0 port 1
Nov 15 10:35:14 localhost opal-prd: I2C: Found Chip: 80000001 engine 0 port 0
Nov 15 10:35:14 localhost opal-prd: I2C: Found Chip: 80000001 engine 0 port 1
Nov 15 10:35:14 localhost opal-prd: I2C: Found Chip: 80000004 engine 0 port 0
Nov 15 10:35:14 localhost opal-prd: I2C: Found Chip: 80000004 engine 0 port 1
Nov 15 10:35:14 localhost opal-prd: I2C: Found Chip: 80000005 engine 0 port 0
Nov 15 10:35:14 localhost opal-prd: I2C: Found Chip: 80000005 engine 0 port 1
Nov 15 10:35:14 localhost opal-prd: I2C: Found Chip: 00000000 engine 0 port 0
Nov 15 10:35:14 localhost rtas_errd: rtas_errd daemon is not supported on the PowerNV platform
Nov 15 10:35:14 localhost opal-prd: I2C: Found Chip: 00000000 engine 0 port 1
Nov 15 10:35:14 localhost opal-prd: I2C: Found Chip: 00000000 engine 1 port 2
Nov 15 10:35:14 localhost opal-prd: CTRL: Listening on control socket /run/opal-prd-control
Nov 15 10:35:14 localhost kernel: i2c /dev entries driver
Nov 15 10:35:14 localhost opal-prd: HBRT: Starting Runtime Services....
Nov 15 10:35:14 localhost opal-prd: HBRT: Initializing modules.
Nov 15 10:35:14 localhost opal-prd: HBRT: #011Initing module libtrace_rt.so...
Nov 15 10:35:14 localhost opal-prd: HBRT: done.
Nov 15 10:35:14 localhost opal-prd: HBRT: #011Initing module liberrl_rt.so...
Nov 15 10:35:14 localhost opal-prd: HBRT: done.
Nov 15 10:35:14 localhost opal-prd: HBRT: #011Initing module libtargeting_rt.so...
Nov 15 10:35:14 localhost opal-prd: HBRT: TARG:Decoded Attribute Section: 0, 0x100000000, 0x3fff7fd50000, 0x16000
Nov 15 10:35:14 localhost opal-prd: HBRT: TARG:Decoded Attribute Section: 2, 0x108000000, 0x3fff7fd66000, 0x1000
Nov 15 10:35:14 localhost opal-prd: HBRT: TARG:Decoded Attribute Section: 3, 0x110000000, 0x3fff7fd67000, 0x6000
Nov 15 10:35:14 localhost opal-prd: HBRT: TARG:Decoded Attribute Section: 10, 0x118000000, 0x3fff7fd6d000, 0x1000
Nov 15 10:35:14 localhost opal-prd: HBRT: TARG:Decoded Attribute Section: 1, 0x120000000, 0x3fff7fd6e000, 0x1000
Nov 15 10:35:14 localhost opal-prd: HBRT: TARG:[TARG] >> TARGETING::TargetService::init()
Nov 15 10:35:14 localhost opal-prd: HBRT: TARG:[TARG] I> Max Nodes to initialize is [1]
[... 871 lines cut ...]
Nov 15 10:35:18 localhost opal-prd: HBRT: PRDF:I>PRD Signature 000D0005 9E8B0000
Nov 15 10:35:18 localhost opal-prd: HBRT: PRDF:I>PRD Signature 000D0005 9E8B0000
Nov 15 10:35:18 localhost opal-prd: HBRT: PRDF:I>PRD Signature 000D0005 9E8B0000
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:fapiChipEcFeature.C: fapiQueryChipEcFeature: Chip (0x3:0x20) has feature (0x2fc3789)
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:mss_maint_cmds.C: mss_get_mark_store(): rank0, chip mark = 255, symbol mark = 255
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:mss_maint_cmds.C: mss_get_steer_mux(): rank0, port0 steer = 255, port1 steer = 255, ecc steer = 255
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:mss_maint_cmds.C: mss_get_steer_mux(): rank0, port0 steer = 255, port1 steer = 255, ecc steer = 255
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:dimmBadDqBitmapFuncs.C: >>dimmGetBadDqBitmap. centaur.mba     k0:n0:s0:p04:c1    :0:0:0
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:dimmBadDqBitmapAccessHwp.C: >>dimmBadDqBitmapAccessHwp: Getting bitmap
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:dimmBadDqBitmapAccessHwp.C: <<dimmBadDqBitmapAccessHwp
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:dimmBadDqBitmapFuncs.C: <<dimmGetBadDqBitmap
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:dimmBadDqBitmapFuncs.C: >>dimmGetBadDqBitmap. centaur.mba     k0:n0:s0:p04:c1    :1:0:0
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:dimmBadDqBitmapAccessHwp.C: >>dimmBadDqBitmapAccessHwp: Getting bitmap
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:dimmBadDqBitmapAccessHwp.C: <<dimmBadDqBitmapAccessHwp
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:dimmBadDqBitmapFuncs.C: <<dimmGetBadDqBitmap
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:dimmBadDqBitmapFuncs.C: >>dimmSetBadDqBitmap. centaur.mba     k0:n0:s0:p04:c1    :0:0:0
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:dimmBadDqBitmapAccessHwp.C: >>dimmBadDqBitmapAccessHwp: Getting bitmap
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:dimmBadDqBitmapAccessHwp.C: <<dimmBadDqBitmapAccessHwp
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:dimmBadDqBitmapAccessHwp.C: >>dimmBadDqBitmapAccessHwp: Setting bitmap
Nov 15 10:35:18 localhost opal-prd: HBRT: FAPI:dimmBadDqBitmapAccessHwp.C: dimmBadDqBitmapAccessHwp: Reconfigure needed, Bad DQ set
Nov 15 10:35:18 localhost systemd: opal-prd.service: main process exited, code=killed, status=11/SEGV
Nov 15 10:35:18 localhost systemd: Unit opal-prd.service entered failed state.
Nov 15 10:35:18 localhost systemd: opal-prd.service failed.
Nov 15 10:35:18 localhost kernel: opal-prd[3243]: unhandled signal 11 at 0000000000000000 nip 00003fff7fdcfcb0 lr 00003fff7fdcfa48 code 30001
Nov 15 10:35:18 localhost kernel: tg3 0001:0c:00.0 net0: Link is up at 1000 Mbps, full duplex

dcrowell77, 2016-11-15:
> latest PNOR Level and Skiboot Level.
Can you be more specific?  Is this the latest op-build release or master-p8 or master-next?

pridhiviraj, 2016-11-15:
PNOR is latest op-build release
ipmitool fru print 43
 Product Name          : OpenPOWER Firmware
 Product Version       : open-power-habanero-f8ed50e
 Product Extra         : 	buildroot-211bd05
 Product Extra         : 	skiboot-5.4.0
 Product Extra         : 	hostboot-94aeacf
 Product Extra         : 	linux-4.4.30-openpower1-418c9ff
 Product Extra         : 	petitboot-v1.3.1-f126151
 Product Extra         : 	habanero-xml-5565b8f-580b6c4
 Product Extra         : 	occ-e6e194f
 Product Extra         : 	hostboot-binaries-56532f5
 Product Extra         : 	capp-uco

Skiboot is latest upstream code
cat /sys/firmware/opal/msglog | grep -i skiboot
[   45.137790355,5] SkiBoot skiboot-5.4.0-20-g3d88d1d starting...

dcrowell77, 2016-11-18:
I'm not seeing the trace at the beginning of the log that tells us where HBRT is being loaded by opal-prd.  Without that I can't decode the crashing address (00003fff7fdcfcb0).  I can't even tell if the crash is coming from inside HBRT or elsewhere.

What are you doing when this crashes?  It looks like maybe some memory error injections?  Based purely on the traces that exist I'm guessing it is somewhere in the i2c path.

stewart-ibm, 2018-08-07:
Closing for now as a lot has changed since reported and we should get more details on what build and test was being run.

closed 2018-08-07 by stewart-ibm