open-power/op-build #751: [   45.318262275,3] DT: dt_attach_root failed, duplicate sensor@58

pridhiviraj, 2016-11-16:
Power Cap Sensor is assigned same sensor id as ps_derating_sensor, so skiboot raises an error.

<globalSetting>
	<id>/sys-0/ps_derating_sensor</id>
	<property>
	<id>IPMI_SENSOR_ID</id>
	<value>0x58</value>
	</property>
</globalSetting>

<globalSetting>
	<id>/sys-0/power_cap_sensor</id>
	<property>
	<id>IPMI_SENSOR_ID</id>
	<value>0x58</value>
	</property>
</globalSetting>

ipmitool sdr elist | grep -i derating
PS Derating Fac  | 58h | ok  | 21.0 | 

ipmitool sdr elist | grep -i cap

So firmware is failed to add power cap sensor as both the sensor's assigned same sensor number.

This is all my anaysis. Not sure above info i am referring correct. Can some one have a look?

pridhiviraj, 2016-11-16:
It is happening on habanero system, with latest firmware. sensor id's are observed from machines.xml

bwmashak, 2016-11-16:
Chris Cain believes the power cap sensor is extraneous and could be deleted, but recommended AMI (BMC provider on habanero) be consulted first.  There are no plans for future development with Habanero and Habanero is NOT an OpenPower reference design, so I don't recommend changing habanero.xml at this time.

stewart-ibm, 2016-11-17:
We still have Habanero in op-build tree and use them heavily for development. So we should fix this for op-build in one way or the other.

We *could* implement something in skiboot to go and mangle things, but I'd argue we should still throw an error as there's no way this is correct.

So, I vote for fixing the XML.

stewart-ibm, 2016-11-17:
Also spotted this in the p8dtu XML

bwmashak, 2016-11-17:
Jayesh Patel has weekly meetings with AMI and will ask if there would be any impact from removing the following from habanero.xml.

<globalSetting>
	<id>/sys-0/power_cap_sensor</id>
	<property>
	<id>IPMI_SENSOR_ID</id>
	<value>0x58</value>
	</property>
</globalSetting>

bwmashak, 2016-11-17:
No concerns from AMI and just removed IPMI defn of power_cap_sensor from habanero.xml -> https://github.com/open-power/habanero-xml/commit/6a78496c286e284719a8201148a5088030b897de

pridhiviraj, 2016-12-20:
Issue is fixed with latest firmware and closing it.

closed 2016-12-20 by pridhiviraj