open-power/op-build #752: [open-power: habenaro] i2c driver is non functional in OpenPower Hardware

pridhiviraj, 2016-11-16:
SrideviRamesh raised an issue regarding i2c driver issue. i2cdetect and other i2c read/write commands getting hang, when she is running i2c driver testcase on habanero system.
https://github.com/open-power/op-test-framework/blob/master/testcases/OpTestI2Cdriver.py

https://github.com/open-power/op-test-framework/issues/85

@stewart-ibm can you have a look at OPAL msg log, there are lot of i2c failures.

stewart-ibm, 2016-11-16:
Hrm... we certainly have a I2C problem, and an SLW problem.

stewart-ibm, 2017-01-09:
So, with latest skiboot, this should be all good. Only really affects systems with a TPM, and we have code in place to mitigate the issues.

closed 2017-01-09 by stewart-ibm