open-power/op-build #755: Witherspoon not on GCC6

stewart-ibm, 2016-11-17:
Witherspoon needs to build with GCC6.

I've set this to 1.13, but I suspect that it won't be met. I could just enable it and force someone to pick up the pieces though...

williamspatrick, 2016-11-18:
Are there details of what isn't working?

williamspatrick, 2016-11-18:
It's not realistic to expect this to be fixed and make its way through the process in the next week.  They probably need instructions on how to enable GCC6 to test and an explanation of what is all broken.

stewart-ibm, 2016-11-20:
To enable GCC6, like we did in https://github.com/open-power/op-build/commit/8b90e5b4c83b1c1d532f3c0cf23090907cd6606f the following needs to be added to witherspoon_defconfig:
```
BR2_GCC_VERSION_6_X=y
```

... and then you work out what doesn't build.

closed 2017-08-03 by stewart-ibm