open-power/op-build #776: 4.4 kernel cannot build with modern binutils

shenki, 2016-12-14:
Modern binutils fail to build the 4.4 kernel due to two linker issues.

One was due to a bad opcode. This is fixed in 4.9-rc3 with https://git.kernel.org/torvalds/c/80f23935cadb1c654e81951f5a8b7c but has not hit the 4.4 backports yet.
```
  OBJCOPY arch/powerpc/kernel/vdso64/vdso64.so
arch/powerpc/kernel/idle_power7.S: Assembler messages:
arch/powerpc/kernel/idle_power7.S:161: Error: missing operand
arch/powerpc/kernel/idle_power7.S:165: Error: missing operand
arch/powerpc/kernel/idle_power7.S:203: Error: missing operand
arch/powerpc/kernel/idle_power7.S:246: Error: missing operand
scripts/Makefile.build:294: recipe for target 'arch/powerpc/kernel/idle_power7.o' failed
```

The other is due to ld being strict about generating ELF binaries. This is fixed in powerpc#next for 4.10: https://git.kernel.org/powerpc/c/ff45000fcb56b5b0f1a14a865d3541746d838a0a
```
  WRAP    arch/powerpc/boot/zImage.pseries
powerpc64le-linux-gnu-ld: arch/powerpc/boot/zImage.pseries: Not enough room for program headers, try linking with -N
powerpc64le-linux-gnu-ld: final link failed: Bad value
arch/powerpc/boot/Makefile:338: recipe for target 'arch/powerpc/boot/zImage.pseries' failed
```

Both patches are backported into the 4.4.35-openpower1 tree.

stewart-ibm, 2016-12-22:
@shenki can we close this issue as fixed with 4.4.38 ?

shenki, 2016-12-22:
Yep. Fixed as of 8ac6ef0512997a8258bb21b3891e1fa141ad67fe

closed 2016-12-22 by shenki