open-power/op-build #801: doesn't build on ppc64

stewart-ibm, 2017-01-10:
habanero_defconfig on fedora 25 ppc64:
````
>>> habanero-xml 6a78496c286e284719a8201148a5088030b897de Configuring
mkdir -p "/home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version"
# Add package name
echo -n "       habanero-xml-" > /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/habanero-xml.version.txt
# If site local
# Add site local and user, local commit, if local is dirty
# Else not local
# Add package version, extraversion if linux, op-build is dirty, op-build patches
if [ "https" == "local" ]; then echo -n "site_local-" >> /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/habanero-xml.version.txt; whoami | xargs echo -n >> /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/habanero-xml.version.txt; echo -n "-" >> /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/habanero-xml.version.txt; cd "https://github.com/open-power/habanero-xml/archive/6a78496c286e284719a8201148a5088030b897de"; (git describe --tags || git log -n1 --pretty=format:'%h' || echo "unknown") | sed 's/\(.*\)-g\([0-9a-f]\{7\}\).*/\2/;s/habanero-xml-//;' | xargs echo -n >> /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/habanero-xml.version.txt; cd "https://github.com/open-power/habanero-xml/archive/6a78496c286e284719a8201148a5088030b897de"; git describe --all --dirty | grep -e "-dirty" | sed 's/.*\(-dirty\)/\1/;' | xargs echo -n >> /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/habanero-xml.version.txt; else [ `echo -n 6a78496c286e284719a8201148a5088030b897de | wc -c` == "40" ] && (echo -n 6a78496c286e284719a8201148a5088030b897de | sed "s/^\([0-9a-f]\{7\}\).*/\1/;s/habanero-xml-//;" >> /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/habanero-xml.version.txt) || echo -n 6a78496c286e284719a8201148a5088030b897de | sed -e 's/habanero-xml-//' >> /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/habanero-xml.version.txt; if [  == "HABANERO_XML" ]; then if ls /home/stewart/op-build/output/build/habanero-xml-6a78496c286e284719a8201148a5088030b897de/Makefile 1>/dev/null; then head /home/stewart/op-build/output/build/habanero-xml-6a78496c286e284719a8201148a5088030b897de/Makefile | grep EXTRAVERSION | cut -d ' ' -f 3 | xargs echo -n >> /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/habanero-xml.version.txt; fi; fi; cd "/home/stewart/op-build/openpower"; git describe --all --dirty | if grep -e "-dirty"; then echo -n "-opdirty" >> /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/habanero-xml.version.txt; fi; if [ -f /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/habanero-xml.patch.txt ]; then echo -n "-" >> /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/habanero-xml.version.txt; cat /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/habanero-xml.patch.txt >> /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/habanero-xml.version.txt; fi fi
/bin/sh: line 0: [: ==: unary operator expected
heads/master-next-dirty
# Add new line to version.txt
echo "" >> /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/habanero-xml.version.txt;
>>> habanero-xml 6a78496c286e284719a8201148a5088030b897de Building
# copy the habanero xml where the common lives
bash -c 'mkdir -p /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch && cp -r /home/stewart/op-build/output/build/habanero-xml-6a78496c286e284719a8201148a5088030b897de/* /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch'
# generate the system mrw xml
perl -I /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/processMrw.pl -x /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/habanero.xml
Loading MRW XML: /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/habanero.xml
Creating XML: /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/HABANERO_hb.mrw.xml
MRW created successfully!
# merge in any system specific attributes, hostboot attributes
/home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/mergexml.sh /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/"HABANERO_hb.system.xml" /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/attribute_types.xml /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/attribute_types_hb.xml /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/attribute_types_openpower.xml /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/target_types_merged.xml /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/target_types_hb.xml /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/target_types_openpower.xml /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/"HABANERO_hb.mrw.xml" > /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/temporary_hb.hb.xml;
cat: /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/attribute_types_openpower.xml: No such file or directory
cat: /home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/target_types_openpower.xml: No such file or directory
# creating the targeting binary
/home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/xmltohb.pl --hb-xml-file=/home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/temporary_hb.hb.xml --fapi-attributes-xml-file=/home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/fapiattrs.xml --src-output-dir=none --img-output-dir=/home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/ --vmm-consts-file=/home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/vmmconst.h --noshort-enums --bios-xml-file=/home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/"HABANERO_bios.xml" --bios-schema-file=/home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/bios.xsd --bios-output-file=/home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/"habanero"_bios_metadata.xml
[FATAL!] Could not find enumeration with ID of PCI_BASE_ADDRS_64

     1: main::getEnumerationType(1298)
     2: main::packSingleSimpleTypeAttribute(4749)
     3: main::packAttribute(4822)
     4: main::generateTargetingImage(5532)
     5: (379)
package/pkg-generic.mk:216: recipe for target '/home/stewart/op-build/output/build/habanero-xml-6a78496c286e284719a8201148a5088030b897de/.stamp_built' failed
make[1]: *** [/home/stewart/op-build/output/build/habanero-xml-6a78496c286e284719a8201148a5088030b897de/.stamp_built] Error 1
Makefile:76: recipe for target '_all' failed
make: *** [_all] Error 2
make: Leaving directory '/home/stewart/op-build/buildroot'
```

stewart-ibm, 2017-01-10:
Interestingly enough, the `output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/attribute_types_openpower.xml` and `output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/target_types_openpower.xml` files both do **not** exist and are only referenced in habanero and firestone XML project .mk files.

stewart-ibm, 2017-01-10:
@bofferdn do you have any idea what's going on here? Your name is at the top of the perl script :)

dcrowell77, 2017-01-10:
The _openpower.xml files are related to some changes @e-liner made, but I don't think that is related to the fail (unless those errors completely short-circuited some copy logic).

Can you attach this file?
/home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/temporary_hb.hb.xml

stewart-ibm, 2017-01-10:
[temporary_hb.hb.zip](https://github.com/open-power/op-build/files/697450/temporary_hb.hb.zip)

dcrowell77, 2017-01-10:
```
<attribute>
		<id>PCI_BASE_ADDRS_64</id>
		<default>0x000000000000000N,0x000000000000000N,0x000000000000000N,0x000000000000000N</default>
	</attribute>
```
The 'N' at the end of those numbers doesn't look good...  Seems like a perl bug of some kind that this environment exposes (or injects).

dcrowell77, 2017-01-10:
Could there be a bug with 64-bit hex numbers?  I see a similar problem with IBSCOM_MCS_BASE_ADDR.  In all of the cases, the values in habanero.xml are 64-bit hex values.  A quick skim of the file doesn't show any other values of that size.

dcrowell77, 2017-01-10:
@stewart-ibm - What part of the equation is new here?  Just a new level of Fedora?  Can you check the perl version and compare to a previously working level of Fedora?

stewart-ibm, 2017-01-10:
Dan <notifications@github.com> writes:
> @stewart-ibm - What part of the equation is new here?  Just a new
> level of Fedora?  Can you check the perl version and compare to a
> previously working level of Fedora?

Both ppc64 (BE) and Fedora 25. I haven't tried fedora on any non-x86
platform to build op-build before.

I'll investigate.

-- 
Stewart Smith
OPAL Architect, IBM.

stewart-ibm, 2017-01-11:
@dcrowell77 I think I have a decent grip on what's going on now. For whatever reason, in setupBars(), around here:
```
                #Note: Hex convert method avoids overflow on 32bit machine
                my $b=sprintf("0x%016s",substr((
                        $i_base+$i_node_offset*$node+
                        $i_proc_offset*$proc+$i_offset*$i)->as_hex(),2));
use Data::Dumper;
print STDERR Dumper($offset, $i_offset);
```

we see the following:
```
$VAR1 = '0x1000000000
                        ';
$VAR2 = bless( {
                 'sign' => 'NaN',
                 'value' => [
                              0
                            ]
               }, 'Math::BigInt' );
```

Which makes it look like the Math::BigInt() constructor isn't converting from hex in the event where there's trailing spaces... which is verified by regexing them out. This is fun as the documentation for Math::BigInt() *explicitly* states that trailing spaces are ignored!

Investigating...

stewart-ibm, 2017-01-11:
This is a bug in the perl Math::BigInt module. I've sent a bug report off with the following details:

The Math::BigInt manual states that leading and trailing whitespace is
ignored, and in fact this has been the behaviour for many years.

However, in recent Math::BigInt (that has shipped in Linux
distributions... e.g. Fedora 25), this behaviour is broken, which breaks
our tools that rely on this.

With Math::BigInt 1.999727:
```
'use Math::BigInt; print Dumper(Math::BigInt->new("0x10\n\t\t\t"));';
$VAR1 = bless( {
		 'sign' => 'NaN',
		 'value' => [
			      0
			    ]
	       }, 'Math::BigInt' );
```
With Math::BigInt 1.9993:
```
'use Math::BigInt; print Dumper(Math::BigInt->new("0x10\n\t\t\t"));';
$VAR1 = bless( {
		 'sign' => '+',
		 'value' => [
			      16
			    ]
	       }, 'Math::BigInt' );
```
The first version to exhibit this buggy behaviour was 1.999712, with
1.999710 and 1.9997_11 both not having the bug. All versions after
1.999712, including the latest, 1.999807 exhibit the bug.

stewart-ibm, 2017-01-11:
Filed a fedora bug to track it in fedora: https://bugzilla.redhat.com/show_bug.cgi?id=1412052

stewart-ibm, 2017-01-11:
This affects all Perl versions from v5.23.7 onwards:
$ git describe --contains 6b10d254fe18e6928745e2db24fec45528237d83
v5.23.7~89

dcrowell77, 2017-01-11:
Going to push up a quick fix, can you verify it in your broken config?

Edit <OPBUILD>/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/processMrw.pl

sub setupBars
{
...stuff...
   $base  =~ s/(^\s+|\s+$)//g;    # new
   $node_offset  =~ s/(^\s+|\s+$)//g;  #new
   $proc_offset  =~ s/(^\s+|\s+$)//g;   #new
   $offset =~ s/(^\s+|\s+$)//g;   #new
        my $i_base = Math::BigInt->new($base);

That eliminates the extra whitespace from the resulting xml file so it should get around the bug.

stewart-ibm, 2017-01-11:
Upstream bug (and patch to the perl module): https://rt.cpan.org/Public/Bug/Display.html?id=119805

stewart-ibm, 2017-01-14:
Fix has started to hit the Fedora25 updates repositories (as mentioned in https://bugzilla.redhat.com/show_bug.cgi?id=1412052 ) - and I got the update on x86-64 at least (still waiting on ppc64, but shouldn't be long).

So, closing this bug as the best bet is for someone to upgrade.

Notably, I have **not** checked other distros such as Ubuntu to see if they're affected.

closed 2017-01-14 by stewart-ibm