open-power/op-build #803: ppe42-binutils doesn't build with recent GCC

stewart-ibm, 2017-01-10:
With GCC6:
```
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I. -I. -I./../include -DHAVE_bfd_elf32_powerpc_vec -DHAVE_rs6000coff_vec -DHAVE_bfd_elf32_powerpcle_vec -DHAVE_ppcboot_vec -DHAVE_bfd_elf64_powerpc_vec -DHAVE_bfd_elf64_powerpcle_vec -DHAVE_bfd_elf64_little_generic_vec -DHAVE_bfd_elf64_big_generic_vec -DHAVE_bfd_elf32_little_generic_vec -DHAVE_bfd_elf32_big_generic_vec -DBINDIR=\"/home/stewart/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/bin\" -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -Wno-error=unused-value -MT elf64-ppc.lo -MD -MP -MF .deps/elf64-ppc.Tpo -c elf64-ppc.c  -fPIC -DPIC -o .libs/elf64-ppc.o
elf64-ppc.c: In function ‘ppc64_elf_size_stubs’:
elf64-ppc.c:12257:16: error: left shift of negative value [-Werror=shift-negative-value]
          & (-1 << htab->plt_stub_align));
                ^~
elf64-ppc.c: In function ‘ppc64_elf_build_stubs’:
elf64-ppc.c:12725:12: error: left shift of negative value [-Werror=shift-negative-value]
      & (-1 << htab->plt_stub_align));
            ^~
cc1: all warnings being treated as errors
Makefile:1612: recipe for target 'elf64-ppc.lo' failed
make[6]: *** [elf64-ppc.lo] Error 1
Makefile:1654: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
Makefile:1138: recipe for target 'all' failed
make[4]: *** [all] Error 2
Makefile:2598: recipe for target 'all-bfd' failed
make[3]: *** [all-bfd] Error 2
Makefile:829: recipe for target 'all' failed
make[2]: *** [all] Error 2
package/pkg-generic.mk:216: recipe for target '/home/stewart/op-build/output/build/host-ppe42-binutils-5b161fc30519a965f16e7e73c3410a388140cba1/.stamp_built' failed
```

Blocks: https://github.com/open-power/op-build/issues/755

stewart-ibm, 2017-01-10:
Commit 29f628db8 from upstream binutils needs to be backported

stewart-ibm, 2017-01-10:
Or, more sensibly, ppe42 needs to make its way upstream.

dgilbert999, 2017-01-11:
The plan is to upstream the PPE toolchain sometime later this year.

dgilbert999, 2017-01-17:
Much more is needed than 29f628db8 to clean up the gnu binutils 2.24 to meet the requirements of gcc 6. The amount of work needed to clean up the ppe binutils based on 2.24 version to bring it up to gcc6 standards is moderate, so I rebased the ppe changes on top of 2.28(latest gnu binutils)  and that compiles under gcc6, however, it now won't compile on our pool machines or event on my openclient machine. It needs a compiler that that has c++ 11 support.   We might be stuck having two versions until our lab machines get updated. I still have testing to do.

stewart-ibm, 2017-01-31:
I'm quickly running out of computers that can build zaius. This probably needs to be bumped up somebody's priority list.

stewart-ibm, 2017-02-02:
Doug Gilbert <notifications@github.com> writes:
> Much more is needed than 29f628db8 to clean up the gnu binutils 2.24
> to meet the requirements of gcc 6. The amount of work needed to clean
> up the ppe binutils based on 2.24 version to bring it up to [...]

Building a newer GCC for lab machines could also be an option.

-- 
Stewart Smith
OPAL Architect, IBM.

stewart-ibm, 2017-02-17:
@dgilbert999 The referenced pull request and commit should be enough to tide us over until ppe42 gets upstream and we just have a single toolchain source for op-build.

It worked for me on fedora25 (host gcc 6.3.1) and I ended up with a zaius.pnor for the first time ever.

dgilbert999, 2017-02-18:
Is this in reference to the pull request I just merged into ppe42-binutils or just your change to the make file?

stewart-ibm, 2017-02-20:
Sorry, I was ambiguous. I was referring to https://github.com/open-power/op-build/issues/879 - the op-build changes.

stewart-ibm, 2017-02-20:
Unless there's any objection, I'll close this issue, as the immediate issue of building it has been solved. There still remains the not-upstream issue.

closed 2017-02-20 by stewart-ibm