open-power/op-build #848: Flashing self-compiled PNOR on Habanero

tk-wfischer, 2017-02-02:
I want to flash a self-compiled PNOR on a Habanero system. The Habanero system is directly from TYAN. It contains the AMI-based BMC-firmware ([OpenBMC is not supported for this system](https://github.com/openbmc/openbmc/issues/605#issuecomment-257350306)).

I've been able to compile the PNOR itself already:
```
[...]
# if this config has an UPDATE_FILENAME defined, create a 32M (1/2 size)
# image that only updates the non-golden side
if [ ""habanero_update.pnor"" != "" ]; then dd if=/home/tk/git/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/pnor/"habanero.pnor" of=/home/tk/git/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/pnor/"habanero_update.pnor" bs=32M count=1; /usr/bin/install /home/tk/git/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/pnor/"habanero_update.pnor" /home/tk/git/op-build/output/images; fi
1+0 records in
1+0 records out
33554432 bytes (34 MB, 32 MiB) copied, 0,0306626 s, 1,1 GB/s
make: Leaving directory '/home/tk/git/op-build/buildroot'
tk@tpt:~/git/op-build$ $ file output/images/habanero_update.pnor
/home/tk/git/op-build/output/images/habanero_update.pnor: Par archive data
```

But now I don't know how I could flash this firmware to the PNOR chip. The BMC web interface and ipmitool support only firmware images in the HPM format.

I have asked already at the open-power mailing list, and got [some valuable hints from Samuel Mendoza-Jonas to use pflash](https://lists.ozlabs.org/pipermail/openpower-firmware/2016-November/000090.html) like described in [this blog post 'Customising OpenPower firmware'](http://jk.ozlabs.org/blog/post/159/customising-openpower-firmware/). But I this blog posting is about a Palmetto system, which is running OpenBMC ([which contains pflash](https://github.com/openbmc/openbmc/blob/master/meta-openbmc-machines/meta-openpower/common/recipes-bsp/skiboot/pflash.bb)).

When I ssh into the AMI-based BMC I don't have pflash there:
```
wfischer@tpw:~$ ssh root@10.1.102.242
root@10.1.102.242's password: 

BusyBox v1.13.2 (2016-03-14 12:42:06 CST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

$ uname -a
Linux AMIA0423F31778A 2.6.28.10-ami #1 Mon Mar 14 12:45:03 CST 2016 armv5tejl unknown
$ pflash
-sh: pflash: not found
$ 
```

Do you have any hints how I could flash the self-compiled image? Or how I could generate a HPM image out of the habanero_update.pnor image? (then I could upload the HPM via the BMC webinterface).

dcrowell77, 2017-02-02:
Unfortunately the tools to create a HPM image are proprietary and require a license from AMI to run.  I'm pretty sure that you can find the source code for pflash in the skiboot tree though.  I'm not sure what it takes to compile it to run on the BMC but it is a pretty standard Linux box so it shouldn't be too hard.  The list you pointed at can probably help you out.  Good luck!

sammj, 2017-02-03:
That's right you can compile pflash yourself from here: https://github.com/open-power/skiboot/tree/master/external/pflash

You can actually run pflash from the host OS which avoids having to build with an ARM toolchain to make it work on the BMC. While easier it does mean that if something _does_ break you do need to then compile a pflash for the BMC.

dcrowell77, 2017-02-03:
@sammj - Does the OS-level pflash allow you to flash both sides of the PNOR separately?  If so, you could flash only one side (the working side) and still rely on the golden side to recover in case of a problem.

sammj, 2017-02-03:
@dcrowell77 Good point! Yes you can just flash one side to be safe.

stewart-ibm, 2017-02-03:
Sam Mendoza-Jonas <notifications@github.com> writes:
> @dcrowell77 Good point! Yes you can just flash one side to be safe.

You can also just keep around a working HPM image and flash it from the
BMC if you mess it up :)

-- 
Stewart Smith
OPAL Architect, IBM.

tk-wfischer, 2017-02-03:
Thank you all for your hints. I'll try this on Thursday (unfortunately I won't be sooner in the lab).

If it is ok for you to keep this issue open, I'll report on Thursday how things went. (Otherwise you can close this issue and I would open one for skiboot in case I would have issues with pflash from the OS.)

tk-wfischer, 2017-02-09:
I have installed Debian 8.0 on the server and compiled pflash like described by @stewart-ibm [in his blog post here](https://www.flamingspork.com/blog/2016/10/20/workaround-for-opal-prd-using-100-cpu/).

I'm getting the following error when I try to query the flash (according to --help the --info option should display information about the flash):
```
root@debian:~/skiboot/external/pflash# ./pflash --info
Couldn't scan '/sys/class/mtd' for MTD
Do you have permission? Are you root?
Couldn't initialise architecture flash structures
root@debian:~/skiboot/external/pflash# uname -a
Linux debian 3.16.0-4-powerpc64le #1 SMP Debian 3.16.39-1 (2016-12-30) ppc64le GNU/Linux
root@debian:~/skiboot/external/pflash# arch
ppc64le
root@debian:~/skiboot/external/pflash#
```

Is the Kernel maybe too old? Should I test it with Ubuntu 16.04 or 16.10 instead?

sammj, 2017-02-10:
I'm not 100% sure about Debian 8's kernel, do you have the MTD driver loaded?
Otherwise yes a more recent Ubuntu should be fine.

tk-wfischer, 2017-02-10:
With Ubuntu 16.04 pflash is working (currently the [original PNOR from TYAN version V1.00](http://www.tyan.com/support_download_bios.aspx?model=B.TN71-BP012) is on the system):
```
tk@ubuntu-16-04:~/skiboot/external/pflash$ sudo ./pflash -i
Flash info:
-----------
Name          = /dev/mtd0
Total size    = 64MB 
Erase granule = 4KB 

TOC@0x00000000 Partitions:
-----------
ID=00            part 00000000..00001000 (actual=00001000) 
ID=01            HBEL 00008000..0002c000 (actual=00024000) [ECC]
ID=02           GUARD 0002c000..00031000 (actual=00005000) [ECC]
ID=03             HBD 00031000..00091000 (actual=00060000) [ECC]
ID=04           DJVPD 00091000..000d9000 (actual=00048000) [ECC]
ID=05            MVPD 000d9000..00169000 (actual=00090000) [ECC]
ID=06            CVPD 00169000..001b1000 (actual=00048000) [ECC]
ID=07             HBI 001b1000..00751000 (actual=005a0000) [ECC]
ID=08            SBEC 00751000..007e1000 (actual=00090000) [ECC]
ID=09             SBE 007e1000..00829000 (actual=00048000) [ECC]
ID=10            WINK 00829000..00949000 (actual=00120000) [ECC]
ID=11            HBRT 00949000..00ca9000 (actual=00360000) [ECC]
ID=12         PAYLOAD 00ca9000..00da9000 (actual=00100000) 
ID=13      BOOTKERNEL 00da9000..01ca9000 (actual=00f00000) 
ID=14        ATTR_TMP 01ca9000..01cb1000 (actual=00008000) [ECC]
ID=15       ATTR_PERM 01cb1000..01cb9000 (actual=00008000) [ECC]
ID=16             OCC 01cb9000..01dd9000 (actual=00120000) [ECC]
ID=17            TEST 01dd9000..01de2000 (actual=00009000) [ECC]
ID=18           NVRAM 01de2000..01e72000 (actual=00090000) [ECC]
ID=19         FIRDATA 01e72000..01e75000 (actual=00003000) [ECC]
ID=20            CAPP 01e7e000..01ea2000 (actual=00024000) [ECC]
ID=21             HBB 01f60000..01ff0000 (actual=00090000) [ECC]
ID=22         VERSION 01ff7000..01ff8000 (actual=00001000) 
ID=23      OTHER_SIDE 03ff8000..04000000 (actual=00000000) 

TOC@0x03ff8000 Partitions:
-----------
ID=00            part 03ff8000..03ff9000 (actual=00001000) 
ID=01            HBEL 00008000..0002c000 (actual=00024000) [ECC]
ID=02           DJVPD 00091000..000d9000 (actual=00048000) [ECC]
ID=03            MVPD 000d9000..00169000 (actual=00090000) [ECC]
ID=04            CVPD 00169000..001b1000 (actual=00048000) [ECC]
ID=05        ATTR_TMP 01ca9000..01cb1000 (actual=00008000) [ECC]
ID=06       ATTR_PERM 01cb1000..01cb9000 (actual=00008000) [ECC]
ID=07           NVRAM 01de2000..01e72000 (actual=00090000) [ECC]
ID=08         FIRDATA 01e72000..01e75000 (actual=00003000) [ECC]
ID=09             HBD 02008000..02068000 (actual=00060000) [ECC]
ID=10             HBI 02188000..02728000 (actual=005a0000) [ECC]
ID=11            SBEC 02728000..027b8000 (actual=00090000) [ECC]
ID=12             SBE 027b8000..02800000 (actual=00048000) [ECC]
ID=13            WINK 02800000..02920000 (actual=00120000) [ECC]
ID=14            HBRT 02920000..02c80000 (actual=00360000) [ECC]
ID=15         PAYLOAD 02c80000..02d80000 (actual=00100000) 
ID=16      BOOTKERNEL 02d80000..03c80000 (actual=00f00000) 
ID=17             OCC 03c80000..03da0000 (actual=00120000) [ECC]
ID=18            CAPP 03da0000..03dc4000 (actual=00024000) [ECC]
ID=19             HBB 03f67000..03ff7000 (actual=00090000) [ECC]
ID=20         VERSION 03ff7000..03ff8000 (actual=00001000) 
ID=21      OTHER_SIDE 00000000..00008000 (actual=00000000) 
tk@ubuntu-16-04:~/skiboot/external/pflash$ ls -l /sys/class/mtd/
total 0
lrwxrwxrwx 1 root root 0 Feb 10 14:07 mtd0 -> ../../devices/platform/ibm,opal:flash@0/mtd/mtd0
lrwxrwxrwx 1 root root 0 Feb 10 14:07 mtd0ro -> ../../devices/platform/ibm,opal:flash@0/mtd/mtd0ro
tk@ubuntu-16-04:~/skiboot/external/pflash$ 
```

To be on the extra safe side, I wanted to read the contents from the flash and write it to a file. At 99% I get the following error:
```
tk@ubuntu-16-04:~/skiboot/external/pflash$ sudo ./pflash -r /home/tk/original-pnor-100
[... 72 lines cut ...]
ID=09         BMC_INV 01e75000..01e7e000 (actual=00009000) 
ID=10         SECBOOT 01ea2000..01ec6000 (actual=00024000) [ECC]
ID=11             HBD 02008000..02068000 (actual=00060000) [ECC]
ID=12             HBI 02188000..02728000 (actual=005a0000) [ECC]
ID=13            SBEC 02728000..027b8000 (actual=00090000) [ECC]
ID=14             SBE 027b8000..02800000 (actual=00048000) [ECC]
ID=15            WINK 02800000..02920000 (actual=00120000) [ECC]
ID=16            HBRT 02920000..02c80000 (actual=00360000) [ECC]
ID=17         PAYLOAD 02c80000..02d80000 (actual=00100000) 
ID=18      BOOTKERNEL 02d80000..03c80000 (actual=00f00000) 
ID=19             OCC 03c80000..03da0000 (actual=00120000) [ECC]
ID=20            CAPP 03da0000..03dc4000 (actual=00024000) [ECC]
ID=21     IMA_CATALOG 03dc4000..03dcd000 (actual=00009000) [ECC]
ID=22             HBB 03f67000..03ff7000 (actual=00090000) [ECC]
ID=23         VERSION 03ff7000..03ff8000 (actual=00001000) 
ID=24     BACKUP_PART 03ff8000..04000000 (actual=00000000) 
ID=25      OTHER_SIDE 00000000..00008000 (actual=00000000) 
tk@ubuntu-16-04:~/skiboot/external/pflash$ 
```

I know that flashing the original HPM via the BMC webinterface should fix my system in case I encounter any issues while updating the PNOR. But I just want to minimize any risks of bricking my system. So I'd appreciate your feedback if the following command should be fine to update the primary side of the image (updating only the 32MB habanero_update.pnor; not the 64MB habanero.pnor):

```
sudo ./pflash -e -p /home/tk/habanero_update.pnor
```

sammj, 2017-02-13:
Flashing just the primary side _should_ be OK, but the bad read is a little concerning - it could just be a corner case in ECC parsing or it could be a problem with the flash chip.
So @cyrilbur-ibm and I can have a look, can you please;
- dd out the whole PNOR:
```dd if=/dev/mtd0 of=somefile```
- read out just the HBB partition (where the read error occurs):
```pflash -P HBB -r hbb_part```
- and read out the whole PNOR again with pflash:
```pflash -r thewhole.pnor```

tk-wfischer, 2017-02-17:
Thank you a lot for your help. I assume it could be a problem with the flash chip. For your background information: this system has been used a test system and when the last external tester sent the system back, it has not been packaged accordingly (a small transport damage of the chassis has been the result, but it seems it has also lead to defects of the system).

Nevertheless I send you the desired output:
```
tk@ubuntu-16-04:~/skiboot/external/pflash$ modinfo mtd
filename:       /lib/modules/4.4.0-62-generic/kernel/drivers/mtd/mtd.ko
description:    Core MTD registration and access routines
author:         David Woodhouse <dwmw2@infradead.org>
license:        GPL
description:    Generic support for concatenating of MTD devices
author:         Robert Kaiser <rkaiser@sysgo.de>
license:        GPL
alias:          char-major-90-*
srcversion:     72F03300430EB6EA241FA69
depends:        
intree:         Y
vermagic:       4.4.0-62-generic SMP mod_unload modversions 
tk@ubuntu-16-04:~/skiboot/external/pflash$ ls -l /dev/mt*
crw------- 1 root root 90, 0 Feb 17 10:23 /dev/mtd0
crw------- 1 root root 90, 1 Feb 17 10:23 /dev/mtd0ro
tk@ubuntu-16-04:~/skiboot/external/pflash$ sudo dd if=/dev/mtd0 of=/home/tk/mtd0.bin
dd: error reading '/dev/mtd0': Input/output error
131071+0 records in
131071+0 records out
67108352 bytes (67 MB, 64 MiB) copied, 32.9265 s, 2.0 MB/s
tk@ubuntu-16-04:~/skiboot/external/pflash$ sudo ./pflash -P HBB -r /home/tk/hbb_part
Reading to "/home/tk/hbb_part" from 0x01f60000..0x01ff0000 !
[==================================================] 100%
tk@ubuntu-16-04:~/skiboot/external/pflash$ sudo ./pflash -r /home/tk/thewhole.pnor
Reading to "/home/tk/thewhole.pnor" from 0x00000000..0x04000000 !
[==================================================] 99% ETA:0s     Flash read error 15 for chunk at 0x03ff0000
tk@ubuntu-16-04:~/skiboot/external/pflash$ 
tk@ubuntu-16-04:~/skiboot/external/pflash$ 
tk@ubuntu-16-04:~/skiboot/external/pflash$ ls -l /home/tk/
total 557384
-rw-rw-r--  1 tk   tk   67108864 Feb 10 15:12 habanero.pnor
-rw-rw-r--  1 tk   tk   33554432 Feb 10 15:12 habanero_update.pnor
-rw-r--r--  1 root root   589824 Feb 17 10:35 hbb_part
-rw-r--r--  1 root root 67108352 Feb 17 10:34 mtd0.bin
-rw-r--r--  1 root root 67043328 Feb 10 16:05 original-pnor-100
-rw-r--r--  1 root root 67043328 Feb 17 10:32 original-pnor-100.2
-rw-r--r--  1 root root 67043328 Feb 10 15:41 origpnor
drwxrwxr-x 19 tk   tk       4096 Feb 10 14:09 skiboot
-rw-rw-r--  1 tk   tk   67108864 Feb 10 16:07 testflash
-rw-rw-r--  1 tk   tk   67108864 Feb 10 16:07 testflashfile
-rw-r--r--  1 root root 67043328 Feb 17 10:36 thewhole.pnor
tk@ubuntu-16-04:~/skiboot/external/pflash$ 
```

I a parallel opened session I received the following messages from syslog:
```
tk@ubuntu-16-04:~$ [  543.197395] mtd mtd0: opal_flash_async_op(op=0) failed (rc -1)
[  680.522429] mtd mtd0: opal_flash_async_op(op=0) failed (rc -1)
[  774.697789] mtd mtd0: opal_flash_async_op(op=0) failed (rc -1)
```

And here are the files (ZIP archive for download, as GitHub does not allow files >10MB): [openpower-analysefiles.zip](http://www.wefi.net/openpower/openpower-analysefiles.zip)

I think I will send the system back to TYAN for repair, but I'll wait for your feedback in case you have any suggestions.

sammj, 2017-02-20:
Thanks for the archive!
The data coming out _looks_ sane, and pulling out the firmware versions I'd be tempted to advise updating Skiboot and retrying - but the I/O errors do sound more like a iffy flash chip/controller. Especially considering this was previously a test system it could simply be that the flash chip is worn out - if it was a machine in the lab I'd just swap it out to check.

If you want to try something before shipping it off you could flash a more up to date Skiboot (or if you're feeling particularly game I'm sure @cyrilbur-ibm could tell you what model flash chip it is), but otherwise sounds like a job for Tyan.

tk-wfischer, 2017-02-28:
Thanks for your information. Finally, I have been able to flash the self-compiled firmware. But because of the hardware defects (that I think are caused by the transport damage), it does not fully boot. Instead, it boots to the Golden Image (after failing 2 times on booting the self-compiled image).

Here are some details for your information:
- Only Centaur3 is operational (the other three are not, according to the IPMI SEL). 
- The TYAN firmware v1.00 is able to boot (it has skiboot-5.0.4 included, according to running "cat /sys/firmware/devicetree/base/ibm,opal/firmware/version")
- The TYAN firmware v1.01 and also the self-compiled firmware stop to boot (with "Error reported by hwpf"), and after some time the Golden Image gets booted automatically (so this works well like designed) Nevertheless, the bootloader petitboot shows "Petitboot (v1.4.0-0847746)" from the self-compiled firmware, although skiboot comes from the Golden image (skiboot-5.0-rc3, according to running "cat /sys/firmware/devicetree/base/ibm,opal/firmware/version")

In case it is interesting to you, here are some outputs from my tests:

1) Tests with TYAN firmware 1.00:
[fw100-console-output.txt](https://github.com/open-power/op-build/files/807348/fw100-console-output.txt)
[fw100-sel-elist-including-shutdown.txt](https://github.com/open-power/op-build/files/807349/fw100-sel-elist-including-shutdown.txt)
[fw100-sel-list-including-shutdown.txt](https://github.com/open-power/op-build/files/807351/fw100-sel-list-including-shutdown.txt)
2) Tests with TYAN firmware 1.01 (and then flashing the self-compiled firmware):
[fw101-console-output-fw101-boot-goldenimage.txt](https://github.com/open-power/op-build/files/807358/fw101-console-output-fw101-boot-goldenimage.txt)
[fw101-ipmitool-fru.txt](https://github.com/open-power/op-build/files/807359/fw101-ipmitool-fru.txt)
[fw101-sel-elist.txt](https://github.com/open-power/op-build/files/807360/fw101-sel-elist.txt)
[fw101-sel-list.txt](https://github.com/open-power/op-build/files/807361/fw101-sel-list.txt)
[fw101-ssh-session-image-flashing.txt](https://github.com/open-power/op-build/files/807362/fw101-ssh-session-image-flashing.txt)
3) Tests with self-compiled firmware
[fw-selfcompiled-console-output.txt](https://github.com/open-power/op-build/files/807370/fw-selfcompiled-console-output.txt)
[fw-selfcompiled-sel-elist.txt](https://github.com/open-power/op-build/files/807372/fw-selfcompiled-sel-elist.txt)
[fw-selfcompiled-sel-list.txt](https://github.com/open-power/op-build/files/807373/fw-selfcompiled-sel-list.txt)

As mentioned, flashing the firmware worked out fine:
```
$ sudo ./pflash -e -p /home/tk/habanero_update.pnor
About to erase 0x00000000..0x02000000 !
WARNING ! This will modify your HOST flash chip content !
Enter "yes" to confirm:yes
Erasing...
[==================================================] 100% ETA:0s     
About to program "/home/tk/habanero_update.pnor" at 0x00000000..0x02000000 !
Programming & Verifying...
[==================================================] 100% ETA:0s     
tk@ubuntu-16-04:~/skiboot/external/pflash$
```

The only thing which I'm a bit surprised of, is that pfash -i showed four partions after flashing the firmware (instead of two partitions before, as mention further above in this issue):

```
tk@ubuntu-16-04:~/skiboot/external/pflash$ sudo ./pflash -i
Flash info:
-----------
Name          = /dev/mtd0
Total size    = 64MB 
Erase granule = 4KB 

TOC@0x00000000 Partitions:
-----------
ID=00            part 00000000..00001000 (actual=00001000) 
ID=01            HBEL 00008000..0002c000 (actual=00024000) [ECC]
ID=02           GUARD 0002c000..00031000 (actual=00005000) [ECC]
ID=03             HBD 00031000..00091000 (actual=00060000) [ECC]
ID=04           DJVPD 00091000..000d9000 (actual=00048000) [ECC]
ID=05            MVPD 000d9000..00169000 (actual=00090000) [ECC]
ID=06            CVPD 00169000..001b1000 (actual=00048000) [ECC]
ID=07             HBI 001b1000..00751000 (actual=005a0000) [ECC]
ID=08            SBEC 00751000..007e1000 (actual=00090000) [ECC]
ID=09             SBE 007e1000..00829000 (actual=00048000) [ECC]
ID=10            WINK 00829000..00949000 (actual=00120000) [ECC]
ID=11            HBRT 00949000..00ca9000 (actual=00360000) [ECC]
ID=12         PAYLOAD 00ca9000..00da9000 (actual=00100000) 
ID=13      BOOTKERNEL 00da9000..01ca9000 (actual=00f00000) 
ID=14        ATTR_TMP 01ca9000..01cb1000 (actual=00008000) 
ID=15       ATTR_PERM 01cb1000..01cb9000 (actual=00008000) [ECC]
ID=16             OCC 01cb9000..01dd9000 (actual=00120000) [ECC]
ID=17            TEST 01dd9000..01de2000 (actual=00009000) [ECC]
ID=18           NVRAM 01de2000..01e72000 (actual=00090000) 
ID=19         FIRDATA 01e72000..01e75000 (actual=00003000) [ECC]
[... 72 lines cut ...]
ID=05        ATTR_TMP 01ca9000..01cb1000 (actual=00008000) [ECC]
ID=06       ATTR_PERM 01cb1000..01cb9000 (actual=00008000) [ECC]
ID=07           NVRAM 01de2000..01e72000 (actual=00090000) [ECC]
ID=08         FIRDATA 01e72000..01e75000 (actual=00003000) [ECC]
ID=09             HBD 02008000..02068000 (actual=00060000) [ECC]
ID=10             HBI 02188000..02728000 (actual=005a0000) [ECC]
ID=11            SBEC 02728000..027b8000 (actual=00090000) [ECC]
ID=12             SBE 027b8000..02800000 (actual=00048000) [ECC]
ID=13            WINK 02800000..02920000 (actual=00120000) [ECC]
ID=14            HBRT 02920000..02c80000 (actual=00360000) [ECC]
ID=15         PAYLOAD 02c80000..02d80000 (actual=00100000) 
[... cut ...]
tk@ubuntu-16-04:~/skiboot/external/pflash$ 
```

Do you maybe have an idea, why I got four partitions?

In any case - I know now everything I need to know to compile the firmware by myself and to flash it afterwards with pflash. After the system is repaired (I assume a new motherboard to be installed), I will continue with my tests.

If you have some feedback to me regarding the partitions, just let me know. Otherwise I will close this issue and thank you all very much for your friendly support.

cyrilbur-ibm, 2017-03-08:
Hi @tk-wfischer,

[so sorry for the late response, I just went back through my 20's of tabs and found I didn't press enter...]

I'm completely with you that this gets confusing. You must have crossed an update boundary where backup TOCs started being generated. Nothing significant has changed in  terms of flash layout, there is still only two copies of the data the 'golden side' and the non golden side. What has been introduced is a BACKUP_PART section within both the primary TOCs (TOC@0x00000000 and TOC@0x02000000) which points you to a backup of those TOC which points to the same address on flash.

Please don't ask me why.

Cyril

tk-wfischer, 2017-03-10:
Thank you @cyrilbur-ibm for this valuable background information, and thank you all others for helping me in this issue. Now all my questions are answered, so I'm happy to hereby close this issue :-)

closed 2017-03-10 by tk-wfischer