open-power/op-build #863: pnor build hangs at "libfakeroot, when sending message: Permission denied"

pridhiviraj, 2017-02-10:
+ for i in '${DEFCONFIGS[@]}'
+ op-build habanero_defconfig
+ make --directory=/home/pridhiviraj/op-build/buildroot O=/home/pridhiviraj/op-build/output habanero_defconfig
make: Entering directory '/home/pridhiviraj/op-build/buildroot'
  GEN     /home/pridhiviraj/op-build/output/Makefile
#
# configuration written to /home/pridhiviraj/op-build/output/.config
#
make: Leaving directory '/home/pridhiviraj/op-build/buildroot'
+ echo BR2_CCACHE=y
+ echo 'BR2_CCACHE_DIR="/home/pridhiviraj/op-build/.op-build_ccache"'
+ echo 'BR2_CCACHE_INITIAL_SETUP=""'
+ op-build olddefconfig
+ make --directory=/home/pridhiviraj/op-build/buildroot O=/home/pridhiviraj/op-build/output olddefconfig
make: Entering directory '/home/pridhiviraj/op-build/buildroot'
  GEN     /home/pridhiviraj/op-build/output/Makefile
BR2_DEFCONFIG='' KCONFIG_AUTOCONFIG=/home/pridhiviraj/op-build/output/build/buildroot-config/auto.conf KCONFIG_AUTOHEADER=/home/pridhiviraj/op-build/output/build/buildroot-config/autoconf.h KCONFIG_TRISTATE=/home/pridhiviraj/op-build/output/build/buildroot-config/tristate.config BR2_CONFIG=/home/pridhiviraj/op-build/output/.config HOST_GCC_VERSION="5" BUILD_DIR=/home/pridhiviraj/op-build/output/build SKIP_LEGACY= /home/pridhiviraj/op-build/output/build/buildroot-config/conf --olddefconfig Config.in
/home/pridhiviraj/op-build/output/.config:2444:warning: override: reassigning to symbol BR2_CCACHE
#
# configuration written to /home/pridhiviraj/op-build/output/.config
#
make: Leaving directory '/home/pridhiviraj/op-build/buildroot'
+ op-build
+ make --directory=/home/pridhiviraj/op-build/buildroot O=/home/pridhiviraj/op-build/output
make: Entering directory '/home/pridhiviraj/op-build/buildroot'
which: no cmake in (/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin)
/usr/bin/make -j1 O=/home/pridhiviraj/op-build/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" silentoldconfig
make[1]: Entering directory '/home/pridhiviraj/op-build/buildroot'
  GEN     /home/pridhiviraj/op-build/output/Makefile
BR2_DEFCONFIG='' KCONFIG_AUTOCONFIG=/home/pridhiviraj/op-build/output/build/buildroot-config/auto.conf KCONFIG_AUTOHEADER=/home/pridhiviraj/op-build/output/build/buildroot-config/autoconf.h KCONFIG_TRISTATE=/home/pridhiviraj/op-build/output/build/buildroot-config/tristate.config BR2_CONFIG=/home/pridhiviraj/op-build/output/.config HOST_GCC_VERSION="5" BUILD_DIR=/home/pridhiviraj/op-build/output/build SKIP_LEGACY= /home/pridhiviraj/op-build/output/build/buildroot-config/conf --silentoldconfig Config.in
make[1]: Leaving directory '/home/pridhiviraj/op-build/buildroot'
>>>   Finalizing target directory
mkdir -p /home/pridhiviraj/op-build/output/target/etc
echo "skiroot" > /home/pridhiviraj/op-build/output/target/etc/hostname
/usr/bin/sed -i -e '$a \127.0.1.1\tskiroot' -e '/^127.0.1.1/d' /home/pridhiviraj/op-build/output/target/etc/hosts
mkdir -p /home/pridhiviraj/op-build/output/target/etc
echo "Welcome to Buildroot" > /home/pridhiviraj/op-build/output/target/etc/issue
mkdir -p /home/pridhiviraj/op-build/output/target/etc/network/
( echo "# interface file auto-generated by buildroot"; echo ; echo "auto lo"; echo "iface lo inet loopback"; ) > /home/pridhiviraj/op-build/output/target/etc/network/interfaces
/usr/bin/sed -i -e s,^root:[^:]*:,root::, /home/pridhiviraj/op-build/output/target/etc/shadow
ln -sf busybox /home/pridhiviraj/op-build/output/target/bin/sh
/usr/bin/sed -i -e '/# GENERIC_SERIAL$/s~^.*#~hvc0::respawn:/sbin/getty -L  hvc0 0 vt100 #~' /home/pridhiviraj/op-build/output/target/etc/inittab
/usr/bin/sed -i -e '/^#.*-o remount,rw \/$/s~^#\+~~' /home/pridhiviraj/op-build/output/target/etc/inittab
Generating locale en_US.UTF-8
Generating locale de_DE.UTF-8
Generating locale es_ES.UTF-8
Generating locale fr_FR.UTF-8
Generating locale it_IT.UTF-8
Generating locale ja_JP.UTF-8
Generating locale ko_KR.UTF-8
Generating locale pt_BR.UTF-8
Generating locale ru_RU.UTF-8
Generating locale zh_CN.UTF-8
Generating locale zh_TW.UTF-8
rm -f /home/pridhiviraj/op-build/output/build/locales.nopurge
for i in C de en es fr it ja ko pt_BR ru zh_CN zh_TW locale-archive; do echo $i >> /home/pridhiviraj/op-build/output/build/locales.nopurge; done
for dir in /home/pridhiviraj/op-build/output/target/usr/share/locale /home/pridhiviraj/op-build/output/target/usr/lib/locale; do for langdir in $dir/*; do if [ -e "${langdir}" ]; then grep -qx "${langdir##*/}" /home/pridhiviraj/op-build/output/build/locales.nopurge || rm -rf $langdir; fi done; done
if [ -d /home/pridhiviraj/op-build/output/target/usr/share/X11/locale ]; then for lang in C de en es fr it ja ko pt_BR ru zh_CN zh_TW; do if [ -f /home/pridhiviraj/op-build/output/target/usr/share/X11/locale/$lang/XLC_LOCALE ]; then echo "$lang/XLC_LOCALE: $lang"; fi done > /home/pridhiviraj/op-build/output/target/usr/share/X11/locale/locale.dir; fi
rm -rf /home/pridhiviraj/op-build/output/target/usr/include /home/pridhiviraj/op-build/output/target/usr/share/aclocal \
	/home/pridhiviraj/op-build/output/target/usr/lib/pkgconfig /home/pridhiviraj/op-build/output/target/usr/share/pkgconfig \
	/home/pridhiviraj/op-build/output/target/usr/lib/cmake /home/pridhiviraj/op-build/output/target/usr/share/cmake
find /home/pridhiviraj/op-build/output/target/usr/{lib,share}/ -name '*.cmake' -print0 | xargs -0 rm -f
find /home/pridhiviraj/op-build/output/target/lib/ /home/pridhiviraj/op-build/output/target/usr/lib/ /home/pridhiviraj/op-build/output/target/usr/libexec/ \
	\( -name '*.a' -o -name '*.la' \) -print0 | xargs -0 rm -f
rm -rf /home/pridhiviraj/op-build/output/target/usr/share/gdb
rm -rf /home/pridhiviraj/op-build/output/target/usr/share/bash-completion
rm -rf /home/pridhiviraj/op-build/output/target/usr/share/zsh
rm -rf /home/pridhiviraj/op-build/output/target/usr/man /home/pridhiviraj/op-build/output/target/usr/share/man
rm -rf /home/pridhiviraj/op-build/output/target/usr/info /home/pridhiviraj/op-build/output/target/usr/share/info
rm -rf /home/pridhiviraj/op-build/output/target/usr/doc /home/pridhiviraj/op-build/output/target/usr/share/doc
[... 18 lines cut ...]
	echo "VERSION_ID=2016.11"; \
	echo "PRETTY_NAME=\"Buildroot 2016.11\"" \
) >  /home/pridhiviraj/op-build/output/target/etc/os-release
>>>   Copying overlay ../openpower/overlay
>>>   Executing post-build script ../openpower/scripts/fixup-target-var
>>>   Executing post-build script ../openpower/scripts/firmware-whitelist
>>>   Generating root filesystem image rootfs.cpio
if [ ! -e /home/pridhiviraj/op-build/output/target/init ]; then /usr/bin/install -m 0755 fs/cpio/init /home/pridhiviraj/op-build/output/target/init; fi
rm -f /home/pridhiviraj/op-build/output/build/_fakeroot.fs
rm -f /home/pridhiviraj/op-build/output/target/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM
rm -f /home/pridhiviraj/op-build/output/build/_users_table.txt
echo '#!/bin/sh' > /home/pridhiviraj/op-build/output/build/_fakeroot.fs
echo "set -e" >> /home/pridhiviraj/op-build/output/build/_fakeroot.fs
echo "chown -h -R 0:0 /home/pridhiviraj/op-build/output/target" >> /home/pridhiviraj/op-build/output/build/_fakeroot.fs
printf '   	- - input -1 * - - - Input device group\n\n' >> /home/pridhiviraj/op-build/output/build/_users_table.txt
PATH="/home/pridhiviraj/op-build/output/host/bin:/home/pridhiviraj/op-build/output/host/sbin:/home/pridhiviraj/op-build/output/host/usr/bin:/home/pridhiviraj/op-build/output/host/usr/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" /home/pridhiviraj/op-build/buildroot/support/scripts/mkusers /home/pridhiviraj/op-build/output/build/_users_table.txt /home/pridhiviraj/op-build/output/target >> /home/pridhiviraj/op-build/output/build/_fakeroot.fs
cat ../openpower/device_table.txt > /home/pridhiviraj/op-build/output/build/_device_table.txt
printf '   	/bin/busybox                     f 4755 0  0 - - - - -\n /dev/console c 622 0 0 5 1 - - -\n\n' >> /home/pridhiviraj/op-build/output/build/_device_table.txt
echo "/home/pridhiviraj/op-build/output/host/usr/bin/makedevs -d /home/pridhiviraj/op-build/output/build/_device_table.txt /home/pridhiviraj/op-build/output/target" >> /home/pridhiviraj/op-build/output/build/_fakeroot.fs
printf '   	cd /home/pridhiviraj/op-build/output/target && find . | cpio --quiet -o -H newc > /home/pridhiviraj/op-build/output/images/rootfs.cpio\n' >> /home/pridhiviraj/op-build/output/build/_fakeroot.fs
chmod a+x /home/pridhiviraj/op-build/output/build/_fakeroot.fs
PATH="/home/pridhiviraj/op-build/output/host/bin:/home/pridhiviraj/op-build/output/host/sbin:/home/pridhiviraj/op-build/output/host/usr/bin:/home/pridhiviraj/op-build/output/host/usr/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" /home/pridhiviraj/op-build/output/host/usr/bin/fakeroot -- /home/pridhiviraj/op-build/output/build/_fakeroot.fs
libfakeroot, when sending message: Permission denied

i am already using sudo for root permissions.

williamspatrick, 2017-03-01:
Is this still a problem?

@shenki - Sounds like a buildroot base problem.

shenki, 2017-03-01:
What distro is this? Which verison?

Are you running it inside docker?

Can you reproduce using the latest `master-next`?

pridhiviraj, 2017-03-02:
@shenki  distro is fedora23
yes, i am running it inside docker, which are the build scripts in ci directory, But there distro version is updated to fedora25. 

Okay, i will check whether it is reproducible with latest master-next on fedora 25

shenki, 2017-03-02:
I think it's related to the permissions your container has (or in this case, does not have).

`fakeroot` uses `CAP_IPC_LOCK`, so try adding `--cap-add IPC_LOCK` to your `docker run` line.

pridhiviraj, 2017-11-14:
Even without containers also i am seeing this error.
```
make[1]: Leaving directory `/root/op-build/output/build/linux-4.13.11'
# Copy the kernel image(s) to its(their) final destination
/usr/bin/install -m 0644 -D /root/op-build/output/build/linux-4.13.11/arch/powerpc/boot/zImage.epapr /root/op-build/output/images/zImage.epapr
# If there is a .ub file copy it to the final destination
test ! -f /root/op-build/output/build/linux-4.13.11/arch/powerpc/boot/zImage.epapr.ub || cp /root/op-build/output/build/linux-4.13.11/arch/powerpc/boot/zImage.epapr.ub /root/op-build/output/images
>>>   Generating root filesystem image rootfs.tar
rm -f /root/op-build/output/build/_fakeroot.fs
rm -f /root/op-build/output/target/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM
rm -f /root/op-build/output/build/_users_table.txt
echo '#!/bin/sh' > /root/op-build/output/build/_fakeroot.fs
echo "set -e" >> /root/op-build/output/build/_fakeroot.fs
echo "chown -h -R 0:0 /root/op-build/output/target" >> /root/op-build/output/build/_fakeroot.fs
printf '   	- - input -1 * - - - Input device group\n\n' >> /root/op-build/output/build/_users_table.txt
PATH="/root/op-build/output/host/bin:/root/op-build/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games::/opt/ibm/pm/bin" /root/op-build/buildroot/support/scripts/mkusers /root/op-build/output/build/_users_table.txt /root/op-build/output/target >> /root/op-build/output/build/_fakeroot.fs
cat ../openpower/device_table.txt > /root/op-build/output/build/_device_table.txt
printf '   	/bin/busybox                     f 4755 0  0 - - - - -\n /dev/console c 622 0 0 5 1 - - -\n\n' >> /root/op-build/output/build/_device_table.txt
echo "/root/op-build/output/host/bin/makedevs -d /root/op-build/output/build/_device_table.txt /root/op-build/output/target" >> /root/op-build/output/build/_fakeroot.fs
printf '   	(cd /root/op-build/output/target; find -print0 | LC_ALL=C sort -z | tar  -cf /root/op-build/output/images/rootfs.tar --null --no-recursion -T - --numeric-owner)\n' >> /root/op-build/output/build/_fakeroot.fs
chmod a+x /root/op-build/output/build/_fakeroot.fs
PATH="/root/op-build/output/host/bin:/root/op-build/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games::/opt/ibm/pm/bin" /root/op-build/output/host/bin/fakeroot -- /root/op-build/output/build/_fakeroot.fs
rootdir=/root/op-build/output/target
table='/root/op-build/output/build/_device_table.txt'
/usr/bin/install -m 0644 support/misc/target-dir-warning.txt /root/op-build/output/target/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM
make: Leaving directory `/root/op-build/buildroot'
root@nwblr2:~/op-build# 
```

shenki, 2017-11-14:
I can't see an error there. It looks like the build completed?

pridhiviraj, 2017-11-14:
I built it for habanero, but i couldn't see any habnero pnor images in output/images directory.
```
ls output/images/
cappucode.bin   firestone.pnor.squashfs.tar  host_fw_debug.tar  rootfs.cpio     rootfs.tar   skiboot.lid.bin
firestone.pnor  firestone_update.pnor        ima_catalog.bin    rootfs.cpio.xz  skiboot.lid  zImage.epapr
```
Whatever the firestone images are for previous firestone build.