open-power/op-build #908: NVRAM Partition is getting erased after a PNOR update.

pridhiviraj, 2017-03-06:
Habanero system is initially with below nvram setting's for petitboot.
# nvram --print-config
"common" Partition
---------------------
test-cfg=test-value
petitboot,bootdevs=uuid:32c2e9d6-7cb6-49ce-8379-d263f53fc9f7 
petitboot,console=/dev/hvc0 [IPMI / Serial]

Then build & updated the PNOR Flash with latest upstream op-build master-next branch.
/tmp/pflash -e -f -p /tmp/habanero.pnor
About to erase 0x00000000..0x04000000 !
Erasing...
[==================================================] 100% ETA:0s     
About to program "/tmp/habanero.pnor" at 0x00000000..0x04000000 !
Programming & Verifying...
[==================================================] 100% ETA:0s     
# echo $?
echo $?
0

After power on of the system NVRAM partion gets erased.
# nvram --print-config
"common" Partition
---------------------

Expectation is NVRAM partion should be preserved across code updates. as per PNOR layout for P8.
 
   <section>
        <description>Nvram (576K)</description>
        <eyeCatch>NVRAM</eyeCatch>
        <physicalOffset>0x1DE2000</physicalOffset>
        <physicalRegionSize>0x90000</physicalRegionSize>
        <side>sideless</side>
        <preserved/>
        <reprovision/>
</section>

williamspatrick, 2017-03-06:
?? You didn't do a code update. You used pflash to raw write the flash module. How do you expect the nvram to be preserved when you raw wrote over it?

closed 2017-03-06 by williamspatrick

stewart-ibm, 2017-03-06:
Yeah, this is by-design. Pflash does a raw overwrite of the whole flash chip and won't maintain anything (it doesn't read the existing flash *at all*).