open-power/op-build #956: linux kernel patch can't be applied in P9 op-build

jimsmc, 2017-03-22:
In these P9 xx_defconfig files, one line is missing,
BR2_LINUX_KERNEL_PATCH="$(BR2_EXTERNAL_OP_BUILD_PATH)/linux"
Buildroot can't find the linux kernel patch folder and skips the patch.

jimsmc, 2017-03-22:
```
The build log of P9,
_[7m>>> linux 4.10-rc7 Patching_[27m
if [ -f /home/smc/p9dsu/p9dsu20170321/output/build/linux-4.10-rc7/tools/perf/Documentation/Makefile ]; then printf "%%:\n\t@:\n" >/home/smc/p9dsu/p9dsu20170321/output/build/linux-4.10-rc7/tools/perf/Documentation/GNUmakefile; fi
for p in  ; do if test -d $p ; then PATH=/home/smc/p9dsu/p9dsu20170321/output/host/usr/bin:$PATH support/scripts/apply-patches.sh  /home/smc/p9dsu/p9dsu20170321/output/build/linux-4.10-rc7 $p \*.patch || exit 1 ; else PATH=/home/smc/p9dsu/p9dsu20170321/output/host/usr/bin:$PATH support/scripts/apply-patches.sh  /home/smc/p9dsu/p9dsu20170321/output/build/linux-4.10-rc7 `dirname $p` `basename $p` || exit 1; fi done
mkdir -p "/home/smc/p9dsu/p9dsu20170321/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version";

The build log of P8,
[7m>>> linux 4.4.24 Patching[27m
for p in /home/smc/ddr4/GA2/p8dtu20170217/openpower/linux ; do if test -d $p ; then support/scripts/apply-patches.sh  /home/smc/ddr4/GA2/p8dtu20170217/output/build/linux-4.4.24 $p \*.patch || exit 1 ; else support/scripts/apply-patches.sh  /home/smc/ddr4/GA2/p8dtu20170217/output/build/linux-4.4.24 `dirname $p` `basename $p` || exit 1; fi done

Applying 0001-xhci-Use-xhci_pci_remove-for-xhci-device-shutdown.patch using patch: 
patching file drivers/usb/host/xhci-pci.c

Applying 0002-drivers-drm-ast-Switch-SCU-to-VGA-output-on-POST.patch using patch: 
patching file drivers/gpu/drm/ast/ast_post.c
...
```

stewart-ibm, 2017-03-22:
Yep, this is a bug.

We initially disabled applying the patches on P9 platforms due to running a more recent kernel on P9 than P8 configs (and it being early bringup).

The fix is to add this to the defconfigs:

```BR2_LINUX_KERNEL_PATCH="$(BR2_EXTERNAL_OP_BUILD_PATH)/linux"```

closed 2017-03-23 by stewart-ibm