open-power/op-build #960: pnor build failed for firestone platform using op-build master-next branch.

pridhiviraj, 2017-03-23:
01:42:30 PATH="/tmp/pnor_20170322_1636/output/host/bin:/tmp/pnor_20170322_1636/output/host/sbin:/tmp/pnor_20170322_1636/output/host/usr/bin:/tmp/pnor_20170322_1636/output/host/usr/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" BR_BINARIES_DIR=/tmp/pnor_20170322_1636/output/images /usr/bin/make -j13 HOSTCC="/tmp/pnor_20170322_1636/output/host/usr/bin/ccache /usr/bin/gcc" HOSTCFLAGS="" ARCH=powerpc INSTALL_MOD_PATH=/tmp/pnor_20170322_1636/output/target CROSS_COMPILE="/tmp/pnor_20170322_1636/output/host/usr/bin/powerpc64le-buildroot-linux-gnu-" DEPMOD=/tmp/pnor_20170322_1636/output/host/sbin/depmod INSTALL_MOD_STRIP=1 -C /tmp/pnor_20170322_1636/output/build/linux-4.10.3 zImage.epapr

01:42:30 make[1]: Entering directory `/tmp/pnor_20170322_1636/output/build/linux-4.10.3'

01:42:30 scripts/kconfig/conf  --silentoldconfig Kconfig

01:42:30 make[1]: Leaving directory `/tmp/pnor_20170322_1636/output/build/linux-4.10.3'

01:42:30 make[1]: Entering directory `/tmp/pnor_20170322_1636/output/build/linux-4.10.3'

01:42:30   CHK     include/config/kernel.release

01:42:30   WRAP    arch/powerpc/include/generated/asm/clkdev.h

01:42:30   WRAP    arch/powerpc/include/generated/asm/div64.h

01:42:30   WRAP    arch/powerpc/include/generated/asm/export.h

01:42:30   WRAP    arch/powerpc/include/generated/asm/irq_regs.h

01:42:30   WRAP    arch/powerpc/include/generated/asm/irq_work.h

01:42:30   WRAP    arch/powerpc/include/generated/asm/local64.h

01:42:30   WRAP    arch/powerpc/include/generated/asm/mcs_spinlock.h

01:42:30   WRAP    arch/powerpc/include/generated/asm/preempt.h

01:42:30   WRAP    arch/powerpc/include/generated/asm/rwsem.h

01:42:30   WRAP    arch/powerpc/include/generated/asm/vtime.h

01:42:30   UPD     include/config/kernel.release

01:42:30   CHK     include/generated/uapi/linux/version.h

01:42:30   UPD     include/generated/uapi/linux/version.h

01:42:30   HOSTCC  scripts/basic/bin2c

01:42:30   CHK     include/generated/utsrelease.h

01:42:30   UPD     include/generated/utsrelease.h

01:42:30   HOSTCC  scripts/kallsyms

01:42:30   HOSTCC  scripts/pnmtologo

01:42:30   HOSTCC  scripts/conmakehash

01:42:30   HOSTCC  scripts/sortextable

01:42:30   HOSTCC  scripts/asn1_compiler

01:42:30   HOSTCC  scripts/extract-cert

01:42:30   HOSTCC  scripts/dtc/dtc.o

01:42:30   HOSTCC  scripts/sign-file

01:42:30   CC      scripts/mod/empty.o

01:42:30   HOSTCC  scripts/mod/mk_elfconfig

01:42:30   HOSTCC  scripts/dtc/fstree.o

01:42:30   HOSTCC  scripts/dtc/flattree.o

01:42:30   CC      scripts/mod/devicetable-offsets.s

01:42:30 scripts/extract-cert.c:21:25: fatal error: openssl/bio.h: No such file or directory

01:42:30  #include <openssl/bio.h>

01:42:30                          ^

01:42:30 compilation terminated.

01:42:30   HOSTCC  scripts/dtc/data.o

01:42:30   HOSTCC  scripts/dtc/livetree.o

01:42:30   HOSTCC  scripts/dtc/treesource.o

01:42:30 scripts/sign-file.c:25:30: fatal error: openssl/opensslv.h: No such file or directory

01:42:30  #include <openssl/opensslv.h>

01:42:30                               ^

01:42:30 compilation terminated.

01:42:30   HOSTCC  scripts/dtc/srcpos.o

01:42:30   HOSTCC  scripts/dtc/checks.o

01:42:30   GEN     scripts/mod/devicetable-offsets.h

01:42:30   MKELF   scripts/mod/elfconfig.h

01:42:30   HOSTCC  scripts/mod/modpost.o

01:42:30   HOSTCC  scripts/mod/file2alias.o

01:42:30   HOSTCC  scripts/mod/sumversion.o

01:42:30   HOSTCC  scripts/dtc/util.o

01:42:30   SHIPPED scripts/dtc/dtc-lexer.lex.c

01:42:30   SHIPPED scripts/dtc/dtc-parser.tab.h

01:42:30   SHIPPED scripts/dtc/dtc-parser.tab.c

01:42:30   HOSTCC  scripts/dtc/dtc-lexer.lex.o

01:42:30 make[2]: *** [scripts/extract-cert] Error 1

01:42:30 make[2]: *** Waiting for unfinished jobs....

01:42:30   HOSTCC  scripts/dtc/dtc-parser.tab.o

01:42:30 make[2]: *** [scripts/sign-file] Error 1

01:42:30   HOSTLD  scripts/dtc/dtc

01:42:30   HOSTLD  scripts/mod/modpost

01:42:30 make[1]: *** [scripts] Error 2

01:42:30 make[1]: Leaving directory `/tmp/pnor_20170322_1636/output/build/linux-4.10.3'

01:42:30 make: *** [/tmp/pnor_20170322_1636/output/build/linux-4.10.3/.stamp_built] Error 2

01:42:30 make: Leaving directory `/tmp/pnor_20170322_1636/buildroot'

dcrowell77, 2017-03-23:
> 01:42:30 scripts/extract-cert.c:21:25: fatal error: openssl/bio.h: No such file or directory
A new ssl dependency seems to have been added recently.  You need to install libssl-dev on your build machine to get it passing.

@stewart-ibm - Someone thought this might be related to some new signing stuff that was recently added to skiboot.  Thoughts?

causten, 2017-03-23:
Other projects also hit this issue.  You will need something like... https://gerrit.openbmc-project.xyz/#/c/3211/

stewart-ibm, 2017-03-23:
Dan <notifications@github.com> writes:
> A new ssl dependency seems to have been added recently.  You need to install libssl-dev on your build machine to get it passing.
>
> @stewart-ibm - Someone thought this might be related to some new [...]

It's the kernel module signing we enabled in
1b5f7274de58b840efbc06a11e27ef68a6dc59db

Luckily, everybody uses a different set of scripts to set up a build
environment, so everybody finds this out (or not) in different ways :/

-- 
Stewart Smith
OPAL Architect, IBM.

pridhiviraj, 2017-03-24:
@stewart-ibm we need to add respective openssl dev/libssl dev package in ci/Dockerfile script to make the build succeed.

closed 2017-03-27 by stewart-ibm