open-power/op-build #1029: ipmitool 1.8.18 does not build with USB support

sammj, 2017-04-18:
The following commit in ipmitool changes USB support defaults in the configure script: https://sourceforge.net/p/ipmitool/source/ci/27ac0c15acc17fd821b29efa8412b9ee4346f467

It's meant to disable USB support on non-Linux systems by chekcing the `host_os` variable for `gnu*` however at least on at least Ubuntu 16.04 the actual value is `linux-gnu` which does not match, and so we build without USB support.

The two main ways to work around this would be either
- Submit a patch to ipmitool upstream to make the check more inclusive, and/or
- Patch Buildroot to allow explicitly setting `--enable-intf-usb` at configure time.

Any thoughts?

stewart-ibm, 2017-07-20:
ipmitool quality. Hrm... maybe we should just patch it for the time being.

sammj, 2017-07-20:
I've posted a patch to the buildroot list which adds an option to enable usb support: https://patchwork.ozlabs.org/patch/791334/

sammj, 2017-07-24:
Merged to master here: https://github.com/buildroot/buildroot/commit/dd4979288666d3dc852d34c4b5d22d78f0221d0a

closed 2017-07-27 by stewart-ibm