open-power/op-build #1093: P9 SBE build process places p9_10.p9n_10.sbe_seeprom.bin file at root of source directory.

rlippert, 2017-05-18:
After running a power9 build, an output file from the SBE build is accidentally placed at the root of the buildroot directory which shows up as an untracked added file in git:

```
p9_10.p9n_10.sbe_seeprom.bin
```

I think the offending code that creates this file is here: https://github.com/open-power/sbe/blob/master/src/import/tools/imageProcs/p9_ipl_build.C#L337

The source comment mentions this is a "debug" file so I don't think it is actually used as part of the build.

dcrowell77, 2017-05-19:
@sgupta2m @pragupta

Shakeebbk, 2017-05-31:
@pragupta @rjknight ipl build tool is creating this temporary file - p9_10.p9n_10.sbe_seeprom.bin, in addition to the actual customized output file. Do we need this file for something or can it be removed?

pragupta, 2017-06-16:
I have pushed up code to fix this.  It is currently in the review process.

pragupta, 2017-07-03:
fixed by the following commit: https://github.com/open-power/sbe/commit/a36220b33148ceb1cb8fe78b1082e2fb3c485858

closed 2017-07-03 by pragupta