open-power/op-build #1194: hostboot patches fail silently

shenki, 2017-06-28:
When a patch in the openpower/packages/hostboot/hostbootP9 directory fails to apply, the build continues silently without applying further patches.

Instead it should fail when applying the patch, as something has gone wrong.

stewart-ibm, 2017-06-28:
Joel Stanley <notifications@github.com> writes:
> When a patch in the openpower/packages/hostboot/hostbootP9 directory fails to apply, the build continues silently without applying further patches.
>
> Instead it should fail when applying the patch, as something has gone
> wrong.

The best way to fix this is probably to split hostboot (and occ) out
into hostboot-p8 and hostboot-p9, that way we'll use the normal
buildroot patching rather than the custom stuff in there now.

-- 
Stewart Smith
OPAL Architect, IBM.

dcrowell77, 2017-06-28:
What do you mean by split?  Into unique repos?  packages?

@scprek @e-liner @williamspatrick - Please weigh in

stewart-ibm, 2017-06-28:
Dan <notifications@github.com> writes:
> What do you mean by split?  Into unique repos?  packages?
>
> @scprek @e-liner @williamspatrick - Please weigh in

Just into unique packages in op-build, they can each point to the
respective branch in the HB repository

-- 
Stewart Smith
OPAL Architect, IBM.

zane131, 2017-08-18:
I think I just ran into this same issue in OCC with today's op-build/master. This patch is stale: https://github.com/open-power/op-build/blob/master/openpower/package/occ/p9Patches/0001-Use-gnu89-as-C-std.patch. It fails to apply because it has already been applied. In my sandbox, I had a second patch behind it that never gets applied because of the issue with the first patch. I would expect op-build to fail. Instead, I didn't realize my patch had not been applied until I tested the code on real hardware.

closed 2018-05-03 by stewart-ibm