open-power/op-build #1199: OS booting issue

gkeishin, 2017-06-28:
Driver info:
```
BMC Driver build:
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-62"
VERSION_ID="v1.99.7-62-g62e2c5e"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-62"
BUILD_ID="v1.99.7"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-9e95993e2d6edc68af5ca3f2345789e0f450d0db #1 Wed Jun 28 15:20:49 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#

PNOR:
IBM-witherspoon-ibm-OP9_v1.17_1.37
        op-build-v1.17-77-g5100413-dirty
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-123-g4df53124cb34
        hostboot-44d178b
        linux-4.11.2-openpower1-p89dc3bc
        petitboot-v1.4.3-p5b26c1a
        machine-xml-d0f86cf
        occ-1dc97a6
        hostboot-binaries-24995fc
        capp-ucode-9c73e9f
        sbe-d770027

```

```
[    0.000000] Linux version 4.11.2-openpower1 (fspbld@gfwa610.aus.stglabs.ibm.com) (gcc version 4.9.4 (Buildroot 2017.02.2-00007-g23118ce) ) #2 SMP Thu Jun 22 08:45:24 CDT 2017
[    0.000000] OPAL: Found memory mapped LPC bus on chip 0
[    0.000000] Using PowerNV machine description
[    0.000000] bootconsole [udbg0] enabled
[    0.000000] CPU maps initialized for 4 threads per core
 -> smp_release_cpus()
spinning_secondaries = 31
 <- smp_release_cpus()
[    0.000000] -----------------------------------------------------
[    0.000000] ppc64_pft_size    = 0x0
[    0.000000] phys_mem_size     = 0x8000000000
[    0.000000] dcache_bsize      = 0x80
[    0.000000] icache_bsize      = 0x80
[    0.000000] cpu_features      = 0x477c7afd18500249
[    0.000000]   possible        = 0x7f7fffff18500649
[    0.000000]   always          = 0x0000000018100040
[    0.000000] cpu_user_features = 0xdc0065c2 0xaec00000
[    0.000000] mmu_features      = 0x3c004041
[    0.000000] firmware_features = 0x0000000010000000
[    0.000000] -----------------------------------------------------
[    0.000000] numa: Initmem setup node 0 [mem 0x00000000-0x3fffffffff]
[    0.000000] numa:   NODE_DATA [mem 0x3fffff6700-0x3fffffffff]
[    0.000000] numa: Initmem setup node 8 [mem 0x200000000000-0x203fffffffff]
[    0.000000] numa:   NODE_DATA [mem 0x203fff776700-0x203fff77ffff]
[    0.000000] numa: Initmem setup node 250
[    0.000000] numa:   NODE_DATA [mem 0x203fff72ce00-0x203fff7366ff]
[    0.000000] numa:     NODE_DATA(250) on node 8
[    0.000000] numa: Initmem setup node 251
[    0.000000] numa:   NODE_DATA [mem 0x203fff723500-0x203fff72cdff]
[    0.000000] numa:     NODE_DATA(251) on node 8
[    0.000000] numa: Initmem setup node 252
[    0.000000] numa:   NODE_DATA [mem 0x203fff719c00-0x203fff7234ff]
[    0.000000] numa:     NODE_DATA(252) on node 8
[    0.000000] numa: Initmem setup node 253
[    0.000000] numa:   NODE_DATA [mem 0x203fff710300-0x203fff719bff]
[    0.000000] numa:     NODE_DATA(253) on node 8
[    0.000000] numa: Initmem setup node 254
[    0.000000] numa:   NODE_DATA [mem 0x203fff706a00-0x203fff7102ff]
[    0.000000] numa:     NODE_DATA(254) on node 8
[    0.000000] numa: Initmem setup node 255
[    0.000000] numa:   NODE_DATA [mem 0x203fff6fd100-0x203fff7069ff]
[    0.000000] numa:     NODE_DATA(255) on node 8
[    0.000000] Initializing IODA2 PHB (/pciex@600c3c0000000)
[    0.000000] PCI host bridge /pciex@600c3c0000000 (primary) ranges:
[    0.000000]  MEM 0x000600c000000000..0x000600c07ffeffff -> 0x0000000080000000
[    0.000000]  MEM 0x0006000000000000..0x0006003fffffffff -> 0x0006000000000000 (M64 #1..31)
[    0.000000]  Using M64 #31 as default window
[    0.000000]   256 (255) PE's M32: 0x80000000 [segment=0x800000]
[    0.000000]                  M64: 0x4000000000 [segment=0x40000000]
[    0.000000]   Allocated bitmap for 4088 MSIs (base IRQ 0xfe000)
[    0.000000] Initializing IODA2 PHB (/pciex@600c3c0100000)
[    0.000000] PCI host bridge /pciex@600c3c0100000  ranges:
[    0.000000]  MEM 0x000600c080000000..0x000600c0fffeffff -> 0x0000000080000000
[    0.000000]  MEM 0x0006004000000000..0x0006007fffffffff -> 0x0006004000000000 (M64 #1..15)
[    0.000000]  Using M64 #15 as default window
[    0.000000]   128 (127) PE's M32: 0x80000000 [segment=0x1000000]
[    0.000000]                  M64: 0x4000000000 [segment=0x80000000]
[    0.000000]   Allocated bitmap for 2040 MSIs (base IRQ 0xfd800)
[    0.000000] Initializing IODA2 PHB (/pciex@600c3c0200000)
```

- [x] Host SOL log : 
[Host_SOL_LOG_hang.txt](https://github.com/open-power/op-build/files/1109582/Host_SOL_LOG_hang.txt)
- [x] BMC journald log : 
[journald_v1.99.7-62-g62e2c5e_os_hang.txt](https://github.com/open-power/op-build/files/1109596/journald_v1.99.7-62-g62e2c5e_os_hang.txt)

gkeishin, 2017-06-28:
@stewart-ibm @sammj  Could you take a look at this.

sammj, 2017-06-28:
A bit odd that it should stop there without an error or any other output. Are you connected via SSH console or IPMI? Does switching to one or the other help? I wonder if it's just a console issue.
The OpenBMC log looks a bit unhappy (a few NCSI kernel warnings) but an OpenBMC person will be better able to interpret that.

gkeishin, 2017-06-29:
For the NSCI kernel warning, we got https://github.com/openbmc/openbmc/issues/1597

The SSH port 2200 is hung but I can check on it once it hit it again if I can access via IPMI as well

gkeishin, 2017-06-30:
Yeah turns out the Host SOL froze on rebooting from OS and when booting it back. Same fate with IPMI SOL too.

Opened a defect for it https://github.com/openbmc/openbmc/issues/1882

Thanks for info @sammj 
Closing this defect.

closed 2017-06-30 by gkeishin