open-power/op-build #1235: Build PNOR for habanero failed with "Cannot allocate memory."

pridhiviraj, 2017-07-11:
16:01:44 mv -f /tmp/pnor_20170711_1546/output/build/glibc-2.25/build/catgets/stamp.oST /tmp/pnor_20170711_1546/output/build/glibc-2.25/build/catgets/stamp.oS

16:01:44 /tmp/pnor_20170711_1546/output/host/usr/bin/powerpc64le-buildroot-linux-gnu-gcc open_catalog.c -c -std=gnu11 -fgnu89-inline  -O2 -Wall -Werror -Wundef -Wwrite-strings -fmerge-all-constants -fno-stack-protector -frounding-math -Wstrict-prototypes -Wold-style-definition -mlong-double-128  -fpic   -ftls-model=initial-exec     -DNLSPATH='"/usr/share/locale/%L/%N:/usr/share/locale/%L/LC_MESSAGES/%N:/usr/share/locale/%l/%N:/usr/share/locale/%l/LC_MESSAGES/%N:"' -I../include -I/tmp/pnor_20170711_1546/output/build/glibc-2.25/build/catgets  -I/tmp/pnor_20170711_1546/output/build/glibc-2.25/build  -I../sysdeps/unix/sysv/linux/powerpc/powerpc64/fpu  -I../sysdeps/unix/sysv/linux/powerpc/powerpc64  -I../sysdeps/unix/sysv/linux/wordsize-64  -I../sysdeps/unix/sysv/linux/powerpc  -I../sysdeps/powerpc/nptl  -I../sysdeps/unix/sysv/linux/include -I../sysdeps/unix/sysv/linux  -I../sysdeps/nptl  -I../sysdeps/pthread  -I../sysdeps/gnu  -I../sysdeps/unix/inet  -I../sysdeps/unix/sysv  -I../sysdeps/unix/powerpc  -I../sysdeps/unix  -I../sysdeps/posix  -I../sysdeps/powerpc/powerpc64/fpu/multiarch  -I../sysdeps/powerpc/powerpc64/fpu  -I../sysdeps/powerpc/powerpc64/multiarch  -I../sysdeps/powerpc/powerpc64  -I../sysdeps/wordsize-64  -I../sysdeps/powerpc/fpu  -I../sysdeps/powerpc  -I../sysdeps/ieee754/ldbl-128ibm/include -I../sysdeps/ieee754/ldbl-128ibm  -I../sysdeps/ieee754/ldbl-opt  -I../sysdeps/ieee754/dbl-64  -I../sysdeps/ieee754/flt-32  -I../sysdeps/ieee754  -I../sysdeps/generic  -I.. -I../libio -I. -nostdinc -isystem /tmp/pnor_20170711_1546/output/host/usr/lib/gcc/powerpc64le-buildroot-linux-gnu/6.3.0/include -isystem /tmp/pnor_20170711_1546/output/host/usr/lib/gcc/powerpc64le-buildroot-linux-gnu/6.3.0/include-fixed -isystem /tmp/pnor_20170711_1546/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/usr/include  -D_LIBC_REENTRANT -include /tmp/pnor_20170711_1546/output/build/glibc-2.25/build/libc-modules.h -DMODULE_NAME=libc -include ../include/libc-symbols.h  -DPIC -DSHARED     -o /tmp/pnor_20170711_1546/output/build/glibc-2.25/build/catgets/open_catalog.os -MD -MP -MF /tmp/pnor_20170711_1546/output/build/glibc-2.25/build/catgets/open_catalog.os.dt -MT /tmp/pnor_20170711_1546/output/build/glibc-2.25/build/catgets/open_catalog.os

16:01:44 echo 'catgets/catgets.o catgets/open_catalog.o' > /tmp/pnor_20170711_1546/output/build/glibc-2.25/build/catgets/stamp.oT

16:01:44 mv -f /tmp/pnor_20170711_1546/output/build/glibc-2.25/build/catgets/stamp.oT /tmp/pnor_20170711_1546/output/build/glibc-2.25/build/catgets/stamp.o

16:01:44 echo 'catgets/catgets.os catgets/open_catalog.os' > /tmp/pnor_20170711_1546/output/build/glibc-2.25/build/catgets/stamp.osT

16:01:44 mv -f /tmp/pnor_20170711_1546/output/build/glibc-2.25/build/catgets/stamp.osT /tmp/pnor_20170711_1546/output/build/glibc-2.25/build/catgets/stamp.os

16:01:44 make[3]: Leaving directory '/tmp/pnor_20170711_1546/output/build/glibc-2.25/catgets'

16:01:44 /usr/bin/make  subdir=math -C math ..=../ subdir_lib

16:01:44 make[3]: Entering directory '/tmp/pnor_20170711_1546/output/build/glibc-2.25/math'

16:01:44 ../Makerules:367: *** Cannot allocate memory.  Stop.

16:01:44 make[3]: Leaving directory '/tmp/pnor_20170711_1546/output/build/glibc-2.25/math'

16:01:44 Makefile:215: recipe for target 'math/subdir_lib' failed

16:01:44 make[2]: *** [math/subdir_lib] Error 2

16:01:44 make[2]: Leaving directory '/tmp/pnor_20170711_1546/output/build/glibc-2.25'

16:01:44 Makefile:9: recipe for target 'all' failed

16:01:44 make[1]: *** [all] Error 2

16:01:44 make[1]: Leaving directory '/tmp/pnor_20170711_1546/output/build/glibc-2.25/build'

16:01:44 package/pkg-generic.mk:227: recipe for target '/tmp/pnor_20170711_1546/output/build/glibc-2.25/.stamp_built' failed

16:01:44 make: *** [/tmp/pnor_20170711_1546/output/build/glibc-2.25/.stamp_built] Error 2

16:01:44 make: Leaving directory '/tmp/pnor_20170711_1546/buildroot'

op-build commit ID:9b8bf2632d7cbfc244470484afd471c6b8756d38

pridhiviraj, 2017-07-11:
Some thing looks memory problem with my build machine where it has 16G is completely using it.
free -h
             total       used       free     shared    buffers     cached
Mem:           15G        15G       278M       280K       838M        12G
-/+ buffers/cache:       2.5G        13G
Swap:          15G       1.0G        14G
I will check and close this if it is some thing memory issue with my build machine

pridhiviraj, 2017-07-12:
Still seeing the issue even after doing system reboot.
Before build starts:
free -h
             total       used       free     shared    buffers     cached
Mem:           15G       792M        14G       856K       267M       230M
-/+ buffers/cache:       293M        15G
Swap:          15G         0B        15G

At the time of build failure:
free -h
             total       used       free     shared    buffers     cached
Mem:           15G       3.7G        11G       872K       342M       2.8G
-/+ buffers/cache:       645M        15G
Swap:          15G         0B        15G

dcrowell77, 2017-07-12:
I've never seen anything similar, maybe @stewart-ibm has some thoughts.

stewart-ibm, 2017-07-17:
Dan <notifications@github.com> writes:
> I've never seen anything similar, maybe @stewart-ibm has some
> thoughts.

I haven't seen it either... Then again, I generally don't build on such
small amounts of memory.

I'll try on my laptop and see if I have an issue... I think it's about
the smallest machine I can find (well, that isn't a service processor).

-- 
Stewart Smith
OPAL Architect, IBM.

williamspatrick, 2017-07-18:
You can limit the memory available to a Docker container.

stewart-ibm, 2017-12-01:
I strongly suspect this could be related to the ratio of CPU threads to memory and you could just be being unfortunate and need to cut down the parallelism of the build.... Ho many CPUs did this bulid machine have?

pridhiviraj, 2017-12-05:
@stewart-ibm It has 12 CPU's.
```
lscpu
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                12
On-line CPU(s) list:   0-11
Thread(s) per core:    2
Core(s) per socket:    6
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 44
Stepping:              2
CPU MHz:               1596.000
BogoMIPS:              6933.62
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              12288K
NUMA node0 CPU(s):     0-11
```
After clearing the cache memory using below command, i am able to build it successfully without any out of memory failures.
`sync && sysctl -w vm.drop_caches=3`

pridhiviraj, 2018-02-06:
As we have a WA, will close this.

closed 2018-02-06 by pridhiviraj