open-power/op-build #1259: Inband IPMI[USB Interface] fails to work in Petitboot kernel.

pridhiviraj, 2017-07-19:
[console-pexpect]#lsmod
Module                  Size  Used by    Not tainted
ext4                  401262  1 
jbd2                   76529  1 ext4
mbcache                 8727  1 ext4
btrfs                 964581  1 
xor                    11350  1 btrfs
raid6_pq              156396  1 btrfs
dm_snapshot            41808  6 
dm_bufio               23691  1 dm_snapshot
dm_mod                114949 15 dm_snapshot,dm_bufio
dax                    88649  1 dm_mod
ofpart                  6297  0 
powernv_flash           6992  0 
mtd                    52704  3 ofpart,powernv_flash
bnx2x                 649427  0 
ast                    58481  1 
sr_mod                 20607  0 
cdrom                  33909  1 sr_mod
sd_mod                 39100  3 
sg                     34738  0 
zlib_inflate           17464  2 btrfs,bnx2x
[console-pexpect]#modprobe sg
[console-pexpect]#lsmod
Module                  Size  Used by    Not tainted
ext4                  401262  1 
jbd2                   76529  1 ext4
mbcache                 8727  1 ext4
btrfs                 964581  1 
xor                    11350  1 btrfs
raid6_pq              156396  1 btrfs
dm_snapshot            41808  6 
dm_bufio               23691  1 dm_snapshot
dm_mod                114949 15 dm_snapshot,dm_bufio
dax                    88649  1 dm_mod
ofpart                  6297  0 
powernv_flash           6992  0 
mtd                    52704  3 ofpart,powernv_flash
bnx2x                 649427  0 
ast                    58481  1 
sr_mod                 20607  0 
cdrom                  33909  1 sr_mod
sd_mod                 39100  3 
sg                     34738  0 
zlib_inflate           17464  2 btrfs,bnx2x
[console-pexpect]#uname -a
Linux skiroot 4.12.1-openpower1 #2 SMP Mon Jul 17 06:39:40 EDT 2017 ppc64le GNU/Linux
[console-pexpect]#cat /etc/os-release 
NAME=Buildroot
VERSION=2017.05.1-00093-g9b91216
ID=buildroot
VERSION_ID=2017.05.1
PRETTY_NAME="Buildroot 2017.05.1"
[console-pexpect]#ipmitool power statu
Invalid chassis power command: statu
[console-pexpect]#ipmitool power status
Chassis Power is on
[console-pexpect]#ipmitool -I usb power status
Error loading interface usb
[console-pexpect]#ipmitool -V
ipmitool version 1.8.18
PNOR Level:
ipmitool fru print 43
 Product Name          : OpenPOWER Firmware
 Product Version       : open-power-habanero-v1.17-179-ga9194c6
 Product Extra         : 	buildroot-2017.05-93-g9b91216
 Product Extra         : 	skiboot-v5.7-rc2
 Product Extra         : 	hostboot-695bd89
 Product Extra         : 	linux-4.12.1-openpower1-p6b5026c
 Product Extra         : 	petitboot-v1.5.0-pe45a591
 Product Extra         : 	machine-xml-6a78496-p871f7b5
 Product Extra         : 	occ-28f2cec
 Product Extra         : 	hostboo

sammj, 2017-07-19:
Unless anything has changed recently (and I don't think it has), this is due to https://github.com/open-power/op-build/issues/1029

sammj, 2017-07-19:
> ipmitool version 1.8.18

Indeed this version has the issue in https://github.com/open-power/op-build/issues/1029.
I'll see if I can scrounge together some free time and give the ipmitool people a patch.

stewart-ibm, 2017-07-20:
Closing as duplicate, Sam is putting together a patch

closed 2017-07-20 by stewart-ibm

pridhiviraj, 2017-07-20:
@sammj @stewart-ibm Thanks. I didn't look at #1029 before raising issue. Is it some thing will affect in host os ipmitool also?

stewart-ibm, 2017-07-20:
Paidipeddi Pridhiviraj <notifications@github.com> writes:
> @sammj @stewart-ibm Thanks. I didn't look at #1029 before raising
> issue. Is it some thing will affect in host os ipmitool also?

It won't affect host, just ipmitool in petitboot.

-- 
Stewart Smith
OPAL Architect, IBM.

pridhiviraj, 2017-07-20:
@stewart-ibm Okay, Thanks.