open-power/op-build #1276: Console buffer data override.

pridhiviraj, 2017-07-25:
When system tried to boot with upstream PNOR, by the time it reaches to petitboot previous boot time console data(Skiboot and Skiroot messages) is getting overwritten with petitboot UI.
So users will be missing important boot time data.

Below is the one of the boot time data we can observe lot of data is getting overwritten.

[-1;-1f[-1;-1fUbuntu 16.04[-1;-1f.  .  .  .[-1;-1fUbuntu 16.04[-1;-1f.  .  .  .reboot: Restarting system
[  301.090849247,5] OPAL: Reboot request...
[  301.090896718,5] RESET: Initiating fast reboot 2...
[  301.298821234,5] PSI: Hot reset!
[  301.299994169,7] FLASH: BOOTKERNEL partition doesn't have ECC
[  301.301364117,7] FLASH: BOOTKERNEL partition isn't signed
[  301.301366173,7] FLASH: computed BOOTKERNEL size 15304552
[  311.191809089,6] STB: sb_verify skipped resource 0, secure_mode=0
[  311.191812131,6] STB: tb_measure skipped resource 0, trusted_mode=0
[  311.191837046,7] FFS: Partition map size: 0x1000
[  311.193224612,7] FLASH: No ROOTFS partition
[  311.252170891,5] PCI: Clearing all devices...
[  316.458362407,5] PCI: Resetting PHBs and training links...
[  319.048940322,5] PCI: Probing slots...
[  319.105384114,5] PHB#0000:00:00.0 [ROOT] 1014 03dc R:00 C:060400 B:01..01 SLOT=Slot3 
[  319.105678459,5] PHB#0000:01:00.0 [EP  ] 14e4 168a R:10 C:020000 (      ethernet) LOC_CODE=Slot3
[  319.114882851,5] PHB#0000:01:00.1 [EP  ] 14e4 168a R:10 C:020000 (      ethernet) LOC_CODE=Slot3
[  319.118046033,5] PHB#0000:01:00.2 [EP  ] 14e4 168a R:10 C:020000 (      ethernet) LOC_CODE=Slot3
 Petitboot (v1.5.0-pe45a591)                          8348-21C 6856D5A
 ──────────────────────────────────────────────────────────────────────────────
    Ubuntu, with Linux 4.4.0-78-generic (recovery mode)
    Ubuntu, with Linux 4.4.0-78-generic
    Ubuntu, with Linux 4.4.0-79-generic (recovery mode)
    Ubuntu, with Linux 4.4.0-79-generic
    Ubuntu, with Linux 4.4.0-84-generic (recovery mode)
    Ubuntu, with Linux 4.4.0-84-generic
    Ubuntu, with Linux 4.4.0-85-generic (recovery mode)
    Ubuntu, with Linux 4.4.0-85-generic
    Ubuntu, with Linux 4.13.0-rc2 (recovery mode)
    Ubuntu, with Linux 4.13.0-rc2
    Ubuntu

  System information
  System configuration
  System status log
  Language
  Rescan devices
  Retrieve config from URL
 *Exit to shell                                        
 ──────────────────────────────────────────────────────────────────────────────
 Enter=accept, e=edit, n=new, x=exit, l=language, g=log, h=help
 [sdb2] Parsed GRUB configuration from /boot/grub/grub.cfg

sammj, 2017-07-26:
Is this just the kernel booting with the `quiet` parameter? This is intentional, as on certain P8 platforms the BMC's SOL implementation can slow down boot considerably.
Otherwise, has anything changed here?

pridhiviraj, 2017-07-26:
it's not about skiroot quiet. it is about console data loss or override during boot time. While booting messages are coming. When it reaches to petitboot, part of previous console data is getting overwritten with petitboot UI. You can see in the above we are able to see half of the skiboot messages remaining boot messages are occupied with petitboot UI.

sammj, 2017-07-26:
Ah ok. What are you using to view the IPMI console?
What's probably happening is that ncurses is just wiping the screen and drawing the UI on top - in most terminals this will take up the entire console.
I don't think there's a bug here - if you reach the UI you'll need to drop to the shell to view the skiboot/kernel logs.

sammj, 2017-07-28:
By the time the UI comes up we _should_ be well and truly through the boot process. Before then and the log will still be intact since ncurses won't have drawn to the screen yet. Is there another scenario?