open-power/op-build #1288: Installing CentOS in Tyan Palmetto

zazola, 2017-07-28:
I have tried to install CentOS 7.3 ppc64le in a Tyan Palmetto server. I have installed it correctly and booted up fine, but after the first reboot I did to the machine, any entry in the petitboot menu regarding the CentOS installation is gone. There is nothing wrong with the hardware or disks, since this is not happening with Ubuntu or Fedora.

Tyan says Ubuntu is the only OS supported in this machine. Did anyone try to install CentOS in this server? Did anyone experience this kind of issue?

sammj, 2017-07-30:
I recall CentOS had an issue where it didn't fully complete the installation on POWER - this blog post covers the issues, have a look and see if that's what affecting you (in particular the grub section at the bottom): https://sthbrx.github.io/blog/2017/01/30/installing-centos-72-on-ibm-power-systems-s822lc-for-high-performance-computing-minksy-with-usb-device/

If CentOS still isn't appearing after that, please post the `pb-sos` output from Petitboot.

zazola, 2017-07-31:
Thanks for the advice sammj.
The problem I have appears when installing CentOS 7.3 but not with CentOS 7.2.
I can actually install 7.3 and after post-installation reboot it's fine. The problem comes when performing the next reboot, Petitboot does not show the installation entry in the menu.
The Petitboot version I have is dev-20141107, which is pretty old. I don't know if the bootloader changed significantly from CentOS 7.2 to 7.3 and that's causing the problem.

zazola, 2017-07-31:
I tried the solution on the thread you pointed out, but it's not working.
I don't have pb-sos in my Petitboot console so, I can just post the logs under /var/log/petitboot
I enclose two files, one (the one labeled with "1st" in the filename) with the menu entries correctly set, and the other one (with "2nd" in the filename), which is the one not working.
I don't think the logs tell anything meaningful, but I don't have any other think I can report.
[petitboot_hvc_1stcentos73.txt](https://github.com/open-power/op-build/files/1188106/petitboot_hvc_1stcentos73.txt)
[petitboot_hvc_2ndcentos73.txt](https://github.com/open-power/op-build/files/1188108/petitboot_hvc_2ndcentos73.txt)

sammj, 2017-08-01:
Ah, I wonder what changed between 7.2 and 7.2.

Those two log files are the logs for the ncurses interfaces, as you say they're not particularly useful in this case. The server log is `pb-discover.log` and will be more helpful, especially if you turn on verbose logging. To do so run
`nvram --update-config petitboot,debug?=true`
and reboot for it to take effect.

`dev-20141107` is indeed very old, but should still be serviceable for this (although if you get a chance I would recommend updating :) )

zazola, 2017-08-01:
I send you the pb-discover logs:
CentOS 7.2 installed in petitboot menu,
CentOS 7.3 after installing correctly shown in petitboot menu,
and CentOS 7.3 after post-installation reboot, without being displayed in the menu.

[petitboot_discover_1stcentos72.txt](https://github.com/open-power/op-build/files/1190046/petitboot_discover_1stcentos72.txt)
[petitboot_discover_1stcentos73.txt](https://github.com/open-power/op-build/files/1190047/petitboot_discover_1stcentos73.txt)
[petitboot_discover_2ndcentos73.txt](https://github.com/open-power/op-build/files/1190048/petitboot_discover_2ndcentos73.txt)

sammj, 2017-08-02:
```
        ID_FS_TYPE          : xfs
...
mounting device /dev/sda2 read-only
couldn't mount device /dev/sda2: mount failed: Input/output error
```

Considering the age of your firmware I'm tempted to think you could be running into a journal issue with XFS. From the non-booting 7.3 case, please post `/var/log/messages`. If we're lucky this can be fixed with just a firmware update.

zazola, 2017-08-02:
Tyan (machine's manufacturer) says there is no newer firmware available. I will rely on CentOS 7.2, which does not look to suffer from this issue at all. Thanks for your help.

sammj, 2017-08-04:
Yep I don't believe Tyan has released any newer builds. If you are comfortable flashing your machine yourself you can build a newer image from this repo, but of course Tyan are unlikely to support that :)

stewart-ibm, 2017-08-04:
A workaround could be to create an ext4 /boot with the newer CentOS, as I wonder if we're running into a journal format incompatibility between the 2014 (i.e. *really* early) firmware and modern XFS in CentOS 7.3.

I *think* we're a bit better off in coping with this on more modern builds.. but maybe we should brainstorm how we could test for that.

I'm not sure there's much more we can do without building+testing newer firmware (or convincing tyan they should ship an update... which is probably unlikely).

Hoping it's okay to close this issue for the time being

closed 2017-08-04 by stewart-ibm