open-power/op-build #1326: Should romulus have secureboot enabled or not

bofferdn, 2017-08-11:
Romulus hostboot config has:

unset CONFIG_SECUREBOOT
unset CONFIG_TPMDD
unset CONFIG_TPM_NUVOTON

But .. the CONFIG_ prefix actually creates the opposite effect of keeping secureboot compiled in, and nobody has complained thus far.  If Romulus really wants secureboot compiled out, the correct 
format is:

unset SECUREBOOT
unset TPMDD
unset TPM_NUVOTON

So, should Romulus keep secureboot logically compiled in (we can remove the existing unset directives), or out (we'll remove the CONFIG_ prefix)?

causten, 2017-08-11:
OpenPOWER Romulus was built as a reference for others designing OpenPOWER style hardware.  I think the default should be to disabled

bofferdn, 2017-08-11:
I would counter that with code in the op-build master today, setting the three policies above will actually keep the secureboot code compiled in and working with development keys.  A customer then has the option of configuring the secure jumper to secure the system or not.  If you want not to be secured, just set the jumper to disable system security.

stewart-ibm, 2017-08-21:
Personally, I don't tihnk there's value in building with secure boot off. The jumper should dictate the behavior, but we should always parse and verify as much as possible