open-power/op-build #1351: op-build pnor output is non deterministic

stewart-ibm, 2017-08-17:
```stewart@ka3:~/op-build$ hexdump -C output/images/p9dsu.pnor|head
00000000  50 41 52 54 00 00 00 01  00 00 00 01 00 00 00 80  |PART............|
00000010  00 00 00 1f 00 00 10 00  00 00 40 00 ff ff ff ff  |..........@.....|
00000020  ff ff ff ff ff ff ff ff  ff ff ff ff 50 41 02 cb  |............PA..|
```
and then `op-build openpower-pnor-rebuild`

```
stewart@ka3:~/op-build$ hexdump -C output/images/p9dsu.pnor|head
00000000  50 41 52 54 00 00 00 01  00 00 00 01 00 00 00 80  |PART............|
00000010  00 00 00 1f 00 00 10 00  00 00 40 00 00 00 00 00  |..........@.....|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 50 41 02 cb  |............PA..|
```

Except that isn't quite reliable, so you may need a few loops.

stewart-ibm, 2017-08-17:
Oh, and yes, this would make the checksum invalid

dcrowell77, 2017-08-17:
@scprek @bofferdn - might have some input

stewart-ibm, 2017-08-17:
I'm working on a rework on how we assemble PNOR that should simplify both how we assemble the PNOR image and how the OpenBMC squashfs.tar image is assembled (while not touching the hostboot scripts that seem also used for FSP builds) which should fix this issue... so it may not be worth investigating too closely (unless everybody universally hates my patch :)

cyrilbur-ibm, 2018-02-01:
I believe this is fixed. Are we right to close it?

stewart-ibm, 2018-05-28:
Yeah, I think this particular issue is fixed thanks to https://github.com/open-power/ffs/commit/fd98239a8beb83a3b3f61513860edb69123828d8

closed 2018-05-28 by stewart-ibm