open-power/op-build #1389: in petitboot fsck does not work as expected

bssrikanth, 2017-08-31:
In petitboot I see fsck command present but that does not work as expected..
# fsck
fsck (busybox 1.26.2, 2017-08-24 11:20:16 PDT)
fsck: fsck.ext2: No such file or directory

Samuel Mendoza-Jonas [5:35 AM] 
hmm, no i expect we should at least have .ext2 support
ah looks like busybox doesn't package it by default, i'll have a look

So this issue. Thank you.

Below are the levels of host fw, petitboot etc.:

 Product Name          : OpenPOWER Firmware
 Product Version       : open-power-SUPERMICRO-P9DSU-V0.12-20170824
 Product Extra         : 	op-build-da78c29-dirty
 Product Extra         : 	skiboot-f5d9049-pa7d7d42
 Product Extra         : 	hostboot-75333b1
 Product Extra         : 	linux-4.12.5-openpower1-pbf81229
 Product Extra         : 	petitboot-v1.5.1-p11bf40f
 Product Extra         : 	machine-xml-a134897
 Product Extra         : 	occ-71b5f68
 Product Extra         : 	sbe-4a62126

bssrikanth, 2017-08-31:
@sammj please take a look. Thank you.

stewart-ibm, 2017-08-31:
Srikanth Aithal <notifications@github.com> writes:
> In petitboot I see fsck command present but that does not work as expected..
> # fsck
> fsck (busybox 1.26.2, 2017-08-24 11:20:16 PDT)
> fsck: fsck.ext2: No such file or directory
>
> Samuel Mendoza-Jonas [...]

I think we may have had it way back in the pre-op-build Tuleta days as
we have more room to play with there (16mb for initramfs).

My guess is the size increase is the big problem.

-- 
Stewart Smith
OPAL Architect, IBM.

sammj, 2017-08-31:
I did a quick build comparison, looks like BR2_PACKAGE_UTIL_LINUX_FSCK adds 108KB. So it fits, but it depends on whether we're going to need that space for something else.

sammj, 2017-08-31:
Actually that could be up to 500KB depending on the combination of util-linux and e2fsprogs options we use.

stewart-ibm, 2017-09-01:
Sam Mendoza-Jonas <notifications@github.com> writes:
> Actually that could be up to 500KB depending on the combination of
> util-linux and e2fsprogs options we use.

That would be 0.5 seconds of boot time for every boot.

-- 
Stewart Smith
OPAL Architect, IBM.

stewart-ibm, 2018-05-03:
I'm inclined to close this as something we won't fix. I don't think this is so essential to have in firmware rather than something that could be a petitboot plugin (or a sysrescuecd type thing) that is loaded from USB or the network.

closed 2018-05-03 by stewart-ibm