open-power/op-build #1430: ppc64_cpu in petitboot ?

agangidi53, 2017-09-13:
There might be good reason for this. But in Power8 or Power9 servers (barreleye) I can't seem to get ppc64_cpu (frequency or smt) to get out an meaningful output (looks different cores are different SMT states ).  
Can someone please explain the reason for this, in a bit of detail ?

sammj, 2017-09-13:
Could you please post some examples of what you're seeing?
There's a new `v1.3.4` release for ppc64_cpu that could help here, we're in the process of updating to that now.

agangidi53, 2017-09-13:
@sammj 
It would nice to know the frequency before we kexec into our public cloud image. This will help our deployment team to dynamically migrate images between different free machines 

/ # ppc64_cpu --frequency
min:	18446744073.710 GHz (cpu -1)
max:	0.000 GHz (cpu -1)
avg:	0.000 GHz

sammj, 2017-09-13:
This upstream commit should fix that: https://github.com/ibm-power-utilities/powerpc-utils/commit/c25ae395db914f1137bb8a718277ea4b6f69136b
We'll pick it up when we update to the `v1.3.4` tag.

sammj, 2017-09-14:
https://github.com/open-power/op-build/pull/1432

stewart-ibm, 2017-09-18:
Merged in with https://github.com/open-power/op-build/pull/1432

closed 2017-09-18 by stewart-ibm