open-power/op-build #1433: Petitboot: Translations do not appear to be built:

sammj, 2017-09-13:
Seen on master:

```
>>> petitboot v1.5.1 Building
PATH="/scratch/sam/op-pseries/host/bin:/scratch/sam/op-pseries/host/sbin:/home/sam/bin:/home/sam/.local/bin:/opt/cross/gcc-4.8.0-nolibc/powerpc64-linux/bin/:/sbin:/usr/sbin:/home/sam/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/scratch/cyrilb/buildroot/builds/obmc/host/usr/bin"  /usr/bin/make -j41  -C /scratch/sam/op-pseries/build/petitboot-v1.5.1/
make[1]: Entering directory '/scratch/sam/op-pseries/build/petitboot-v1.5.1'
  YACC     discover/grub2/grub2-parser.c
  LEX      discover/grub2/grub2-lexer.h
  LEX      discover/grub2/grub2-lexer.c
updating discover/grub2/grub2-parser.h
/usr/bin/make  all-recursive
make[2]: Entering directory '/scratch/sam/op-pseries/build/petitboot-v1.5.1'
Making all in po
make[3]: Entering directory '/scratch/sam/op-pseries/build/petitboot-v1.5.1/po'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/scratch/sam/op-pseries/build/petitboot-v1.5.1/po'
make[3]: Entering directory '/scratch/sam/op-pseries/build/petitboot-v1.5.1'
  CC       ui/ncurses/ui_ncurses_petitboot_nc-generic-main.o
  CC       lib/fold/lib_libpbcore_la-fold.lo
...
```

sammj, 2017-09-14:
From master:
```
sam@ka3:/scratch/sam/op-pseries/target $ find . -name LC_MESSAGES | wc -l
0
```
From v1.19:
```
sam@ka3:/scratch/sam/op-pseries-alt $ find . -name LC_MESSAGES | wc -l
105
```
Most noticeably there was a Buildroot bump in between, perhaps a default option has changed.

sammj, 2017-09-14:
```
017.08-rc1, Released August 2nd, 2017

      ...

        - Major revamp of the gettext handling, with user-visible
          effect:

          - prior to this revamp, when BR2_ENABLE_LOCALE=y, each
            package could decide to enable or not NLS support. When
            BR2_ENABLE_LOCALE was disabled, NLS support was forced off
            by passing --disable-nls to packages.

          - after this revamp, a new BR2_SYSTEM_ENABLE_NLS option
            controls whether NLS support should be enabled or not in
            packages. This option defaults to disabled, which means
            that now, NLS support is by default disabled in all
            packages.

          Therefore, if you need NLS support in packages, you must now
          explicitly enable the BR2_SYSTEM_ENABLE_NLS option.
```

SIGH, patch incoming

stewart-ibm, 2017-09-18:
Merged, so closing issue.

closed 2017-09-18 by stewart-ibm