open-power/op-build #1458: Building PNOR images for Witherspoon failed.

pridhiviraj, 2017-09-19:
Combining Hash strings...

Writing updated hash||string file (../img/hbotStringFile)...

Hashing Started at Tue Sep 19 09:53:43 2017
Hashing Finished at Tue Sep 19 09:53:45 2017
make[1]: Leaving directory '/root/pridhvi/op-build/output/build/hostboot-d79906ed72501bf59cc4f1684aa58b9ee0c987fb'
>>> hostboot d79906ed72501bf59cc4f1684aa58b9ee0c987fb Installing to images directory
cd /root/pridhvi/op-build/output/build/hostboot-d79906ed72501bf59cc4f1684aa58b9ee0c987fb && source ./env.bash && /root/pridhvi/op-build/output/build/hostboot-d79906ed72501bf59cc4f1684aa58b9ee0c987fb/src/build/tools/hbDistribute --openpower /root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/
sbroot= sbname=
target dir /root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/
sbfw dir = invalid directory
In secure mode
Non FSP build, using default pnor xml layout
make[1]: Entering directory '/root/pridhvi/op-build/output/build/hostboot-d79906ed72501bf59cc4f1684aa58b9ee0c987fb/src/build/mkrules'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//fsp/attribute_types.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//fsp/attribute_types.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//tools/attribute_types.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//tools/attribute_types.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//openpower/attribute_types.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//openpower/attribute_types.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//vpo/attribute_types.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//vpo/attribute_types.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//attribute_types.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//attribute_types.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//fsp/attribute_types_hb.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//fsp/attribute_types_hb.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//tools/attribute_types_hb.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//tools/attribute_types_hb.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//openpower/attribute_types_hb.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//openpower/attribute_types_hb.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//vpo/attribute_types_hb.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//vpo/attribute_types_hb.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//attribute_types_hb.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//attribute_types_hb.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//fsp/target_types_hb.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//fsp/target_types_hb.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//tools/target_types_hb.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//tools/target_types_hb.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//openpower/target_types_hb.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//openpower/target_types_hb.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//vpo/target_types_hb.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//vpo/target_types_hb.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//target_types_hb.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//target_types_hb.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//fsp/attribute_types_oppowervm.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//fsp/attribute_types_oppowervm.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//tools/attribute_types_oppowervm.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//tools/attribute_types_oppowervm.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//openpower/attribute_types_oppowervm.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//openpower/attribute_types_oppowervm.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//vpo/attribute_types_oppowervm.xml'
dist.rules.mk:184: warning: ignoring old recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//vpo/attribute_types_oppowervm.xml'
dist.rules.mk:216: warning: overriding recipe for target '/root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//attribute_types_oppowervm.xml'
[... cut ...]
Can't use string ("0xFFFFFFFF") as a HASH ref while "strict refs" in use at /root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/Targets.pm line 253.
package/pkg-generic.mk:227: recipe for target '/root/pridhvi/op-build/output/build/machine-xml-f8aaa73b75564d6b845147afd3b2ab7c433838bc/.stamp_built' failed
make: *** [/root/pridhvi/op-build/output/build/machine-xml-f8aaa73b75564d6b845147afd3b2ab7c433838bc/.stamp_built] Error 255
make: Leaving directory '/root/pridhvi/op-build/buildroot'

The above failure is for witherspoon_sequoia defconfig.

dcrowell77, 2017-09-19:
Did you start from a fresh clone or are you running inside an old op-build?

Copy /root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/witherspoon.xml somewhere we can see it.  Also grab /root/pridhvi/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/Targets.pm

pridhiviraj, 2017-09-19:
I cloned earlier. But did a git pull and then tried building it.

[files.zip](https://github.com/open-power/op-build/files/1314398/files.zip)

dcrowell77, 2017-09-19:
We do see issues trying to rebuild after pulls sometimes so that doesn't shock me.  Which packages did you specifically update?  And you just did a full 'op-build' with no args?

dcrowell77, 2017-09-19:
Line 253 in Targets.pm is inside this function.

sub storeEnumerations
{
    my $self = shift;
    my $baseptr = $self->{xml}->{enumerationType};
    if ($self->{xml_version} == 1)
    {
        $baseptr = $self->{xml}->{enumerationTypes}->{enumerationType};
    }
    foreach my $enumType (keys(%{ $baseptr }))
    {
        foreach my $enum (
            keys(%{$baseptr->{$enumType}->{enumerator}}))
        {
            $self->{enumeration}->{$enumType}->{$enum} =
              $baseptr->{$enumType}->{enumerator}->{$enum}->{value};  <<<<
        }
    }
}

Unfortunately it is hard to tell which specific tag has the unrecognized value.  I suspect you've got some mismatch between your xml and the HB code.

stewart-ibm, 2017-12-01:
I think we've only ever seen this when there's a mismatch between XML and HB code... So, I'm going to close this for the time being.

A separate bug could be the awfulness of the error message.

closed 2017-12-01 by stewart-ibm