open-power/op-build #1650: PNOR Build for witherspoon failed.

pridhiviraj, 2017-11-23:
18:33:07 /home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/ppetools/ppetracepp /home/jenkins/workspace/buildPNOR/op-build/output/host/bin/powerpc64le-buildroot-linux-gnu-gcc -c -g -Wall -fsigned-char -msoft-float -m32 -mbig-endian -mcpu=405 -mmultiple -mstring -meabi -msdata=eabi -ffreestanding -fno-common -fno-inline-functions-called-once -std=gnu89 -pipe -Wa,-m405 -Os -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405 -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/.. -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/rtls -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/thread -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/incl -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/sensor -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/errl -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/trac -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/pss -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/timer -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/proc -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/cmdh -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/dcom -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/amec -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/cent -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/dimm -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/gpu -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/mem -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/lock -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/wof -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/../common -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occ_405/pgpe  -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/ssx/ssx -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/ssx/ppc32 -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/ssx/ppc405 -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/ssx/trace -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/ssx/occhw -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/ssx/../lib/common -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/ssx/../include -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/ssx/../include/registers -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/lib/occlib -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/lib/common -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/lib/ssxlib -I/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/lib/ppc405lib -DTRAC_TO_SIMICS=0 -DSIMICS_ENVIRONMENT=0 -DIMAGE_NAME=occ_405 -DSSX_TIMER_SUPPORT=1 -DSSX_THREAD_SUPPORT=1 -DPPC405_MMU_SUPPORT=0 -DSSX_TRACE_SUPPORT=1 -DSSX_TRACE_HASH_PREFIX=38241 -DSSX_USE_INIT_SECTION=0 -DUSE_SSX_APP_CFG_H=1 -D__SSX__=1 -DSSX_NO_BOOT_VECTOR0=1 -o /home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/ppc405lib/strcasecmp.o strcasecmp.c
18:33:07 /home/jenkins/workspace/buildPNOR/op-build/output/host/bin/powerpc64le-buildroot-linux-gnu-ar crs /home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/ppc405lib/libppc405.a /home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/ppc405lib/*.o
18:33:07 make[3]: Leaving directory '/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/lib/ppc405lib'
18:33:07 /home/jenkins/workspace/buildPNOR/op-build/output/host/bin/powerpc64le-buildroot-linux-gnu-cpp -E -x c -P -DTRAC_TO_SIMICS=0 -DSIMICS_ENVIRONMENT=0 -DIMAGE_NAME=occ_405 -DSSX_TIMER_SUPPORT=1 -DSSX_THREAD_SUPPORT=1 -DPPC405_MMU_SUPPORT=0 -DSSX_TRACE_SUPPORT=1 -DSSX_TRACE_HASH_PREFIX=38241 -DSSX_USE_INIT_SECTION=0 -DUSE_SSX_APP_CFG_H=1 -D__SSX__=1 -DSSX_NO_BOOT_VECTOR0=1 linkocc.cmd -o /home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/linkscript
18:33:07 /home/jenkins/workspace/buildPNOR/op-build/output/host/bin/powerpc64le-buildroot-linux-gnu-ld -e __ssx_boot -T/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/linkscript --oformat=elf32-powerpc -melf32ppc -Map /home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/occ_405.map -Bstatic -o /home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/occ_405.out -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405 -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/ssx -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/ssx/ppc405 -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/ssx/ppc32 -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/commonlib -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/occlib -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/ppc405lib -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/cmdh -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/dcom -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/dimm -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/errl -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/gpu -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/lock -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/pss -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/rtls -L/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occ_405/sensor -L/home/jenkins/workspace/buildPNOR/op-build/out [... cut ...]
18:33:12 make[1]: Leaving directory '/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src'
18:33:12 package/pkg-generic.mk:227: recipe for target '/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/.stamp_built' failed
18:33:12 make: *** [/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/.stamp_built] Error 2
18:33:12 make: Leaving directory '/home/jenkins/workspace/buildPNOR/op-build/buildroot

pridhiviraj, 2017-11-23:
Another instance it failed with below signatures(This time build with devel mode).
19:06:53     MAKE       hwas GEN
19:06:53     MAKE       test GEN
19:06:53     MAKE       fsiscom GEN
19:06:53     MAKE       test GEN
19:06:53     MAKE       intr GEN
19:06:53     MAKE       test GEN
19:06:53     MAKE       util GEN
19:06:53     MAKE       mbox GEN
19:06:53     MAKE       test GEN
19:06:53     MAKE       test GEN
19:06:53     MAKE       runtime GEN
19:06:53     MAKE       vpd GEN
19:06:53     MAKE       test GEN
19:06:53 ../../../src/build/mkrules/cc.rules.mk:74: warning: overriding recipe for target '../../../img/libvpd.so'
19:06:53 ../../../src/build/mkrules/cc.rules.mk:74: warning: ignoring old recipe for target '../../../img/libvpd.so'
19:06:53 ../../../src/build/mkrules/beam.rules.mk:75: warning: overriding recipe for target 'beamgen'
19:06:53 ../../../src/build/mkrules/beam.rules.mk:75: warning: ignoring old recipe for target 'beamgen'
19:06:53 ../../../src/build/mkrules/beam.rules.mk:90: warning: overriding recipe for target 'make_beamdir'
19:06:53 ../../../src/build/mkrules/beam.rules.mk:90: warning: ignoring old recipe for target 'make_beamdir'
19:06:53 ../../../src/build/mkrules/gcov.rules.mk:34: warning: overriding recipe for target 'make_gcovdir'
19:06:53 ../../../src/build/mkrules/gcov.rules.mk:34: warning: ignoring old recipe for target 'make_gcovdir'
19:06:53 ../../../src/build/mkrules/cscope.rules.mk:30: warning: overriding recipe for target 'cscope'
19:06:53 ../../../src/build/mkrules/cscope.rules.mk:30: warning: ignoring old recipe for target 'cscope'
19:06:53 ../../../src/build/mkrules/cscope.rules.mk:38: warning: overriding recipe for target 'ctags'
19:06:53 ../../../src/build/mkrules/cscope.rules.mk:38: warning: ignoring old recipe for target 'ctags'
19:06:53 ../../../src/build/mkrules/passes.env.mk:86: warning: overriding recipe for target 'clobber'
19:06:53 ../../../src/build/mkrules/passes.env.mk:86: warning: ignoring old recipe for target 'clobber'
19:06:53 ../../../src/build/mkrules/passes.rules.mk:45: warning: overriding recipe for target 'suppress_nothing_to_do'
19:06:53 ../../../src/build/mkrules/passes.rules.mk:45: warning: ignoring old recipe for target 'suppress_nothing_to_do'
19:06:53 ../../../src/build/mkrules/passes.rules.mk:105: warning: overriding recipe for target '_BUILD/PASSES/CODE/BODY'
19:06:53 ../../../src/build/mkrules/passes.rules.mk:105: warning: ignoring old recipe for target '_BUILD/PASSES/CODE/BODY'
19:06:53     MAKE       scan GEN
19:06:53 ../../../src/build/mkrules/passes.rules.mk:105: warning: overriding recipe for target '_BUILD/PASSES/GCOV/BODY'
19:06:53 ../../../src/build/mkrules/passes.rules.mk:105: warning: ignoring old recipe for target '_BUILD/PASSES/GCOV/BODY'
19:06:53 ../../../src/build/mkrules/passes.rules.mk:105: warning: overriding recipe for target '_BUILD/PASSES/GCOV/PRE'
19:06:53 ../../../src/build/mkrules/passes.rules.mk:105: warning: ignoring old recipe for target '_BUILD/PASSES/GCOV/PRE'
19:06:53 ../../../src/build/mkrules/passes.rules.mk:105: warning: overriding recipe for target '_BUILD/PASSES/BEAM/PRE'
19:06:53 ../../../src/build/mkrules/passes.rules.mk:105: warning: ignoring old recipe for target '_BUILD/PASSES/BEAM/PRE'
19:06:53 ../../../src/build/mkrules/cc.rules.mk:74: warning: overriding recipe for target '../../../img/libvpd.so'
19:06:53 ../../../src/build/mkrules/cc.rules.mk:74: warning: ignoring old recipe for target '../../../img/libvpd.so'
19:06:53 ../../../src/build/mkrules/beam.rules.mk:75: warning: overriding recipe for target 'beamgen'
19:06:53 ../../../src/build/mkrules/beam.rules.mk:75: warning: ignoring old recipe for target 'beamgen'
19:06:53 ../../../src/build/mkrules/beam.rules.mk:90: warning: overriding recipe for target 'make_beamdir'
19:06:53 ../../../src/build/mkrules/beam.rules.mk:90: warning: ignoring old recipe for target 'make_beamdir'
19:06:53 ../../../src/build/mkrules/gcov.rules.mk:34: warning: overriding recipe for target 'make_gcovdir'
19:06:53 ../../../src/build/mkrules/gcov.rules.mk:34: warning: ignoring old recipe for target 'make_gcovdir'
19:06:53 ../../../src/build/mkrules/cscope.rules.mk:30: warning: overriding recipe for target 'cscope'
19:06:53 ../../../src/build/mkrules/cscope.rules.mk:30: warning: ignoring old recipe for target 'cscope'
19:06:53 ../../../src/build/mkrules/cscope.rules.mk:38: warning: overriding recipe for target 'ctags'
19:06:53 ../../../src/build/mkrules/cscope.rules.mk:38: warning: ignoring old recipe for target 'ctags'
19:06:53 ../../../src/build/mkrules/passes.env.mk:86: warning: overriding recipe for target 'clobber'
19:06:53 ../../../src/build/mkrules/passes.env.mk:86: warning: ignoring old recipe for target 'clobber'
19:06:53 ../../../src/build/mkrules/passes.rules.mk:45: warning: overriding recipe for target 'suppress_nothing_to_do'
19:06:53 ../../../src/build/mkrules/passes.rules.mk:45: warning: ignoring old recipe for target 'suppress_nothing_to_do'
19:06:53 ../../../src/build/mkrules/passes.rules.mk:105: warning: overriding recipe for target '_BUILD/PASSES/CODE/BODY'
19:06:53 ../../../src/build/mkrules/passes.rules.mk:105: warning: ignoring old recipe for target '_BUILD/PASSES/CODE/BODY'
19:06:53     MAKE       test GEN
19:06:53 ../../../src/build/mkrules/passes.rules.mk:105: warning: overriding recipe for target '_BUILD/PASSES/GCOV/BODY'
[... cut ...]
19:06:53 package/pkg-generic.mk:227: recipe for target '/home/jenkins/workspace/Secure_container_creation_dev_mode/op-build/output/build/hostboot-9a2e268f1799061d8eedaadd4aedc7c54cc6589e/.stamp_built' failed
19:06:53 make: *** [/home/jenkins/workspace/Secure_container_creation_dev_mode/op-build/output/build/hostboot-9a2e268f1799061d8eedaadd4aedc7c54cc6589e/.stamp_built] Error 2
19:06:53 make: Leaving directory '/home/jenkins/workspace/Secure_container_creation_dev_mode/op-build/buildroot'
19:06:53 Building PNOR images for witherspoon is failedBuild step 'Execute shell' marked build as failure

dcrowell77, 2017-11-25:
It seems like the error is in the targeting/xmltohb area but I can't see the smoking gun.  Can you rerun 'op-build hostboot-rebuild' and save off all the output to a file?

pridhiviraj, 2017-11-30:
yeah, i have triggered it through jenkins which will delete the workspace once build is done. I have r-econfigured it to take the repo backup once build is finished. But most of the times it fails with below occ related errors
```09:30:05 >>> occ dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c Extracting
09:30:05 gzip -d -c /home/jenkins/workspace/buildPNOR/op-build/dl/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c.tar.gz | tar --strip-components=1 -C /home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c   -xf -
09:30:05 >>> occ dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c Patching
09:30:05 if [ "y" == "y" ]; then PATH=/home/jenkins/workspace/buildPNOR/op-build/output/host/bin:$PATH support/scripts/apply-patches.sh  /home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c /home/jenkins/workspace/buildPNOR/op-build/openpower/package/occ/p9Patches \*.patch; if [ -d /home/jenkins/workspace/buildPNOR/op-build/openpower/custom/patches/occ/p9Patches ]; then PATH=/home/jenkins/workspace/buildPNOR/op-build/output/host/bin:$PATH support/scripts/apply-patches.sh  /home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c /home/jenkins/workspace/buildPNOR/op-build/openpower/custom/patches/occ/p9Patches \*.patch; fi; fi; if [ "" == "y" ]; then PATH=/home/jenkins/workspace/buildPNOR/op-build/output/host/bin:$PATH support/scripts/apply-patches.sh  /home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c /home/jenkins/workspace/buildPNOR/op-build/openpower/package/occ/p8Patches \*.patch; if [ -d /home/jenkins/workspace/buildPNOR/op-build/openpower/custom/patches/occ/p8Patches ]; then PATH=/home/jenkins/workspace/buildPNOR/op-build/output/host/bin:$PATH support/scripts/apply-patches.sh  /home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c /home/jenkins/workspace/buildPNOR/op-build/openpower/custom/patches/occ/p8Patches \*.patch; fi; fi;
09:30:05 
09:30:05 Applying 0001-Use-gnu89-as-C-std.patch using patch: 
09:30:05 patching file src/occBootLoader/img_defs.mk
09:30:05 Reversed (or previously applied) patch detected!  Skipping patch.
09:30:05 1 out of 1 hunk ignored -- saving rejects to file src/occBootLoader/img_defs.mk.rej
09:30:05 patching file src/occ_405/img_defs.mk
09:30:05 Reversed (or previously applied) patch detected!  Skipping patch.
09:30:05 1 out of 1 hunk ignored -- saving rejects to file src/occ_405/img_defs.mk.rej
09:30:05 mkdir -p "/home/jenkins/workspace/buildPNOR/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version";
09:30:05 # Remove patch file to start off fresh
09:30:05 if [ -f /home/jenkins/workspace/buildPNOR/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/occ.patch.txt ]; then rm -rf /home/jenkins/workspace/buildPNOR/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/occ.patch.txt; fi
09:30:05 # Check all global patch directories
09:30:05 if ls ""/occ/*.patch 2>/dev/null; then sha512sum ""/occ/*.patch | sha512sum | xargs echo >> /home/jenkins/workspace/buildPNOR/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/occ.tmp_patch.txt; fi; if ls "/home/jenkins/workspace/buildPNOR/op-build/openpower/custom/patches"/occ/*.patch 2>/dev/null; then sha512sum "/home/jenkins/workspace/buildPNOR/op-build/openpower/custom/patches"/occ/*.patch | sha512sum | xargs echo >> /home/jenkins/workspace/buildPNOR/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/occ.tmp_patch.txt; fi;
09:30:05 # Check the package patch dir, /home/jenkins/workspace/buildPNOR/op-build/openpower/package/occ/ doesn't exist when running the version rules
09:30:05 if [ -n "/home/jenkins/workspace/buildPNOR/op-build/openpower/package/occ/" ]; then if ls /home/jenkins/workspace/buildPNOR/op-build/openpower/package/occ/*.patch 2>/dev/null; then sha512sum /home/jenkins/workspace/buildPNOR/op-build/openpower/package/occ/*.patch | sha512sum | xargs echo >> /home/jenkins/workspace/buildPNOR/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/occ.tmp_patch.txt; fi; else if ls /home/jenkins/workspace/buildPNOR/op-build/openpower/package/occ/*.patch 2>/dev/null; then sha512sum /home/jenkins/workspace/buildPNOR/op-build/openpower/package/occ/*.patch | sha512sum | xargs echo >> /home/jenkins/workspace/buildPNOR/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/occ.tmp_patch.txt; fi; fi
09:30:05 # If this is for linux, also check openpower/linux
09:30:05 if [ "LINUX" == "OCC" ]; then if ls /home/jenkins/workspace/buildPNOR/op-build/openpower/occ/*.patch 2>/dev/null; then sha512sum /home/jenkins/workspace/buildPNOR/op-build/openpower/occ/*.patch | sha512sum | xargs echo >> /home/jenkins/workspace/buildPNOR/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/occ.tmp_patch.txt; fi; fi;
09:30:05 # Combine all the patches found in the package and global package directories
09:30:05 if [ -f /home/jenkins/workspace/buildPNOR/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/occ.tmp_patch.txt ]; then cat /home/jenkins/workspace/buildPNOR/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/occ.tmp_patch.txt | sha512sum | cut -c 1-7 | xargs echo -n > /home/jenkins/workspace/ [... cut ...]
09:30:07 make[2]: Leaving directory '/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/src/occBootLoader'
09:30:07 Makefile:155: recipe for target '/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occBootLoader/bootloader.out' failed
09:30:07 make[1]: *** [/home/jenkins/workspace/buildPNOR/op-build/output/build/occ-dbb4d7e88cf3a4f83d4b67b6ff90cb222503c60c/obj/occBootLoader/bootloader.out] Error 2
09:30:07 make[1]: *** Waiting for unfinished jobs....
```

pridhiviraj, 2017-11-30:
[op_build_log_occ_errors_witherspoon.txt](https://github.com/open-power/op-build/files/1517604/op_build_log_occ_errors_witherspoon.txt)
Can some one have a look at these failure errors related to OCC, build is failing for witherspoon system with upstream commit 64187e71709c5f468ed5fc0301b9d495dee8d961.

dcrowell77, 2017-12-01:
Can you describe what you are doing?  What is unique about these build attempts?

pridhiviraj, 2017-12-08:
@dcrowell77 I am getting heavily these OCC related failures when building it in Power arch. But after again if i re-trigger in the same state it works fine. But that is not a good thing to do always,

dcrowell77, 2017-12-08:
I can't actually see what is failing in what you have above so I've got no guesses.  The errors messages are all the generic ones that show up at the bottom, not the real failure.  If these are the only error messages then I am really confused.  In that case it seems like some sort of fundamental OS/environment issue.

ngruend, 2017-12-08:
@pridhiviraj  You are building on a Power based machine?  What processor/stepping are you using?
I've found this issue when building on P9 based Romulus.  The push (@INC, "../../xmltohb"); line is not getting picked up for some reason.  I have to manually changed the scripts "create_ekb_targattr.pl" "handle_duplicate.pl"  and "handle_fapi_attr_mapping.pl" to use this instead.
push (@INC, "/home/test/op-build/output/build/hostboot-665c9a18838f155ae32c74c829fd4605bfd8d30a/src/usr/targeting/xmltohb");
Also, if you are building on anything besides DD2.1, I've had to add powersave=off to the kernel command line.  Otherwise, it seems to be random errors with gcc.

pridhiviraj, 2017-12-11:
@ngruend I am building it on a P8 based system, and it is a P8E DD2.1 revision. and gcc level `gcc version 5.1.1 20150618 (Red Hat 5.1.1-4) (GCC)`. you are right these errors looks like some random errors. @dcrowell77 I am also confused, the real errors not able to identify from those failures.