open-power/op-build #1697: PNOR build for witherspoon failed in host-squashfs build step.

pridhiviraj, 2017-12-11:
```
op-build openpower-pnor-rebuild
make: Entering directory '/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/buildroot'

rm -f /home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/openpower-pnor-4c844a5ef7efc6605c4c877b6c92cf0a1a0df0b5/.stamp_staging_installed
rm -f /home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/openpower-pnor-4c844a5ef7efc6605c4c877b6c92cf0a1a0df0b5/.stamp_target_installed
rm -f /home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/openpower-pnor-4c844a5ef7efc6605c4c877b6c92cf0a1a0df0b5/.stamp_images_installed
rm -f /home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/openpower-pnor-4c844a5ef7efc6605c4c877b6c92cf0a1a0df0b5/.stamp_host_installed
rm -f /home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/openpower-pnor-4c844a5ef7efc6605c4c877b6c92cf0a1a0df0b5/.stamp_built
/usr/bin/make -j1 O=/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" silentoldconfig
  GEN     /home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/Makefile
>>> host-squashfs 3de1687d7432ea9b302c2db9521996f506c140a3 Building
PATH="/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/bin:/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/var/perf/pm/bin:/root/bin" PKG_CONFIG="/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/lib/pkgconfig:/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/share/pkgconfig" /usr/bin/make -j161 CC="/usr/bin/gcc" EXTRA_CFLAGS="-O2 -I/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/include" EXTRA_LDFLAGS="-L/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/lib -Wl,-rpath,/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/lib" XATTR_SUPPORT=1 XZ_SUPPORT=1 GZIP_SUPPORT=1 LZ4_SUPPORT=1 LZO_SUPPORT=1 LZMA_XZ_SUPPORT=1 -C /home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/host-squashfs-3de1687d7432ea9b302c2db9521996f506c140a3/squashfs-tools/
/usr/bin/gcc  -L/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/lib -Wl,-rpath,/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/lib mksquashfs.o read_fs.o action.o swap.o pseudo.o compressor.o sort.o progressbar.o read_file.o info.o restore.o process_fragments.o caches-queues-lists.o gzip_wrapper.o lzma_xz_wrapper.o xz_wrapper.o lzo_wrapper.o lz4_wrapper.o xattr.o read_xattrs.o -lpthread -lm -lz -llzma -llzma  -llzo2 -llz4 -o mksquashfs
/usr/bin/gcc  -L/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/lib -Wl,-rpath,/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/lib unsquashfs.o unsquash-1.o unsquash-2.o unsquash-3.o unsquash-4.o swap.o compressor.o unsquashfs_info.o gzip_wrapper.o lzma_xz_wrapper.o xz_wrapper.o lzo_wrapper.o lz4_wrapper.o read_xattrs.o unsquashfs_xattr.o -lpthread -lm -lz -llzma -llzma  -llzo2 -llz4 -o unsquashfs
/usr/bin/ld: cannot find -llzo2
collect2: error: ld returned 1 exit status
/usr/bin/ld: cannot find -llzo2
collect2: error: ld returned 1 exit status
Makefile:279: recipe for target 'unsquashfs' failed
make[2]: *** [unsquashfs] Error 1
make[2]: *** Waiting for unfinished jobs....
Makefile:228: recipe for target 'mksquashfs' failed
make[2]: *** [mksquashfs] Error 1
package/pkg-generic.mk:234: recipe for target '/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/host-squashfs-3de1687d7432ea9b302c2db9521996f506c140a3/.stamp_built' failed
make[1]: *** [/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/host-squashfs-3de1687d7432ea9b302c2db9521996f506c140a3/.stamp_built] Error 2
Makefile:79: recipe for target '_all' failed
make: *** [_all] Error 2
make: Leaving directory '/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/buildroot'
```

pridhiviraj, 2017-12-11:
seen multiple times this issue on Power arch.

pridhiviraj, 2017-12-11:
Looks like some packages missing(Need to document those in README), After installing those getting another error which i am not much have an idea on it.
```
op-build host-squashfs-rebuild
make: Entering directory '/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/buildroot'
rm -f /home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/host-squashfs-3de1687d7432ea9b302c2db9521996f506c140a3/.stamp_staging_installed
rm -f /home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/host-squashfs-3de1687d7432ea9b302c2db9521996f506c140a3/.stamp_target_installed
rm -f /home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/host-squashfs-3de1687d7432ea9b302c2db9521996f506c140a3/.stamp_images_installed
rm -f /home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/host-squashfs-3de1687d7432ea9b302c2db9521996f506c140a3/.stamp_host_installed
rm -f /home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/host-squashfs-3de1687d7432ea9b302c2db9521996f506c140a3/.stamp_built
>>> host-squashfs 3de1687d7432ea9b302c2db9521996f506c140a3 Building
PATH="/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/bin:/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/var/perf/pm/bin:/root/bin" PKG_CONFIG="/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/lib/pkgconfig:/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/share/pkgconfig" /usr/bin/make -j161 CC="/usr/bin/gcc" EXTRA_CFLAGS="-O2 -I/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/include" EXTRA_LDFLAGS="-L/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/lib -Wl,-rpath,/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/lib" XATTR_SUPPORT=1 XZ_SUPPORT=1 GZIP_SUPPORT=1 LZ4_SUPPORT=1 LZO_SUPPORT=1 LZMA_XZ_SUPPORT=1 -C /home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/host-squashfs-3de1687d7432ea9b302c2db9521996f506c140a3/squashfs-tools/
/usr/bin/gcc  -L/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/lib -Wl,-rpath,/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/lib mksquashfs.o read_fs.o action.o swap.o pseudo.o compressor.o sort.o progressbar.o read_file.o info.o restore.o process_fragments.o caches-queues-lists.o gzip_wrapper.o lzma_xz_wrapper.o xz_wrapper.o lzo_wrapper.o lz4_wrapper.o xattr.o read_xattrs.o -lpthread -lm -lz -llzma -llzma  -llzo2 -llz4 -o mksquashfs
/usr/bin/gcc  -L/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/lib -Wl,-rpath,/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/lib unsquashfs.o unsquash-1.o unsquash-2.o unsquash-3.o unsquash-4.o swap.o compressor.o unsquashfs_info.o gzip_wrapper.o lzma_xz_wrapper.o xz_wrapper.o lzo_wrapper.o lz4_wrapper.o read_xattrs.o unsquashfs_xattr.o -lpthread -lm -lz -llzma -llzma  -llzo2 -llz4 -o unsquashfs
>>> host-squashfs 3de1687d7432ea9b302c2db9521996f506c140a3 Installing to host directory
PATH="/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/bin:/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/var/perf/pm/bin:/root/bin" PKG_CONFIG="/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/lib/pkgconfig:/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/share/pkgconfig" /usr/bin/make -j161 XATTR_SUPPORT=1 XZ_SUPPORT=1 GZIP_SUPPORT=1 LZ4_SUPPORT=1 LZO_SUPPORT=1 LZMA_XZ_SUPPORT=1 -C /home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/host-squashfs-3de1687d7432ea9b302c2db9521996f506c140a3/squashfs-tools/ INSTALL_DIR=/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/bin install
mkdir -p /home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/host/bin
[... cut ...]
package/pkg-generic.mk:244: recipe for target '/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/host-squashfs-3de1687d7432ea9b302c2db9521996f506c140a3/.stamp_host_installed' failed
make[1]: *** [/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/output/build/host-squashfs-3de1687d7432ea9b302c2db9521996f506c140a3/.stamp_host_installed] Error 1
Makefile:79: recipe for target '_all' failed
make: *** [_all] Error 2
make: Leaving directory '/home/jenkins_data/crtl/repo/OpenPower/PNOR/failed_opbuilds/pnor_witherspoon_20171211_1359/op-build/buildroot'
```

pridhiviraj, 2017-12-11:
@stewart-ibm We need to add these packages into README  ``` lz4, squashfs-tools, lzo-devel, lzo-minilzo, python-lzo ```(These are the names for Power arch/fedora i got it). And who can help us for debugging above squashfs related failures here?

hellerda, 2017-12-11:
See my comment in pull request #1324 for an explanation of "installs executables without proper RPATH" error.

hellerda, 2017-12-11:
I never hit this problem, but on my system, the lzo libraries are installed on both my host system, and as a "host" package under op-build.  I did not check to see which one it (the squashfs package) is using, but my guess is, since it's built as a "host" package, it want's to use the one under op-build:

```
[build@mybox op-build]$ find /usr/lib64/ -name liblzo2.so\* -exec ls -l {} \;
lrwxrwxrwx. 1 root root 16 Feb 11  2017 /usr/lib64/liblzo2.so.2 -> liblzo2.so.2.0.0
-rwxr-xr-x. 1 root root 138800 Feb 11  2017 /usr/lib64/liblzo2.so.2.0.0
[build@ltc254 op-build]$ find . -name liblzo2.so -exec ls -l {} \;
lrwxrwxrwx. 1 build build 16 Oct 26 15:36 ./output/build/host-lzo-2.09/src/.libs/liblzo2.so -> liblzo2.so.2.0.0
lrwxrwxrwx. 1 build build 16 Oct 26 15:36 ./output/host/lib/liblzo2.so -> liblzo2.so.2.0.0
```
So, to fix your original problem, did you install lzo to your host system? (i.e. your build machine)?  Or did you build it under op-build?  What happens if you `op-build host-lzo-rebuild` ?

pridhiviraj, 2017-12-12:
@hellerda Installed all the required packages still seeing the same RPATH error in the same cloned repo. In another fresh attempt of clone and build got success,

pridhiviraj, 2017-12-12:
@hellerda I saw your comment in #1324 . So for that any changes need to be done?

stewart-ibm, 2018-02-28:
This looks like it's been resolved in the new PRs that we ended up merging?

I'll close it for now, and re-open if we see it as an issue

closed 2018-02-28 by stewart-ibm