open-power/op-build #1711: PNOR built from master failed to boot on P8 habanero system.

pridhiviraj, 2017-12-14:
```
  3.15283|Ignoring boot flags, incorrect version 0x0
  3.17108|ECC error in PNOR flash in section offset 0x01CB1000

  3.17111|System shutting down with error status 0x60F
  3.17116|System shutting down with error status 0x90000002
```
op-build commit id: `e74b3907e720a02cb7a23693bd338efedce54ff8`

pridhiviraj, 2017-12-14:
This looks related to permanent attr override seciton.
```
    <section>
        <description>Permanent Attribute Override (32K)</description>
        <eyeCatch>ATTR_PERM</eyeCatch>
        <physicalOffset>0x1CB1000</physicalOffset>
        <physicalRegionSize>0x8000</physicalRegionSize>
        <side>sideless</side>
        <ecc/>
        <reprovision/>
</section>```

sammj, 2017-12-15:
Confirmed on `4f51ffeb50f97a1ee0fd8c3f0a9049225c7d8881` via flashing the whole PNOR.
Looks like the ATTR_PERM partition is has just been set to all ones rather than being ECC formatted correctly:

```
➜  pflash git:(master_openpower) ./pflash -F ~/Downloads/habanero.pnor -r /tmp/attr -P ATTR_PERM -S 0
Reading to "/tmp/attr" from 0x01cb1000..0x01cb9000 !
[==================================================] 100%
➜  pflash git:(master_openpower) hexdump -C -n 256 /tmp/attr 
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000100
```

sammj, 2017-12-15:
I reckon this patch has broken it: https://github.com/open-power/pnor/commit/dccb9d427b9fb95b08af814218d06d9bee5392e6
They don't build the ATTR_PERM partition as a workaround, but since that skips ECC-correcting the partition that is _going_ to exist, it won't boot. This presumably breaks booting on all P8 platforms.

stewart-ibm, 2017-12-15:
Oh, look, this was actually caught by CI but the CI scripts are so broken they don't class not booting as a failure:
```
spawn ipmitool -H 9.3.29.140 -I lanplus -U ADMIN -P admin sol activate
[SOL Session operational.  Use ~? for help]
  3.15788|Ignoring boot flags, incorrect version 0x0
  3.17619|ECC error in PNOR flash in section offset 0x01CB1000

  3.17623|System shutting down with error status 0x60F
  3.17628|System shutting down with error status 0x90000002

sol_logger.exp: EOF detected

```

dcrowell77, 2017-12-15:
Sigh... the joy of not branching every repo, only some of them... I agree the commit above is the culprit.

pridhiviraj, 2017-12-20:
@stewart-ibm Is this fixed or not? I still see the issue with latest PNOR built from master.

stewart-ibm, 2018-01-10:
@pridhiviraj not yet. Nagendra is looking at it.

mzipse, 2018-01-11:
@nagendra, this is blocking us from tagging any OPBuild snapshots for the P8 platforms.  Can you expedite this?

nagendraIBM, 2018-01-12:
https://github.com/open-power/pnor/pull/87

closed 2018-01-15 by stewart-ibm