open-power/op-build #1729: Hostboot memd_creation.pl tool leaves dirty git tree

sammj, 2017-12-20:
https://github.com/open-power/op-build/blob/master/openpower/package/machine-xml/machine-xml.mk#L112 runs the `memd_creation.pl` tool from Hostboot. This script creates a temporary file as part creating the MEMD binary, but due to how it is called the file is created in the buildroot submodule directory and not cleaned up. This results in a dirty tree and gross version strings, for example:

```
IBM-witherspoon-ibm-OP9_v1.19_1.99
        op-build-v1.19-293-g418d38f-dirty
        buildroot-2017.08-8-g5e23247
        skiboot-v5.9.6
        hostboot-b4a89ac
        linux-4.13.16-openpower1-pd22e9b4
        petitboot-v1.6.3-pf9ef270
        machine-xml-592a6dd
        occ-c62d875
        hostboot-binaries-c5203ed
        capp-ucode-p9-dd2-v2
        sbe-b8e05a5
```

The below patch is probably all we need:
```
diff --git a/src/build/buildpnor/memd_creation.pl b/src/build/buildpnor/memd_creation.pl
index 419d63569..84c87d599 100755
--- a/src/build/buildpnor/memd_creation.pl
+++ b/src/build/buildpnor/memd_creation.pl
@@ -143,6 +143,7 @@ foreach my $file (@memd_files)
     }
     close $in_file;
     $offset = $offset + ($max_file_size * ROUNDING_DIVISOR);
+    unlink $new_file;
 }
```
But I don't believe I have access to the Hostboot gerrit, so someone can just commit that themselves if they like.

dcrowell77, 2017-12-21:
I don't think that change will leave a functional version of the code.  The script is trying to create files that will be consumed by later steps.  If we unlink the file at the end of the script then those files aren't there anymore, right?

@cvswen @e-liner - Please take a look while I'm out if you're around

sammj, 2017-12-21:
Based on the existing code I think `$new_file` is only used here and written into a different file:
```
...
      # Removing the first byte of the MEMD binary
      my $new_file = File::Basename::basename "$file.tmp";
      run_command("tail -c +2 $file > $new_file");
  
      seek($fh, $offset, 0);
      print "Writing the file $new_file...\n";
      open(my $in_file, '<', $new_file) or die "Could not open file '$new_file' $!";
  
      while ( <$in_file> )
      {
          print $fh $_;
      }
      close $in_file;
      $offset = $offset + ($max_file_size * ROUNDING_DIVISOR);
 }
  
  
  close $fh;
  print "Done. Memd binary is $memd_output.\n";

```

e-liner, 2018-01-09:
I tried out the change, and everything worked fine.  I'll bring them into hostboot today.

stewart-ibm, 2018-01-18:
@e-liner have these gone in to hostboot yet?

e-liner, 2018-02-09:
We pulled these into hostboot as of yesterday.  Here's today's pull request with the change: https://github.com/open-power/op-build/pull/1852

e-liner, 2018-02-09:
Here's the commit: 7136fd9 - Elizabeth Liner - 2018-02-08 - Cleaning up git tree from memd_creation.pl

closed 2018-02-09 by e-liner