open-power/op-build #1822: Barreleye fails to build on powerpc64le Ubuntu 16.04.3

cyrilbur-ibm, 2018-02-01:
Steps to reproduce:
```
. op-build-env
op-build clean
op-build barreleye_defconfig
op-build
```
Information:
```
cat /etc/os-release 
NAME="Ubuntu"
VERSION="16.04.3 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.3 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial
```

```
commit 7a8ea59358a5db6df72709e4e528447a8013c78d
Merge: ef03903 4994a2c
Author: William Hoffa <wihoffa@gmail.com>
Date:   Fri Jan 26 07:44:36 2018 -0600

    Merge pull request #1806 from op-jenkins/op-build-update_181_1-25-2018
    
    op-build update 1-25-2018
```

Failing component:
```
WARNING: no hash file for openpower-mrw-0729a4a68c0e731a8d3720f1a5aee190d165400b.tar.gz
>>> openpower-mrw 0729a4a68c0e731a8d3720f1a5aee190d165400b Extracting
gzip -d -c /home/cyrilb/ffspart-op-build/dl/openpower-mrw-0729a4a68c0e731a8d3720f1a5aee190d165400b.tar.gz | tar --strip-components=1 -C /home/cyrilb/ffspart-op-build/output/build/openpower-mrw-0729a4a68c0e731a8d3720f1a5aee190d165400b   -xf -
>>> openpower-mrw 0729a4a68c0e731a8d3720f1a5aee190d165400b Patching
>>> openpower-mrw 0729a4a68c0e731a8d3720f1a5aee190d165400b Configuring
>>> openpower-mrw 0729a4a68c0e731a8d3720f1a5aee190d165400b Building
INSTALL_DIRECTORY=/home/cyrilb/ffspart-op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/usr/bin/ bash -c 'cd /home/cyrilb/ffspart-op-build/output/build/openpower-mrw-0729a4a68c0e731a8d3720f1a5aee190d165400b && /usr/bin/make -j161'
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER xmlutil.C -o xmlutil.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwTargetParser.C -o mrwTargetParser.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwParserCommon.C -o mrwParserCommon.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwMemCommon.C -o mrwMemCommon.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwPCIEParser.C -o mrwPCIEParser.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwPCIECommon.C -o mrwPCIECommon.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwPowerBusParser.C -o mrwPowerBusParser.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwPowerBusCommon.C -o mrwPowerBusCommon.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwFSICommon.C -o mrwFSICommon.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwI2CCommon.C -o mrwI2CCommon.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwPowerCommon.C -o mrwPowerCommon.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwLocationCodeParser.C -o mrwLocationCodeParser.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwDMICommon.C -o mrwDMICommon.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwMergeElements.C -o mrwMergeElements.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwChipIDs.C -o mrwChipIDs.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwCecChips.C -o mrwCecChips.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwFSIParser.C -o mrwFSIParser.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwMemParser.C -o mrwMemParser.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwI2CParser.C -o mrwI2CParser.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwDMIParser.C -o mrwDMIParser.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwPower.C -o mrwPower.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwMruIdParser.C -o mrwMruIdParser.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwProcSpiParser.C -o mrwProcSpiParser.o
g++ -c -g -I. -I/usr/include/libxml2/ -DMRW_OPENPOWER mrwProcSpiCommon.C -o mrwProcSpiCommon.o
In file included from xmlutil.C:34:0:
./xmlutil.H:43:27: fatal error: libxml/parser.h: No such file or directory
compilation terminated.
[... 119 lines cut ...]
make[2]: *** [mrwDMIParser.o] Error 1
makefile:96: recipe for target 'mrwPower.o' failed
make[2]: *** [mrwPower.o] Error 1
makefile:96: recipe for target 'mrwMruIdParser.o' failed
make[2]: *** [mrwMruIdParser.o] Error 1
makefile:96: recipe for target 'mrwPCIEParser.o' failed
make[2]: *** [mrwPCIEParser.o] Error 1
makefile:96: recipe for target 'mrwProcSpiParser.o' failed
make[2]: *** [mrwProcSpiParser.o] Error 1
makefile:96: recipe for target 'mrwI2CCommon.o' failed
make[2]: *** [mrwI2CCommon.o] Error 1
makefile:96: recipe for target 'mrwParserCommon.o' failed
make[2]: *** [mrwParserCommon.o] Error 1
makefile:96: recipe for target 'mrwProcSpiCommon.o' failed
make[2]: *** [mrwProcSpiCommon.o] Error 1
makefile:96: recipe for target 'mrwChipIDs.o' failed
make[2]: *** [mrwChipIDs.o] Error 1
makefile:96: recipe for target 'mrwMemParser.o' failed
make[2]: *** [mrwMemParser.o] Error 1
package/pkg-generic.mk:234: recipe for target '/home/cyrilb/ffspart-op-build/output/build/openpower-mrw-0729a4a68c0e731a8d3720f1a5aee190d165400b/.stamp_built' failed
make[1]: *** [/home/cyrilb/ffspart-op-build/output/build/openpower-mrw-0729a4a68c0e731a8d3720f1a5aee190d165400b/.stamp_built] Error 2
Makefile:79: recipe for target '_all' failed
make: *** [_all] Error 2
make: Leaving directory '/home/cyrilb/ffspart-op-build/buildroot'
```

cyrilbur-ibm, 2018-02-01:
Has also happened for `firestone_defconfig`

stewart-ibm, 2018-03-06:
It looks like you're missing `libxml-dev` ?

(Do we have this documented as a dependency somewhere?)