open-power/op-build #1839: PNOR failed to boot on p9dsu platform after building upstream op-build with p9dsu defconfig.

pridhiviraj, 2018-02-06:
```
../pUpdate -h $BMCIP -i lan -u ADMIN -p ADMIN -pnor /var/www/html/sath/pridhivi/smc_opbuild_images/images/p9dsu.pnor 
failed to get version string.
Failed to find partition [VERSION].
Version info are not available.
If the PNOR update fails, PLEASE TRY AGAIN.
Transfer data ................
validating PNOR image ................
Image validation failed.
Invalid PNOR image!
ERROR: failed to update PNOR. [0xffffffff]
```

hellerda, 2018-02-06:
Hmm, looks like it failed to flash, is that really the subject of this bug?  Can you flash any image built via op-build?   Do other defconfigs give you a problem, on their respective machines?  Or is it just the p9dsu defconfig that is a problem?

pridhiviraj, 2018-02-06:
@hellerda Yeah, thanks. I changed the subject now. The problem is with flashing itself. Other platform PNOR's are not having any problems. Only facing problem with p9dsu defconfig.

pridhiviraj, 2018-02-06:
@hellerda Using pflash tool works fine and system boots fine to petitboot.

hellerda, 2018-04-06:
Is this still a problem?  Was this an issue only for development and test, and using pflash is a workaround?
Is this an OpenBMC problem?  If so, is it related to development version vs. production version of BMC image?

hellerda, 2018-04-08:
Previously with briggs-stratton I had a similar problem, which was due to a small delta in file size between production PNOR and PNOR I built myself.  I found I could make it work by stripping the trailer from a production image and appending it to mine.  That only worked for the development level BMC firmware, not the production BMC firmware.  But I think it means there is some additional step for our PNOR, to make it flashable by the BMC.  (I think I mentioned this before but I don't recall if it was this issue.)

pridhiviraj, 2018-04-17:
yeah, it is the same issue we discussed earlier.