open-power/op-build #1848: op-build failed to build signed PNOR for p9dsu platform in developement mode

pridhiviraj, 2018-02-08:
After enabling secureboot in hostboot defconfig, op-build failed to build and crashed in create-container step.
```
--> crtSignedContainer.sh: Signing mode: local
--> crtSignedContainer.sh: Using existing cache dir: /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//SIGNTOOL_1518078849, created: Thu Feb  8 14:04:09 IST 2018
--> crtSignedContainer.sh: Creating new cache subdir: /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//SIGNTOOL_1518078849/WOFDATA
--> crtSignedContainer.sh: Generating signing requests...
*** Error in `create-container': double free or corruption (!prev): 0x000001003dc52030 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x85b74)[0x3fff80a15b74]
/lib64/libc.so.6(cfree+0x164)[0x3fff80a25f44]
create-container[0x10003ed0]
create-container[0x10001f6c]
/lib64/libc.so.6(+0x22ebc)[0x3fff809b2ebc]
/lib64/libc.so.6(__libc_start_main+0xb8)[0x3fff809b30b8]
======= Memory map: ========
10000000-10010000 r-xp 00000000 fd:05 2348839682                         /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/bin/create-container
10010000-10020000 r--p 00000000 fd:05 2348839682                         /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/bin/create-container
10020000-10030000 rw-p 00010000 fd:05 2348839682                         /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/bin/create-container
1003dc50000-1003dc80000 rw-p 00000000 00:00 0                            [heap]
3fff806e0000-3fff80960000 r--p 00000000 fd:05 2350802052                 /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch/WOFDATA.staged
3fff80960000-3fff80970000 r-xp 00000000 fd:00 262782                     /usr/lib64/libdl-2.23.so
3fff80970000-3fff80980000 r--p 00000000 fd:00 262782                     /usr/lib64/libdl-2.23.so
3fff80980000-3fff80990000 rw-p 00000000 00:00 0 
3fff80990000-3fff80b60000 r-xp 00000000 fd:00 262776                     /usr/lib64/libc-2.23.so
3fff80b60000-3fff80b70000 r--p 001c0000 fd:00 262776                     /usr/lib64/libc-2.23.so
3fff80b70000-3fff80b80000 rw-p 001d0000 fd:00 262776                     /usr/lib64/libc-2.23.so
3fff80b80000-3fff80b90000 rw-p 00000000 00:00 0 
3fff80b90000-3fff80d80000 r-xp 00000000 fd:05 1092775041                 /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/lib/libcrypto.so.1.0.0
3fff80d80000-3fff80da0000 r--p 001e0000 fd:05 1092775041                 /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/lib/libcrypto.so.1.0.0
3fff80da0000-3fff80db0000 rw-p 00200000 fd:05 1092775041                 /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/lib/libcrypto.so.1.0.0
3fff80db0000-3fff80e20000 r-xp 00000000 fd:05 1092775044                 /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/lib/libssl.so.1.0.0
3fff80e20000-3fff80e30000 r--p 00060000 fd:05 1092775044                 /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/lib/libssl.so.1.0.0
3fff80e30000-3fff80e40000 rw-p 00070000 fd:05 1092775044                 /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/lib/libssl.so.1.0.0
3fff80e40000-3fff80e60000 r-xp 00000000 00:00 0                          [vdso]
3fff80e60000-3fff80ea0000 r-xp 00000000 fd:00 262769                     /usr/lib64/ld-2.23.so
3fff80ea0000-3fff80eb0000 r--p 00030000 fd:00 262769                     /usr/lib64/ld-2.23.so
3fff80eb0000-3fff80ec0000 rw-p 00040000 fd:00 262769                     /usr/lib64/ld-2.23.so
3fffcc650000-3fffcc680000 rw-p 00000000 00:00 0                          [stack]
/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/usr/bin//crtSignedContainer.sh: line 626: 150534 Aborted                 (core dumped) create-container $HW_KEY_ARGS $SW_KEY_ARGS --payload "$PAYLOAD" --imagefile "$OUTPUT" --dumpPrefixHdr "$T/prefix_hdr" --dumpSwHdr "$T/software_hdr" $DEBUG_ARGS $ADDL_ARGS
crtSignedContainer.sh: Call to create-container failed with error: 134
Error running command: /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/usr/bin//crtSignedContainer.sh --scratchDir /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch/  --hwPrivKeyA /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/etc/keys//hw_key_a.key --hwPrivKeyB /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/etc/keys//hw_key_b.key --hwPrivKeyC /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/etc/keys//hw_key_c.key --swPrivKeyP /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/etc/keys//sw_key_a.key  --flags 0x80000000 --sign-project-FW-token WOFDATA  --protectedPayload /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//WOFDATA.staged --out /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//rand-218306800.WOFDATA.temp.hdr.bin. Nonzero return code of (256) returned.
Error running command: cd /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch/ && /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//genPnorImages.pl --binDir /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch/ --systemBinFiles HBRT=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HBRT.staged,GUARD=EMPTY,BOOTKERNEL=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//BOOTKERNEL.staged,SBKT=EMPTY,FIRDATA=EMPTY,RINGOVD=EMPTY,WOFDATA=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//WOFDATA.staged,VERSION=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//VERSION.staged,ATTR_TMP=EMPTY,OCC=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//OCC.staged,PAYLOAD=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//PAYLOAD.staged,ATTR_PERM=EMPTY,MVPD=EMPTY,SBE=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//SBE.staged,MEMD=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//MEMD.staged,HDAT=EMPTY,HBB=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HBB.staged,CAPP=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//CAPP.staged,DJVPD=EMPTY,SECBOOT=EMPTY,HBI=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HBI.staged,NVRAM=EMPTY,IMA_CATALOG=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//IMA_CATALOG.staged,CVPD=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//CVPD.staged,HBBL=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HBBL.staged,HBD=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/open [... cut ...]
make[1]: *** [/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/openpower-pnor-de0f8e47de145df11a18d065742c7d583307f307/.stamp_images_installed] Error 2
Makefile:79: recipe for target '_all' failed
make: *** [_all] Error 2
make: Leaving directory '/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/buildroot'
```

Steps to reproduce:
1. Clone upstream op-build with commit c69c2a73b0451b8dfa27adf036e89776d665d5c1
2. Enable secureboot in p9dsu.defconfig by commenting these three lines.
#unset SECUREBOOT
#unset TPMDD
#unset TPM_NUVOTON

3. Build the PNOR.

stewart-ibm, 2018-02-09:
Paidipeddi Pridhiviraj <notifications@github.com> writes:
> After enabling secureboot in hostboot defconfig, op-build failed to build and crashed in create-container step.
> ```
> --> crtSignedContainer.sh: Signing mode: local
> --> crtSignedContainer.sh: Using [...]

If with op-build with sb-signing-utils v0.4, we think it was a
regression introduced there.

It should be fixed by the patch I just pushed that reverts us back to
0.3 for the timebeing while @hellerda takse a look.

-- 
Stewart Smith
OPAL Architect, IBM.

pridhiviraj, 2018-02-09:
@stewart-ibm Yes, the failure occurs with v0.4. I saw the revert patch. @hellerda Can you have a look at this failure, we are hinting when op-build has v0.4 SB util's in it.

hellerda, 2018-04-06:
This is now fixed with the following commit, so if no more issues, we should be able to close this now.  Thx -Dave
https://github.com/open-power/sb-signing-utils/commit/bfa902bfe26bc5bb4b1e12f6baaca80397f64730

pridhiviraj, 2018-04-06:
@hellerda Yes, we can close this, as pretty much we are able to build PNOR successfully in development mode now, Thanks. Closing it.

closed 2018-04-06 by pridhiviraj