open-power/op-build #1849: VERSION partition is not signed due to which seceure boot enforcement is happening during boot.

pridhiviraj, 2018-02-08:
```
/ # cat /sys/firmware/opal/msglog | grep -i STB
[   65.422464780,6] MEM: parsing reserved memory from node /ibm,hostboot/reserved-memory
[   69.220411069,7] STB: Secureboot not yet initialized, initializing now...
[   69.220415299,5] STB: Found ibm,secureboot-v2
[   69.231123306,5] STB: secure mode off
[   69.233254531,6] STB: Found CVC @ 203ffd260000-203ffd26ffff
[   69.233256579,6] STB: Found CVC-sha512 @ 203ffd260040, version=1
[   69.233258373,6] STB: Found CVC-verify @ 203ffd260050, version=1
[   69.233260363,0] STB: VERSION verification FAILED. log=0xffffffffffff8120
[   69.350953563,5] STB: trusted mode off
[   69.487967689,5] STB: IMA_CATALOG verified
[   70.119872468,5] STB: CAPP verified
[   77.265263634,5] STB: BOOTKERNEL verified
```
From the op-build build logs i have observed that VERSION partition is not getting signed, which can lead to boot abort during OPAL boot time.,

pridhiviraj, 2018-02-08:
```
[root@ltc-jenkins op-build]# ls /home/jenkins/workspace/Secure_container_creation_dev_mode/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch/SIGNTOOL_1518088387/
BOOTKERN  CAPP  HBB  HBBL  HBD  HBI  HBRT  HCODE  HDAT  IMA_CATA  MEMD  OCC  PAYLOAD  SBE  WOFDATA
```
These are the partitions we are signing in op-build, and VERSION partition is missing here

dcrowell77, 2018-02-08:
I would not expect VERSION to be signed.
@bofferdn - Thoughts?

bofferdn, 2018-02-08:
It certainly isn't signed now and it's not code, so kinda a toss up.  It doesn't hurt to sign it obviously.  If you manipulated it in arbitrary fashion, would a security hole be created? That is typically the litmus test.  Even if you signed it, you could take a file from any other driver signed by the same keys and try to fool whomever is looking at it.  Secure Boot is also not designed to stop denial of service; i.e. if trashing that file prevents system boot, then Secure Boot is not necessarily protecting against that.  But, if you can manipulate the content to force OPAL or hostboot to execute arbitrary code, that would be a problem.

stewart-ibm, 2018-02-09:
I'd like us to sign it - we parse it in OPAL to put it into the device tree, and I'd prefer a big giant guard over someone trying to use it as an attack vector. String parsing in C is... well, you know :)

pridhiviraj, 2018-02-12:
@hellerda What are the changes needed for this and who can fix this?

hellerda, 2018-02-20:
@bofferdn Is this something that needs to be added to the buildpnor scripts?

bofferdn, 2018-02-22:
Yeah we'd have to add it to the build PNOR script .. but there is the limitation in the sense that someone could replace it with some previously signed blob of version from earlier releases and pretend it's something its not.  As long as nobody cares, it's certainly fine to secure it.  I'll put together a work item.

stewart-ibm, 2018-02-23:
Nick Bofferding <notifications@github.com> writes:
> Yeah we'd have to add it to the build PNOR script .. but there is the
> limitation in the sense that someone could replace it with some
> previously signed blob of version from earlier releases and [...]

Yeah, that's okay - at least it won't be parsing something unknown then.

-- 
Stewart Smith
OPAL Architect, IBM.