open-power/op-build #1888: OCCs were not loaded/started successfully with latest PNOR.

pridhiviraj, 2018-02-20:
```
--== Welcome to Hostboot hostboot-28927a7/hbicore.bin ==--

  4.04231|secure|SecureROM valid - enabling functionality
  4.05125|secure|Booting in non-secure mode.
  5.55027|Booting from SBE side 0 on master proc=00050000
  5.59912|ISTEP  6. 5 - host_init_fsi
  5.81158|ISTEP  6. 6 - host_set_ipl_parms
  5.86199|ISTEP  6. 7 - host_discover_targets
  7.63258|HWAS|PRESENT> DIMM[03]=AAAA000000000000
  7.63259|HWAS|PRESENT> Proc[05]=8800000000000000
  7.63260|HWAS|PRESENT> Core[07]=FFF0FC3FF3FC0000
  7.66154|ISTEP  6. 8 - host_update_master_tpm
 18.40266|SECURE|Security Access Bit> 0x0000000000000000
 18.40266|SECURE|Secure Mode Disable (via Jumper)> 0xC000000000000000
 18.40278|ISTEP  6. 9 - host_gard
 18.43450|HWAS|FUNCTIONAL> DIMM[03]=AAAA000000000000
 18.43451|HWAS|FUNCTIONAL> Proc[05]=8800000000000000
 18.43452|HWAS|FUNCTIONAL> Core[07]=FFF0FC3FF3FC0000
 18.44109|ISTEP  6.10 - host_revert_sbe_mcs_setup
 18.44270|ISTEP  6.11 - host_start_occ_xstop_handler
 19.29191|ISTEP  6.12 - host_voltage_config
 19.40781|ISTEP  7. 1 - mss_attr_cleanup
 19.73846|ISTEP  7. 2 - mss_volt
 19.83192|ISTEP  7. 3 - mss_freq
 22.23235|ISTEP  7. 4 - mss_eff_config
 23.38300|ISTEP  7. 5 - mss_attr_update
 23.39291|ISTEP  8. 1 - host_slave_sbe_config
 23.50267|ISTEP  8. 2 - host_setup_sbe
 23.51366|ISTEP  8. 3 - host_cbs_start
 23.58936|ISTEP  8. 4 - proc_check_slave_sbe_seeprom_complete
 30.81685|ISTEP  8. 5 - host_attnlisten_proc
 30.81784|ISTEP  8. 6 - host_p9_fbc_eff_config
 30.82399|ISTEP  8. 7 - host_p9_eff_config_links
 30.83286|ISTEP  8. 8 - proc_attr_update
 30.83541|ISTEP  8. 9 - proc_chiplet_fabric_scominit
 30.86717|ISTEP  8.10 - proc_xbus_scominit
 32.76151|ISTEP  8.11 - proc_xbus_enable_ridi
 32.76693|ISTEP  8.12 - host_set_voltages
 32.85852|ISTEP  9. 1 - fabric_erepair
 32.90811|ISTEP  9. 2 - fabric_io_dccal
 33.60786|ISTEP  9. 3 - fabric_pre_trainadv
 33.61204|ISTEP  9. 4 - fabric_io_run_training
 33.74155|ISTEP  9. 5 - fabric_post_trainadv
 33.74437|ISTEP  9. 6 - proc_smp_link_layer
 33.75145|ISTEP  9. 7 - proc_fab_iovalid
 33.77622|ISTEP  9. 8 - host_fbc_eff_config_aggregate
 33.78410|ISTEP 10. 1 - proc_build_smp
 33.92268|ISTEP 10. 2 - host_slave_sbe_update
 34.83451|ISTEP 10. 4 - proc_cen_ref_clk_enable
 34.91909|ISTEP 10. 5 - proc_enable_osclite
 34.92042|ISTEP 10. 6 - proc_chiplet_scominit
 34.97314|ISTEP 10. 7 - proc_abus_scominit
 34.98318|ISTEP 10. 8 - proc_obus_scominit
 35.05510|ISTEP 10. 9 - proc_npu_scominit
 35.09440|ISTEP 10.10 - proc_pcie_scominit
 35.14051|ISTEP 10.11 - proc_scomoverride_chiplets
 35.14251|ISTEP 10.12 - proc_chiplet_enable_ridi
 35.14536|ISTEP 10.13 - host_rng_bist
 35.14752|ISTEP 10.14 - host_update_redundant_tpm
 35.15158|ISTEP 11. 1 - host_prd_hwreconfig
 35.19741|ISTEP 11. 2 - cen_tp_chiplet_init1
 35.20130|ISTEP 11. 3 - cen_pll_initf
 35.20536|ISTEP 11. 4 - cen_pll_setup
 35.21153|ISTEP 11. 5 - cen_tp_chiplet_init2
 35.21714|ISTEP 11. 6 - cen_tp_arrayinit
 35.22146|ISTEP 11. 7 - cen_tp_chiplet_init3
 35.22547|ISTEP 11. 8 - cen_chiplet_init
 35.22938|ISTEP 11. 9 - cen_arrayinit
 35.23476|ISTEP 11.10 - cen_initf
[... 276 lines cut ...]
 62.91106|  FFDC                       : HW_IMG_SIZE
 62.91107|     00000000
 62.91110|------------------------------------------------
 62.91111|  HwpReturnCode              : RC_HW_IMAGE_INVALID_SIZE
 62.91112|  FFDC                       : MAX_HW_IMG_SIZE
 62.91112|     00100000
 62.91115|------------------------------------------------
 62.91116|  Callout type             : Procedure Callout
 62.91116|  Procedure                : EPUB_PRC_HB_CODE
 62.91117|  Priority                 : SRCI_PRIORITY_HIGH
 62.91117|------------------------------------------------
 62.91118|  Callout type             : Procedure Callout
 62.91118|  Procedure                : EPUB_PRC_LVL_SUPP
 62.91119|  Priority                 : SRCI_PRIORITY_MED
 62.91119|------------------------------------------------
 62.91120|  Callout type             : Procedure Callout
 62.91121|  Procedure                : EPUB_PRC_HB_CODE
 62.91121|  Priority                 : SRCI_PRIORITY_MED
 62.91122|------------------------------------------------
 62.91122|  Hostboot Build ID: hostboot-28927a7/hbicore.bin
 62.91123|================================================
 63.24955|ISTEP 21. 2 - host_verify_hdat
 63.23005|ISTEP 21. 3 - host_start_payload
[  103.381908445,5] OPAL skiboot-v5.10-rc3 starting...
```

pridhiviraj, 2018-02-20:
PNOR Level:
```
cat /var/lib/phosphor-software-manager/pnor/ro/VERSION 
open-power-witherspoon-v1.21-rc2-8-gcf83faa-dirty
	buildroot-2017.11-5-g65679be
	skiboot-v5.10-rc3
	hostboot-28927a7
	linux-4.14.20-openpower1-pf1c13fc
	petitboot-v1.6.6-p7cfd0fc
	machine-xml-58554bf-p1f16afe
	occ-f72f857
	hostboot-binaries-6924d6b
	capp-ucode-p9-dd2-v3
	sbe-0aae9a8
```
BMC Level:
```
cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.1"
VERSION_ID="ibm-v2.1-55-g35bc913-r1-ctm-0-g8b7a6f2"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.1"
BUILD_ID="ibm-v2.1-55-g35bc913-r1-ctm"
```

dcrowell77, 2018-02-20:
@cjcain

cjcain, 2018-02-20:
This 2A09 is a VRM VDD temperature timeout (1 sec) and the OCC will request a reset when this occurs.  This timeout has since been increased to 5 seconds in a MRW change: SW415109 (end of Jan?)

A PM Complex Reset was issued and I believe succeeded (dont have full eSEL data to know) and then a PM Complex Load/Start, but looks like the hardware procedure failed:
```
 62.09025|Error reported by hwpf (0x0900) PLID 0x9000003F
 62.09026|  HW Procedure generated error. See User Data.
 62.09026|  ModuleId   0x01 MOD_FAPI2_RC_TO_ERRL
 62.09027|  ReasonCode 0x090f RC_HWP_GENERATED_ERROR
 62.09029|  UserData1  RC value from HWP : 0x00000000000e6cc2
 62.09030|  UserData2  <unused> : 0x0000000000000000
 62.09030|------------------------------------------------
 62.09031|  HwpReturnCode              : RC_HW_IMAGE_INVALID_SIZE
 62.09032|  HWP Error description      : Hardware Image size is not in expected range
 62.09033|------------------------------------------------
 62.09033|  HwpReturnCode              : RC_HW_IMAGE_INVALID_SIZE
 62.09034|  FFDC                       : HW_IMG_SIZE
 62.09035|     00000000
 62.09038|------------------------------------------------
 62.09039|  HwpReturnCode              : RC_HW_IMAGE_INVALID_SIZE
 62.09039|  FFDC                       : MAX_HW_IMG_SIZE
 62.09040|     00100000
 62.09043|------------------------------------------------
```

vanderp, 2018-02-20:
This is caused by hostboot not pointing to the correct location during hcode_image_build.   

There was a similar problem in HB where Secure Boot changes caused the image rebuild using stale data, thus the failure (0 in image size).   A fix is being worked in SW417728.

stewart-ibm, 2018-05-03:
From reading everything here, I'll close this Issue as something that has been fixed through various means. We can re-open if it's spotted with latest code.

closed 2018-05-03 by stewart-ibm