open-power/op-build #1889: build gets stopped after generating root filesystem.

pridhiviraj, 2018-02-20:
I see lot of times the build gets stopped after generating root file system when we try to build a PNOR for a platform,
```
# Copy the kernel image(s) to its(their) final destination
/usr/bin/install -m 0644 -D /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/linux-4.14.16/arch/powerpc/boot/zImage.epapr /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/images/zImage.epapr
# If there is a .ub file copy it to the final destination
test ! -f /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/linux-4.14.16/arch/powerpc/boot/zImage.epapr.ub || cp /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/linux-4.14.16/arch/powerpc/boot/zImage.epapr.ub /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/images
>>>   Generating root filesystem image rootfs.tar
rm -f /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_fakeroot.fs
rm -f /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/target/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM
rm -f /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_users_table.txt
echo '#!/bin/sh' > /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_fakeroot.fs
echo "set -e" >> /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_fakeroot.fs
echo "chown -h -R 0:0 /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/target" >> /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_fakeroot.fs
printf '   	- - input -1 * - - - Input device group\n\n' >> /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_users_table.txt
PATH="/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/bin:/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/var/perf/pm/bin:/root/bin" /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/buildroot/support/scripts/mkusers /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_users_table.txt /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/target >> /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_fakeroot.fs
cat ../openpower/device_table.txt > /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_device_table.txt
printf '   	/bin/busybox                     f 4755 0  0 - - - - -\n /dev/console c 622 0 0 5 1 - - -\n\n' >> /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_device_table.txt
echo "/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/bin/makedevs -d /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_device_table.txt /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/target" >> /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_fakeroot.fs
printf '   	(cd /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/target; find -print0 | LC_ALL=C sort -z | tar  -cf /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/images/rootfs.tar --null --no-recursion -T - --numeric-owner)\n' >> /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_fakeroot.fs
chmod a+x /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_fakeroot.fs
PATH="/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/bin:/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/var/perf/pm/bin:/root/bin" /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/host/bin/fakeroot -- /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_fakeroot.fs
rootdir=/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/target
table='/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/build/_device_table.txt'
/usr/bin/install -m 0644 support/misc/target-dir-warning.txt /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/target/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM
make: Leaving directory '/home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/buildroot'
[root@system op-build]# echo $?
0
[root@system op-build]# 
```

pridhiviraj, 2018-02-20:
It should build all the images and also build PNOR at the end.

sammj, 2018-02-20:
What do you get from
`grep BR2_DEFCONFIG /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/.config`
after such a build?

pridhiviraj, 2018-02-21:
```
# grep BR2_DEFCONFIG /home/jenkins_data/sath/pridhivi/boston_p9dsu_opbuild/op-build/output/.config
BR2_DEFCONFIG="/var/www/html/sath/pridhivi/boston_p9dsu_opbuild/op-build/openpower/configs/p9dsu_defconfig"
```