open-power/op-build #1946: Mismatch in PHB's under chip1 with upstream PNOR and official released SMC PNOR

pridhiviraj, 2018-03-08:
```
With official SMC released PNOR:
=====================================
[  199.196649672,5] PHB#0000:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..01 SLOT=UIO Slot1 
[  199.205840123,5] PHB#0000:01:00.0 [EP  ] 15b3 1019 R:00 C:020700 (       network) LOC_CODE=UIO Slot1
[  199.215402871,5] PHB#0000:01:00.1 [EP  ] 15b3 1019 R:00 C:020700 (       network) LOC_CODE=UIO Slot1
[  199.222910146,5] PHB#0001:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..09 SLOT=PLX 
[  199.232343639,5] PHB#0001:01:00.0 [SWUP] 10b5 8725 R:ca C:060400 B:02..09 LOC_CODE=PLX
[  199.238180468,5] PHB#0001:02:01.0 [SWDN] 10b5 8725 R:ca C:060400 B:03..03 SLOT=S000103 
[  199.245118086,5] PHB#0001:03:00.0 [EP  ] 1000 00c9 R:01 C:010700 (           sas) LOC_CODE=S000103
[  199.252103144,5] PHB#0001:02:08.0 [SWDN] 10b5 8725 R:ca C:060400 B:04..08 
[  199.261927683,5] PHB#0001:02:09.0 [SWDN] 10b5 8725 R:ca C:060400 B:09..09 SLOT=S000109 
[  199.269628966,5] PHB#0001:09:00.0 [EP  ] 8086 1589 R:02 C:020000 (      ethernet) LOC_CODE=S000109
[  199.279671348,5] PHB#0001:09:00.1 [EP  ] 8086 1589 R:02 C:020000 (      ethernet) LOC_CODE=S000109
[  199.288411766,5] PHB#0001:09:00.2 [EP  ] 8086 1589 R:02 C:020000 (      ethernet) LOC_CODE=S000109
[  199.298710312,5] PHB#0001:09:00.3 [EP  ] 8086 1589 R:02 C:020000 (      ethernet) LOC_CODE=S000109
[  199.308745380,5] PHB#0001:01:00.1 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) LOC_CODE=PLX
[  199.314231853,5] PHB#0001:01:00.2 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) LOC_CODE=PLX
[  199.323166595,5] PHB#0001:01:00.3 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) LOC_CODE=PLX
[  199.329778713,5] PHB#0001:01:00.4 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) LOC_CODE=PLX
[  199.340275051,5] PHB#0002:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..01 SLOT=UIO Slot3 
[  199.345215298,5] PHB#0002:01:00.0 [EP  ] 1000 00c9 R:01 C:010700 (           sas) LOC_CODE=UIO Slot3
[  199.353224160,5] PHB#0003:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..01 SLOT=Onboard SAS 
[  199.360214311,5] PHB#0003:01:00.0 [EP  ] 9005 028d R:01 C:010700 (           sas) LOC_CODE=Onboard SAS
[  199.370542554,5] PHB#0004:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..02 SLOT=Onboard BMC 
[  199.379014833,5] PHB#0004:01:00.0 [ETOX] 1a03 1150 R:04 C:060400 B:02..02 LOC_CODE=Onboard BMC
[  199.392669099,5] PHB#0004:02:00.0 [PCID] 1a03 2000 R:41 C:030000 (           vga) LOC_CODE=Onboard BMC
[  199.401542263,5] PHB#0005:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..01 SLOT=Onboard USB 
[  199.407257299,5] PHB#0005:01:00.0 [EP  ] 104c 8241 R:02 C:0c0330 (      usb-xhci) LOC_CODE=Onboard USB
[  199.415822722,5] PHB#0030:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..01 SLOT=WIO Slot3 
[  199.424765328,5] PHB#0030:01:00.0 [EP  ] 15b3 1019 R:00 C:020700 (       network) LOC_CODE=WIO Slot3
[  199.433209870,5] PHB#0030:01:00.1 [EP  ] 15b3 1019 R:00 C:020700 (       network) LOC_CODE=WIO Slot3
[  199.441785378,5] PHB#0031:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..01 SLOT=WIO-R Slot 
[  199.448918818,5] PHB#0031:01:00.0 [EP  ] 1000 00c9 R:01 C:010700 (           sas) LOC_CODE=WIO-R Slot
[  199.458755051,5] PHB#0032:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..0d SLOT=PLX 
[  199.464344583,5] PHB#0032:01:00.0 [SWUP] 10b5 8725 R:ca C:060400 B:02..0d LOC_CODE=PLX
[  199.471739628,5] PHB#0032:02:01.0 [SWDN] 10b5 8725 R:ca C:060400 B:03..03 SLOT=S003203 
[  199.478611930,5] PHB#0032:03:00.0 [EP  ] 1000 00c9 R:01 C:010700 (           sas) LOC_CODE=S003203
[  199.488182724,5] PHB#0032:02:08.0 [SWDN] 10b5 8725 R:ca C:060400 B:04..08 
[  199.496884292,5] PHB#0032:02:09.0 [SWDN] 10b5 8725 R:ca C:060400 B:09..0d SLOT=S003209 
[  199.504837100,5] PHB#0032:01:00.1 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) LOC_CODE=PLX
[  200.000719166,5] PHB#0032:01:00.2 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) LOC_CODE=PLX
[  200.012473813,5] PHB#0032:01:00.3 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) LOC_CODE=PLX
[  200.023043749,5] PHB#0032:01:00.4 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) LOC_CODE=PLX
[  200.037724350,5] PHB#0033:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..01 SLOT=WIO Slot4 
[  200.046689507,5] PHB#0033:01:00.0 [EP  ] 11f8 f117 R:06 C:010802 (  mass-storage) LOC_CODE=WIO Slot4

With upstream PNOR(build from op-build):
============================================
[14268.141320870,7] PCI Summary:
[14268.141326475,5] PHB#0000:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..01 
[14268.144553886,5] PHB#0000:01:00.0 [EP  ] 15b3 1019 R:00 C:020700 (       network) 
[14268.148101224,5] PHB#0000:01:00.1 [EP  ] 15b3 1019 R:00 C:020700 (       network) 
[14268.152372630,5] PHB#0001:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..09 
[14268.155909719,5] PHB#0001:01:00.0 [SWUP] 10b5 8725 R:ca C:060400 B:02..09 
[14268.159450460,5] PHB#0001:02:01.0 [SWDN] 10b5 8725 R:ca C:060400 B:03..03 SLOT=S000103 
[14268.163033251,5] PHB#0001:03:00.0 [EP  ] 1000 00c9 R:01 C:010700 (           sas) LOC_CODE=S000103
[14268.167974986,5] PHB#0001:02:08.0 [SWDN] 10b5 8725 R:ca C:060400 B:04..08 
[14268.171538721,5] PHB#0001:02:09.0 [SWDN] 10b5 8725 R:ca C:060400 B:09..09 SLOT=S000109 
[14268.175076520,5] PHB#0001:09:00.0 [EP  ] 8086 1589 R:02 C:020000 (      ethernet) LOC_CODE=S000109
[14268.180036533,5] PHB#0001:09:00.1 [EP  ] 8086 1589 R:02 C:020000 (      ethernet) LOC_CODE=S000109
[14268.184298837,5] PHB#0001:09:00.2 [EP  ] 8086 1589 R:02 C:020000 (      ethernet) LOC_CODE=S000109
[14268.189271979,5] PHB#0001:09:00.3 [EP  ] 8086 1589 R:02 C:020000 (      ethernet) LOC_CODE=S000109
[14268.193531430,5] PHB#0001:01:00.1 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) 
[14268.197789287,5] PHB#0001:01:00.2 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) 
[14268.201354711,5] PHB#0001:01:00.3 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) 
[14268.205615323,5] PHB#0001:01:00.4 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) 
[14268.209876390,5] PHB#0002:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..01 
[14268.213403586,5] PHB#0002:01:00.0 [EP  ] 1000 00c9 R:01 C:010700 (           sas) 
[14268.216966368,5] PHB#0003:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..01 
[14268.220513738,5] PHB#0003:01:00.0 [EP  ] 9005 028d R:01 C:010700 (           sas) 
[14268.224099614,5] PHB#0004:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..02 
[14268.227609335,5] PHB#0004:01:00.0 [ETOX] 1a03 1150 R:04 C:060400 B:02..02 
[14268.231171047,5] PHB#0004:02:00.0 [PCID] 1a03 2000 R:41 C:030000 (           vga) 
[14268.235420397,5] PHB#0005:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..01 
[14268.238966263,5] PHB#0005:01:00.0 [EP  ] 104c 8241 R:02 C:0c0330 (      usb-xhci) 
[14268.242522638,5] PHB#0030:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..01 
[14268.246089105,5] PHB#0030:01:00.0 [EP  ] 15b3 1019 R:00 C:020700 (       network) 
[14268.249678686,5] PHB#0030:01:00.1 [EP  ] 15b3 1019 R:00 C:020700 (       network) 
[14268.253879258,5] PHB#0031:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..01 
[14268.257426155,5] PHB#0031:01:00.0 [EP  ] 1000 00c9 R:01 C:010700 (           sas) 
[14268.260978314,5] PHB#0032:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..0d 
[14268.264596441,5] PHB#0032:01:00.0 [SWUP] 10b5 8725 R:ca C:060400 B:02..0d 
[14268.268093632,5] PHB#0032:02:01.0 [SWDN] 10b5 8725 R:ca C:060400 B:03..03 SLOT=S003203 
[14268.272337585,5] PHB#0032:03:00.0 [EP  ] 1000 00c9 R:01 C:010700 (           sas) LOC_CODE=S003203
[14268.276603787,5] PHB#0032:02:08.0 [SWDN] 10b5 8725 R:ca C:060400 B:04..08 
[14268.280152714,5] PHB#0032:02:09.0 [SWDN] 10b5 8725 R:ca C:060400 B:09..0d SLOT=S003209 
[14268.284408777,5] PHB#0032:01:00.1 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) 
[14268.288665552,5] PHB#0032:01:00.2 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) 
[14268.292227532,5] PHB#0032:01:00.3 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) 
[14268.296475637,5] PHB#0032:01:00.4 [EP  ] 10b5 87d0 R:ca C:088000 (system-peripheral) 
[14268.300735089,5] PHB#0033:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:01..01 
[14268.303586325,5] PHB#0033:01:00.0 [EP  ] 11f8 f117 R:06 C:010802 (  mass-storage) 
[14268.307846450,5] PHB#0034:00:00.0 [ROOT] 1014 04c1 R:00 C:060400 B:00..00 
```

pridhiviraj, 2018-03-08:
With official SMC released PNOR:
============================
chip0 has 6 PHB's and chip1 has 4 PHB's

With upstream PNOR:
====================
chip0 has 6 PHB's and chip1 has 5 PHB's

mikey, 2018-03-08:
I thought the second chip on boston only had 4 PHBs used, not 5. Did the XML change accidentally?