open-power/op-build #1977: op-build fails to build devel to prod key transition PNOR when pass_on_validation_error=n

pridhiviraj, 2018-03-20:
```
13:31:50 //============= Generate PNOR Image Settings ===========//
13:31:50 PNOR Layout = /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/build/openpower-pnor-b210f15c69933e21494323a8f5017501e7b2c1de/p9Layouts/defaultPnorLayout_64.xml
13:31:50 Emit ECC-less versions of output files, when possible = No
13:31:50 Test Mode = No
13:31:50 Secureboot = Enabled
13:31:50 Sign Mode = development
13:31:50 Key Transition Mode = production
13:31:50 Lab security override (valid for SBE partition only) = Yes
13:31:50 //======================================================//
13:31:50 
13:31:50 Loading bin files...
13:31:50 TRACE: manipulateImages
13:31:50 HBBL raw size (no padding/ecc) = 19400/20480
13:31:50 TRACE: cp /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HBBL.staged /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//rand-2869566857.HBBL.temp.bin
13:31:50 TRACE: dd if=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//rand-2869566857.HBBL.temp.bin of=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HBBL.staged ibs=20480 conv=sync
13:31:50 0+1 records in
13:31:50 40+0 records out
13:31:50 20480 bytes (20 kB) copied, 0.000242541 s, 84.4 MB/s
13:31:50 TRACE: dd if=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//imprintHwKeyHash conv=notrunc of=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HBBL.staged bs=1 seek=$((20416))
13:31:50 64+0 records in
13:31:50 64+0 records out
13:31:50 64 bytes (64 B) copied, 0.000195698 s, 327 kB/s
13:31:50 TRACE: /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/usr/bin//crtSignedContainer.sh --scratchDir /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch/  --hwPrivKeyA /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/etc/keys//hw_key_a.key --hwPrivKeyB /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/etc/keys//hw_key_b.key --hwPrivKeyC /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/etc/keys//hw_key_c.key --swPrivKeyP /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/etc/keys//sw_key_a.key  --flags 0x80000000 --sign-project-FW-token HBBL  --protectedPayload /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HBBL.staged --out /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//rand-2869566857.HBBL.temp.hdr.bin
13:31:50 --> crtSignedContainer.sh: Signing mode: local
13:31:50 --> crtSignedContainer.sh: Parsing INI file: /home/jenkins/userContent/opalci/stb/project.ini
13:31:50 --> crtSignedContainer.sh: Using existing cache dir: /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//SIGNTOOL_1521532603, created: Tue Mar 20 13:26:43 IST 2018
13:31:50 --> crtSignedContainer.sh: Using existing cache subdir: /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//SIGNTOOL_1521532603/HBBL
13:31:50 --> crtSignedContainer.sh: Generating signing requests...
13:31:50 --> crtSignedContainer.sh: Found signature for HW key A.
13:31:50 --> crtSignedContainer.sh: Found signature for HW key B.
13:31:50 --> crtSignedContainer.sh: Found signature for HW key C.
13:31:50 --> crtSignedContainer.sh: Found signature for SW key P.
13:31:50 --> crtSignedContainer.sh: Have signatures for keys A,B,C,P, adding to container...
13:31:50 --> crtSignedContainer.sh: Container HBBL build completed.
13:31:50 
13:31:50 Container validity check PASSED. Container verification check FAILED.
13:31:50 
13:31:50 Error running command: /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/usr/bin//crtSignedContainer.sh --scratchDir /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch/  --hwPrivKeyA /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/etc/keys//hw_key_a.key --hwPrivKeyB /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/etc/keys//hw_key_b.key --hwPrivKeyC /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/etc/keys//hw_key_c.key --swPrivKeyP /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/etc/keys//sw_key_a.key  --flags 0x80000000 --sign-project-FW-token HBBL  --protectedPayload /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HBBL.staged --out /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//rand-2869566857.HBBL.temp.hdr.bin. Nonzero return code of (256) returned.
13:31:50 Error running command: cd /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch/ && /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//genPnorImages.pl --binDir /home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch/ --systemBinFiles ATTR_TMP=EMPTY,CVPD=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//CVPD.staged,GUARD=EMPTY,VERSION=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//VERSION.staged,MEMD=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//MEMD.staged,OCC=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//OCC.staged,HBD=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HBD.staged,DJVPD=EMPTY,HBB=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HBB.staged,RINGOVD=EMPTY,CAPP=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//CAPP.staged,FIRDATA=EMPTY,SBE=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//SBE.staged,HCODE=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HCODE.staged,HBEL=EMPTY,IMA_CATALOG=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//IMA_CATALOG.staged,SBKT=EMPTY,WOFDATA=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//WOFDATA.staged,ATTR_PERM=EMPTY,HBBL=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HBBL.staged,NVRAM=EMPTY,HBRT=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HBRT.staged,BOOTKERNEL=/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/host/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//BOOTKERNEL.staged,SECBOOT=EMPTY,HDAT=EMPTY,MVPD=EMPTY,HBI=/home/jenkins/workspace/Stb_key_transition_dev [... cut ...]
13:31:50 package/pkg-generic.mk:305: recipe for target '/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/build/openpower-pnor-b210f15c69933e21494323a8f5017501e7b2c1de/.stamp_images_installed' failed
13:31:50 make: *** [/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/output/build/openpower-pnor-b210f15c69933e21494323a8f5017501e7b2c1de/.stamp_images_installed] Error 2
13:31:50 make: Leaving directory '/home/jenkins/workspace/Stb_key_transition_dev_to_prod/op-build/buildroot'
```
Build will work only when we have `export SB_PASS_ON_ERROR=y` in environment or set `pass_on_validation_error=y` in INI file.

pridhiviraj, 2018-03-20:
@hellerda We have earlier discussed this issue, I have raised this issue here, so that we will not loose the track of it.

dcrowell77, 2018-03-20:
@bofferdn - fyi

hellerda, 2018-03-20:
Thx Pridhiviraj.  The issue is that for transition containers of the dev_to_production type, the "to" and "from" containers are signed by different sets of HW keys, and so require different values for SB_VERIFY, which there is currently no way to provide.  We need a way to tell signtool to "use this HW keys hash for the from-container, and this different value for the to-container".  (The "to" container being the "inner" container of SBKT, i.e. SBKTRAND).

hellerda, 2018-03-20:
I see two ways to fix this:
1. Add a new config property to signtool, e.g. "SB_VERIFY_TRANSITION".  I'd rather not do this.
2. The better way IMO is to feed signtool a different SB_PROJECT_INI for the transition container.  How is this different than 1?  It's a more complete solution, future-proofing against the need to support production_to_production transition containers.  The INI file can hold the SB_VERIFY value but it can also hold a boatload more properties, to apply to the "to" container.  If we ever want to support production_to_production transition containers, we'd need to feed a bunch of different values for the "to" container, to allow it to interface with signframework and retrieve keys/sigs from a different signing project than the "from" containers.  Feeding signtool a completely different INI would solve all the requirements.

hellerda, 2018-03-20:
That said, I haven't yet determined the best way to introduce a separate INI.  I may need to add a new config property after all, something like "SB_PROJECT_INI_TRANSITION".  I'll work on it a bit and put together a test patch.

hellerda, 2018-04-02:
This should now be fixed with the latest commit: https://github.com/open-power/sb-signing-utils/commit/61a4f1bd4be7503d18aa1a73be091ff42f730e0a.  I added both SB_VERIFY_TRANS and SB_PROJECT_INI_TRANS.  The former is the simpler way to address the issue for Dev-to-Production so I suggest using that.  The latter is there in case we want to eventually support Production-to-Production.  (But you can use it for Dev-to-Production if you want).

I updated the wiki accordingly.  let me know if questions.  https://github.com/open-power/sb-signing-utils/wiki

hellerda, 2018-04-23:
We found in chat, you had the wrong SB_VERIFY value in your INI, and that overrides your correct value you set bye env var. Sorry, too many ways to set the same properties!! But hopefully good now, if you change it in your INI, or just comment it out of the INI and let env var set it.

pridhiviraj, 2018-04-23:
Thanks @hellerda for pointing the root cause for the failure, after using right value for SB_VERIFY by exporting it and commenting it in INI file, it worked fine. So closing this issue, as it is resolved,

closed 2018-04-23 by pridhiviraj

hellerda, 2018-04-24:
Thanks for raising it.

After giving it more thought I decided the best way to handle it is to fix it in the code.  We can never reuse FW key signatures for SBKTRAND or SBKT (since the latter *contains* the former) so I made crtSignedContainer.sh check the component id and never reuse FW key sigs for these.  So now it's OK to keep the cache, even for transition containers, and it works just like the other cases, except that it will always retrieve or generate new P,Q,R sigs for SBKTRAND and SBKT.  See  https://github.com/open-power/sb-signing-utils/pull/24.

While I'd prefer not to have component-specific code in the program, there's already a case where it checks for SBKTRAND (https://github.com/open-power/sb-signing-utils/commit/61a4f1bd4be7503d18aa1a73be091ff42f730e0a), so what the heck.