open-power/op-build #1993: HB failed to laod latest skiboot, system hangs at ISTEP 21.3

pridhiviraj, 2018-03-28:
```
  1.58140|ERRL|Dumping errors reported prior to registration
  3.34323|Ignoring boot flags, incorrect version 0x0
  3.59302|ISTEP  6. 3
  4.02684|ISTEP  6. 4
  4.02743|ISTEP  6. 5
  8.41109|HWAS|PRESENT> DIMM[03]=AAAA0000AAAA0000
  8.41110|HWAS|PRESENT> Membuf[04]=CC00000000000000
  8.41110|HWAS|PRESENT> Proc[05]=8000000000000000
 23.17721|ISTEP  6. 6
 23.26571|ISTEP  6. 7
 23.26647|ISTEP  6. 8
 23.29613|ISTEP  6. 9
 24.27559|ISTEP  6.10
 24.27611|ISTEP  6.11
 24.29593|ISTEP  6.12
 24.56318|ISTEP  6.13
 24.56432|ISTEP  7. 1
 24.64260|ISTEP  7. 2
 24.81029|ISTEP  7. 3
 24.83077|ISTEP  7. 4
 24.86842|ISTEP  7. 5
 25.04502|ISTEP  7. 6
 25.07335|ISTEP  7. 7
 25.08315|ISTEP  7. 8
 25.14522|ISTEP  7. 9
 25.16431|ISTEP  8. 1
 25.30522|ISTEP  8. 2
 25.30621|ISTEP  8. 3
 25.31639|ISTEP  8. 4
 25.31723|ISTEP  8. 5
 25.31776|ISTEP  8. 6
 25.93430|ISTEP  8. 7
 25.93477|ISTEP  8. 8
 25.94520|ISTEP  9. 1
 26.31981|ISTEP  9. 2
 26.81013|ISTEP 10. 1
 27.15954|ISTEP 10. 2
 27.91265|ISTEP 10. 3
 27.91320|ISTEP 10. 4
 27.91361|ISTEP 10. 5
 27.91753|ISTEP 10. 6
 27.91813|ISTEP 10. 7
 27.91861|ISTEP 10. 8
 28.00044|ISTEP 10. 9
 28.00179|ISTEP 10.10
 28.00280|ISTEP 10.11
 28.07132|ISTEP 10.12
 28.07186|ISTEP 10.13
 28.07242|ISTEP 10.14
 28.22297|ISTEP 11. 1
 28.25416|ISTEP 11. 2
 28.25480|ISTEP 11. 3
 28.40606|ISTEP 11. 4
 28.51318|ISTEP 11. 5
 28.60656|ISTEP 11. 6
 29.65578|ISTEP 11. 7
 29.65647|ISTEP 11. 8
 29.98767|ISTEP 11. 9
 30.12675|ISTEP 11.10
 30.18686|ISTEP 11.11
 30.18776|ISTEP 11.12
 34.63232|ISTEP 11.13
 34.63377|ISTEP 12. 1
 34.71747|ISTEP 12. 2
 34.78279|ISTEP 12. 3
 34.81170|ISTEP 12. 4
 35.06744|ISTEP 12. 5
 35.06814|ISTEP 13. 1
 35.37504|ISTEP 13. 2
 35.46936|ISTEP 13. 3
 35.49797|ISTEP 13. 4
 36.18937|ISTEP 13. 5
 36.19120|ISTEP 13. 6
 37.19141|ISTEP 13. 7
 37.37387|ISTEP 13. 8
 37.54316|ISTEP 13. 9
 40.51566|ISTEP 13.10
 40.53560|ISTEP 13.11
 40.63730|ISTEP 13.12
 40.63851|ISTEP 14. 1
 40.70403|ISTEP 14. 2
 40.71668|ISTEP 14. 3
 44.30843|ISTEP 14. 4
 44.33835|ISTEP 14. 5
 44.40414|ISTEP 14. 6
 44.40506|ISTEP 14. 7
 44.55716|ISTEP 14. 8
 44.55963|ISTEP 15. 1
 48.37827|ISTEP 15. 2
 48.38833|ISTEP 15. 3
 48.51131|ISTEP 16. 1
 49.68651|ISTEP 16. 2
 49.91728|ISTEP 16. 3
 49.94783|ISTEP 16. 4
 49.96167|ISTEP 18.13
 50.28273|ISTEP 18.14
 50.29307|ISTEP 21. 1
 56.64292|htmgt|User power limit has been set to 990W
 57.14594|htmgt|OCCs are now running in ACTIVE state
 67.48724|ISTEP 21. 2
 67.50806|ISTEP 21. 3
```
tried flashing below skiboot level.
```
0a1d25fbae1a4d4697214f8befdf65b2e1fef3a6
```

pridhiviraj, 2018-03-28:
With upstream habanero PNOR system booted fine and able to flash upstream skiboot and boots fine. 
After observed that  problem exists with official released latest PNOR(which has fixes for spectre/meltdown) we can't flash skiboot.lid.xz as HB regression of not having fix for loading xz compressed lid, and if we try to flash skiboot.lid (uncompressed one 1.3 MB) OPAL is not booting as 0.3MB skiboot image is cropping.

dcrowell77, 2018-03-28:
>  as HB regression of not having fix for loading xz compressed lid

I'm not familiar with what you are referring to here.  Can you point me to another issue?

pridhiviraj, 2018-03-28:
@dcrowell77 I am guessing previous official released PNOR's has support for loading xz compressed skiboot PAYLOAD during IPL. But with latest released PNOR it is getting hang at ISTEP 21.3 hence i am thinking it may be a regression in HB driver. But yes, still we need to confirm when that support was available in official drivers.

dcrowell77, 2018-03-29:
We haven't removed any of that support (at least not on purpose) so I'm at a loss as to why it wouldn't work.  We haven't really touched our P8 tree much in awhile.

If the system is truly hung (no attentions) then the only way to debug is using our debug framework on top of cronus.  Can you get a dump of our memory in the hang state?  Run output/build/hostboot-5fc3b529c69246a6706577351bccd7898f1c227b/src/build/debug/fsp-memdump.sh from inside a cronus shell.  Post the binary it produces along with the output/images/host_fw_debug.tar for the build that was running.

stewartsmith, 2020-11-09:
Since POWER8 and palmetto support was removed in 40f804058414ddf4518ac07866bb58af74db20a2, this can probably be closed now.

closed 2020-11-09 by klauskiwi