open-power/op-build #1994: pnor: Mark HBB/HBI partitions as read-only

anoo1, 2018-03-28:
This is a follow-up issue for commit https://github.com/open-power/pnor/commit/2f4e88c64aace62165122af6e5d336c7465c65be where the PNOR partitions were marked read-only, except for HBB/HBI due to some issues found during testing.

This issue is to resolve the problems found and mark HBB/HBI as read-only partitions, otherwise they can be modified on the PNOR chip.

stewart-ibm, 2018-03-28:
I know @amboar has been working on some mbox fun with r/w windows for r/o sections and solidifying that behavior. Do we know what the issues were?

dcrowell77, 2018-03-29:
If I remember correctly, it is tied to how our VMM works.  There is a strange (and likely incorrect) association between the PNOR permissions and our page permissions.  So something in the bowels of that was leading to the problem where we ended up having pages stuck in memory because we couldn't get rid of them.  My memory is pretty foggy though so this is a lot of speculation.

amboar, 2018-04-09:
For the record, the BMC-side patches @stewart-ibm mentioned (which are currently only relevant on Witherspoon) are now integrated into OpenBMC, though from the input above sound unlikely have any impact on the problem.

stewart-ibm, 2018-11-09:
Do we have all the pieces for this upstream now? Should we close this issue?