open-power/op-build #2079: AMD Polaris GPU using 32-bit DMA

madscientist159, 2018-05-01:
Not sure where to file this; can move to the proper location when it is found.

On our Talos II systems, Polaris GPUs fall back to 32-bit DMA.  This breaks the amdgpu driver and X in "interesting" ways, in many cases preventing the GPU from working properly or even at all with 3D applications.

My understanding is that a fix is in the works for this, involving the creation of an increase DMA window, but that this fix is not yet ready.

madscientist159, 2018-05-01:
https://bugs.freedesktop.org/show_bug.cgi?id=106258 may be related

luke-jr, 2018-05-05:
@stewart-ibm ping :)

madscientist159, 2018-07-07:
This was addressed in the following patches:

https://bugs.freedesktop.org/attachment.cgi?id=140257

https://lists.ozlabs.org/pipermail/linuxppc-dev/2018-June/174985.html

https://scm.raptorengineering.com/scm/#changesetPanel;6IQvJHpVSM;578e2d761554130f7c6abfe821c2509912a00ac6;commit

https://scm.raptorengineering.com/scm/#changesetPanel;6IQvJHpVSM;461333cedd470140eb1e1667608da2a6d65a58e5;commit

https://bugs.freedesktop.org/show_bug.cgi?id=107012

Currently awaiting upstreaming.

closed 2018-07-07 by madscientist159