open-power/op-build #2082: Logo Design

nunojesus, 2018-05-02:
Hi,

I have read your work so I decided to creat a logo for you. I hope you like it and you want to use it. I can give you all the formats of the design free. If you want to make changes please specify.

The logo concept are the letters O and P mixed together.

Greetings

[image]

matthiasbeyer, 2018-05-02:
Beware: this user opened The exact same post on one of my repositories which is named "op". Might be a scam.

nunojesus, 2018-05-02:
It´s not a scam dont worry, if you want the logo I will do it, if you dont want just say no and close this issue.

I contribute to many projects with logo designs,

greetings

matthiasbeyer, 2018-05-02:
Yes, that's why I wondered. I didn't want to offend you if this is legit, so apologies in this case. I hope you understand that this is rather unusual to the normal github user who gets maybe PRs with code, but never any free logo contributions.

If everything is legit, this is really great!

nunojesus, 2018-05-02:
You can see here 2 of my logo contributions, and you can see in the Readme file my name as logo designer:

https://github.com/nityanandagohain/SoundHaven
https://github.com/MarxSoul55/cats_vs_dogs

Greetings
Nuno

stewart-ibm, 2018-05-03:
Thanks for the idea and art, but I think if we end up having a use for a logo, we'll end up going with something consistent with the OpenPOWER branding through the foundation.

closed 2018-05-03 by stewart-ibm