open-power/op-build #2089: hcode package fails to build when no parent git directory present

sammj, 2018-05-04:
```
>>> hcode a3a060cd02c7ecab9abc660afc7f403d6aa54f7e Building
CONFIG_FILE=/home/sam/git/op-build/openpower/configs/hcode/ LD_LIBRARY_PATH=/scratch/sam/op-rom/host/usr/lib OPENPOWER_BUILD=1 CROSS_COMPILER_PATH=/scratch/sam/op-rom/host/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux PPE_TOOL_PATH=/scratch/sam/op-rom/host/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux PPE_PREFIX=/scratch/sam/op-rom/host/powerpc64le-buildroot-linux-gnu/sysroot/ppe42-binutils/linux/bin/powerpc-eabi- RINGFILEPATH=/scratch/sam/op-rom/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_binaries __EKB_PREFIX= CONFIG_IONV_FILE_LOCATION=/scratch/sam/op-rom/host/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_binaries/"ionv.bin" CONFIG_INCLUDE_IONV= bash -c 'cd /scratch/sam/op-rom/build/hcode-a3a060cd02c7ecab9abc660afc7f403d6aa54f7e && source ./env.bash && /usr/bin/make -j41 '
fatal: Not a git repository (or any parent up to mount point /scratch)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
Could not find git root
package/pkg-generic.mk:221: recipe for target '/scratch/sam/op-rom/build/hcode-a3a060cd02c7ecab9abc660afc7f403d6aa54f7e/.stamp_built' failed
make: *** [/scratch/sam/op-rom/build/hcode-a3a060cd02c7ecab9abc660afc7f403d6aa54f7e/.stamp_built] Error 255
make: Leaving directory '/home/sam/git/op-build/buildroot'
```

The hcode build command sources `env.bash` which has:
```
GIT_ROOT=`git rev-parse --show-toplevel`
if [ $? -ne 0 ]; then
    echo "Could not find git root"
    exit -1
fi
```
Which it then uses to set up some paths. However buildroot does not build packages in their git directories and so this fails unless the machine _happens_ to have a `.git` directory somewhere up the directory tree. Adding `/scratch/sam/.git` allows the hcode build to go ahead even though it's obivously incorrect*
The hcode package needs to be updated to set itself up relative to its build directory rather than assuming it's in a git tree.

\* and then host-libflash blows up later, possibly related to setting a bunch of git variables.

rjknight, 2018-05-04:
as a test, can you remove the lines you listed above from env.bash and change the following line from  
export PROJECT_ROOT=$GIT_ROOT
to
export PROJECT_ROOT=$ROOTDIR?

sammj, 2018-05-09:
Yep, looks like that works.
```
diff --git a/env.bash b/env.bash
index f1f7412..8be9627 100755
--- a/env.bash
+++ b/env.bash
@@ -29,15 +29,9 @@ if [ -e ./customrc ]; then
     source ./customrc
 fi
 
-GIT_ROOT=`git rev-parse --show-toplevel`
-if [ $? -ne 0 ]; then
-    echo "Could not find git root"
-    exit -1
-fi
-
 # Setup some global variables
 export PROJECT_NAME=HCODE
-export PROJECT_ROOT=$GIT_ROOT
+export PROJECT_ROOT=$ROOTDIR
 export TOOLSDIR=$PROJECT_ROOT/tools
 export PERLMODULES=$TOOLSDIR/perl.modules
 export HOOKSDIR=$PROJECT_ROOT/.git/hooks
```

rjknight, 2018-05-09:
Good deal, I'll get a fix pushed up in the morning

sammj, 2018-08-21:
This should be fixed now.

closed 2018-08-21 by sammj